onitake/gsl-firmware #3: Get firmware for Chuwi vi8

barni2000, 2016-02-02:
@sigboe has firmware for vi8

sigboe, 2016-02-02:
My firmware is formatted with tools made by @onitake and works with his drivers. But the firmware has wrong resolution and requires calibration in X.

barni2000, 2016-02-02:
but you have old style firmware

sigboe, 2016-02-02:
@barni2000 Looks like @onitake already has Chuwi Vi8 firmware in this repository.

barni2000, 2016-02-02:
That firmware is wrong.

sigboe, 2016-02-02:
@barni2000 okay I will post it tomorrow after work, @onitake wants the regular firmware as is from manufacturer right?

onitake, 2016-02-03:
@sigboe If possible, yes. Compatible firmware is ok, as long as it works, of course.

sigboe, 2016-02-03:
@onitake I can just zip both the firmware files I have with a description and post it here. After I get home of course. Then you can use which one you like your self and put them in the repository. 

Or if you really want, I can fork and do a pull request.

onitake, 2016-02-03:
@sigboe Sending a zip file is ok. What's the difference between those firmwares?

sigboe, 2016-02-03:
@onitake One I used fwtool from your other repo to decrypt and add 4095x4095 (or whatever the max was) resolution.

Have you looked more into reading out the original resolution from the firmware?

onitake, 2016-02-03:
@sigboe No, I haven't had time to do that. The GSL_TS_CFG.h "firmware" contains a block of values before the actual firmware. I'm positive these are device parameters, but I'm not sure what they mean exactly.
This also applies to the unscrambled SileadTouch.fw and probably also SileadTouch.sys, as I assume the same header file is used to compile the firmware.

sigboe, 2016-02-03:
@onitake Do you think they tried to obscure it intentionally?

onitake, 2016-02-03:
@sigboe I have no idea what they were thinking. In any case, I think it's really, really stupid to obfuscate firmware.

sigboe, 2016-02-03:
@onitake How did you deduce the XORpad? Maybe its another XORpad for that section?

onitake, 2016-02-03:
@sigboe I disassembled SileadTouch.sys and searched for a firmware loading routine.

sigboe, 2016-02-03:
@onitake If it is another XORpad, and its more advanced one like Nintendo uses to encrypt their games, then I would think the whole XORpad should be obviously visible in a dissembled driver. If it is a more simple one, like XOR every byte with a specific byte it might be harder to see.

onitake, 2016-02-03:
@sigboe The code didn't look very sophisticated to me. I was confused at first, because there were lots of "magic" values there, but then I realised this was just code that parses the header file and converts the hex values into binary. So, it's nothing complex at all.

They just XOR everything with 0x88, then skim over the result (which is exactly the content of GSL_TS_CFG.h) until they find "GSL_TS_CFG", then convert the hex values that follow into binary, and that's the firmware image.

The firmware image that's compiled into the driver however, is not scrambled at all. It's just what you would get if you compiled GSL_TS_CFG.h into a program.

sigboe, 2016-02-03:
[Touch_PG.zip](https://github.com/onitake/gsl-firmware/files/115940/Touch_PG.zip)

onitake, 2016-02-03:
Thanks. Committed as a19629d7.

closed 2016-02-03 by onitake