onitake/gsl-firmware #6: tools/fw_extractor my_android_gslx680.ko firmware.fw

youling257, 2016-03-07:
my tablet onda v891w，8.9 inch，copy this /lib/modules/gslx68x_ts.ko ，isn't system/vendor/modules/，tools/fw_extractor gslx68x_ts.ko  firmware.fw，not success

onitake, 2016-03-07:
Umm... No success?

I get this:
`tools/fw_extractor gslx68x_ts.ko 
Found fw_index offset 100340 count 4
Found gsl_load_fw offset 63040 count 933
Found GSLX680_FW_TEST offset 62240 count 7656
Found GSLX680_FW_gsl_customer offset 69952 count 39336
Found GSLX680_FW_I81_GSL3676B_8 offset 109312 count 37224
Found GSLX680_FW_I802_GSL3676B_ offset 146560 count 36960
Found GSLX680_FW_I802_GSL3676B_ offset 183520 count 40392
Found GSLX680_FW_I100_GSL3692_1 offset 223936 count 38808
Found GSLX680_FW_I89_GSL3676B_1 offset 262752 count 37224
Found GSLX680_FW_I71_GSL1686F_1 offset 300000 count 36960
Found FW_I89_GSL3676B_19201200_ offset 455296 count 40392
Found GSLX680_FW_I86_GSL3676B_8 offset 536128 count 39336
Found gslx680_fw_grp offset 62272 count 152`

All the GSLX680_FW* files look like proper firmware files. To find out which one to use, check which chip you have (you may have to open the device), then test the matching ones. The `I89` may hint to the panel size, so try those first, perhaps.

onitake, 2016-03-07:
Aaah, now I see your problem. The documentation is wrong. Sorry. Will fix that in a moment.

onitake, 2016-03-07:
Documentation fix applied in 537bf55639942f94d066078c3db15fec146955c1.

youling257, 2016-03-07:
./fwtool -c FW_I89_GSL3676B_19201200_.fw  -w 1920 -h 1200 -t 10 silead_ts.fw，it is right ？
developer options，show touches，show visual feedback for touches，enable，the finger click  position, touchscreen White dot ，Offset to the right，why?

but i use chuwi hi8 's firmware.fw，convert  silead_ts.fw，the finger click  position, touchscreen White dot ，Offset to the left，why? it's not calibration problem，can't calibration touchscreen

onitake, 2016-03-07:
I can't tell you if it's correct. I don't know the exact specs of your device. You need to test and find them out yourself.

Offsets or inaccessible areas are probably caused by incorrect touchscreen dimensions. These can be different from the display resolution. There is currently no reliable way to determine them, you need to do some experimentation.

If calibration using `xinput_calibrator` is not an option, then you need to set the correct parameters in the firmware. Use `xinput test <device number>` (find the number with `xinput list`) to see the all touch events, then move your finger around the corners a bit. This should give an idea what the dimensions of the touchscreen has. Set those using `fwtool -w <width> -h <height>`.

If you don't get multitouch events, or if the axes seem to be swapped or mirrored, set the corresponding parameters using `fwtool -f`.

youling257, 2016-03-07:
oh，no，i don't know how to do，-m -w -h ，correct parameters，oh ，no，i don't know

onitake, 2016-03-07:
As I said, some experimentation is necessary.

Convert the firmware using default parameters, then test using `xinput test`. Once you've found out the approximate width and height of the the touch panel, set those parameters using `fwtool -w <width> -h <height>`. If you encounter any problems like a mirrored axis or broken multitouch, set the appropriate flag using `fwtool -f <flag>`.

I'm sorry if I can't help you any further.

youling257, 2016-03-07:
how to use fwtool -f ? please example

onitake, 2016-03-08:
Let's say your touchscreen is mirrored on the horizontal axis, i.e. when you touch on the left, it registers on the right and vice-versa. In this case, you would use: `fwtool -f xflip`

Or, you notice that multitouch does not work. In this case, you can try enabling finger tracking: `fwtool -f track`

If you want to combine the options, separate them by a comma: `fwtool -f xflip,track`

onitake, 2016-03-08:
The other options are:
```
-f    Sets flags, separated by commas (optional, supported flags: xflip, yflip, swap, track)
      xflip enables horizontal flipping
      yflip enables vertical flipping
      swap enables axis swapping
      track enables in-driver finger tracking (use for controllers that don't support it)
      Each flag may be prefixed with 'no' to disable it.
```

youling257, 2016-03-08:
Up and down，Upside down ; left  right，not Upside down. how to fix ?

onitake, 2016-03-08:
Ok, I think I'm beginning to understand what your problem is. You seem to be experiencing a mirrored y axis.

Try this:
```
./fwtool -c FW_I89_GSL3676B_19201200_.fw -w 1920 -h 1200 -t 10 -f yflip silead_ts.fw
sudo cp silead_ts.fw /lib/firmware
```

Do you also need help with the driver installation?

youling257, 2016-03-08:
I know  modprobe -r gslx680_ts_acpi，modprobe gslx680_ts_acpi

onitake, 2016-03-08:
Ah, very good.

So, just create silead_ts.fw, copy to /lib/firmware, and you should be good to go. If it doesn't work, try the same procedure with the other firmwares.

youling257, 2016-03-08:
ok，already  vertical flipping，but -w -h parameter still incorrect，i can't understand xinput test result.

youling257, 2016-03-08:
not  perfect，click not Precision，I do these，in oder to on Android x86 6.0.1，use touchscreen，still has problem，already use calibration app ，Quick Settings drop-down and Side slide，can't do this

youling257, 2016-03-08:
This is Android x86 6.0.1，http://i.imgur.com/ZDq4PKB.png  http://i.imgur.com/fcUALEc.png

onitake, 2016-03-09:
Nice, at least there's some progress.

`xinput test` will print the coordinates of all input events. If you move your finger across the screen, especially around the edges, you will see the maxima for the pixel coordinates. That way, you should be able to determine the accurate values for -w and -h.

I don't have a working test device right now, so I can't provide you with example output. But I'll post here when I have some.

youling257, 2016-03-10:
can't find correct parameters，it must be -w 1920 -h 1200，because 
“This is Android x86 6.0.1，http://i.imgur.com/ZDq4PKB.png http://i.imgur.com/fcUALEc.png
This is dual os 4.4.4，http://i.imgur.com/hBrmVNi.png http://i.imgur.com/KMdpGyL.png”
whatever i do ，every time boot Ubuntu，must calibration touchscreen
every time boot Androidx86 6.0.1(www.android-x86.org)，need calibration touchscreen，it's can't，can't calibration

youling257, 2016-03-11:
oh，no，The left side，not offset，the finger  from left side to right side，offset  bigger and bigger

onitake, 2016-03-11:
If you have an offset or scaling Problem, then the -w 1920 -h 1200 is most likely NOT correct.

Try smaller values, like -w 1024 -h 600 for example. Or test using `xinput test` as I described above to determine the correct values.

youling257, 2016-03-11:
i find my tablet SileadTouch.fw，I890_3676B_19201200_PG_WJ_WJ888_Drivers-20150604.zip，then 
./fwtool -c SileadTouch.fw -3 -w 1920 -h 1200 -t 10 silead_ts.fw，
Up and down，Upside down ; left right，also Upside down；how to do ?
-w 1920 -h 1200 -f yflip xclip ?what terminal command ?

onitake, 2016-03-11:
To enable both horizontal and vertical mirroring, use this:
`./fwtool -c SileadTouch.fw -3 -w 1920 -h 1200 -t 10 -f xflip,yflip silead_ts.fw`

youling257, 2016-03-11:
This SileadTouch.fw has more problem，Not as good as FW_I89_GSL3676B_19201200_.fw

youling257, 2016-03-11:
no matter what parameters， xinput test ， top left corner ，a[0]=20，bottom left corner ，a[0]=40

onitake, 2016-03-12:
Hmm I see... But that could just mean that the firmware does not work with your chip.

Can you open your device and check if you really have a GSL3676B?

youling257, 2016-03-12:
what mean ? extract gslx68x_ts.ko，get GSL3676B firmware，gslx68x_ts.ko from onda v891w dual-os Android 4.4.4，what mean “open your device” ?

onitake, 2016-03-12:
I mean, you should disassemble your tablet physically and look inside what chips you have on the circuit board. Check which Silead chip you have exactly (GSL3676B, GSL2681, GSL1680, ....) to make sure you use the correct firmware.

youling257, 2016-03-12:
yes，onda v891w，gsl 3676，Disassemble is my hobby，Just bought，first to do disassemble 
 like this，
http://i3.bbs.fd.zol-img.com.cn/t_s800x5000/g2/M00/0E/04/Cg-4WVWuCTqISLsCAAIWZNEp7FUAAHclAOOEHwAAhZ8023.jpg
http://bbs.zol.com.cn/quanzi/d26_795_uid_a757338903.html

onitake, 2016-03-12:
Ok, ok. Very good! :)

If you use the wrong firmware for the chip, you will get very strange results. Inaccuracy is normal, however. The chip sends out touch events without much interpolation, and gslx680-acpi doesn't interpolate either. I know that this leads to all kinds of problems, but I assumed that user space would handle this better. Unfortunately, that's not the case.

My test environment is Ubuntu 15.10, where basic stuff like touching buttons and typing work more or less, but dragging is a big problem, and I had lots of reliability problems close to the edges.

Unfortunately, I don't know a good way to solve these.

If you need a quicker solution, better try this driver here: https://github.com/rastersoft/gsl1680
It uses the FW_I89_GSL3676B_19201200_.fw firmware directly (no need to convert with fwtool), but runs in user space and requires some manual setup for the GPIO pins.

youling257, 2016-03-16:
offset 109312 count 37224
offset 146560 count 36960
offset 183520 count 40392
offset 223936 count 38808
offset 262752 count 37224
offset 300000 count 36960
offset 455296 count 40392
offset 536128 count 39336
what mean ?

onitake, 2016-03-16:
`Found <n> offset <x> count <y>`

That means it found a firmware image at file offset <x>, of length <y> bytes. The firmware will then be written out to `<n>.fw`.

youling257, 2016-03-16:
Found FW_I89_GSL3676B_19201200_ offset 455296 count 40392，
./fwtool -c FW_I89_GSL3676B_19201200_.fw -w ？ -h ？，Is there any relationship between the two?

onitake, 2016-03-16:
I don't know, really.

The only known fact is that the limit for -w and -h is 4095, because they're 12 bit values. Other than that - unknown...

youling257, 2016-03-16:
https://github.com/sigboe/gslX68X，“You will need to place a firmware file at /lib/firmware/[hid].fw where you replace [hid] with the HID ID of your device.”  ，I can't understand

onitake, 2016-03-17:
Yes, that part is a bit unfortunate. [HID] is the ACPI or DT ID of the device.
In most cases, this is `MSSL1680`, but you'll have to check.

youling257, 2016-03-18:
gslx680_ts_acpi.ko，need silead_ts.fw ; silead.ko need what fw ?

onitake, 2016-03-18:
Please don't mix up the drivers.

gslx680_ts_acpi.ko (from gslx680-acpi) needs silead_ts.fw created with fwtool.

silead.ko (from gslX68X) needs the firmware directly as it comes out of fw_extractor, and you need to rename it to [hid].fw, where [hid] is usually mssl1680. If mssl1680.fw does not work, look at the dmesg output. It will tell you which file it's expecting.

youling257, 2016-03-19:
https://github.com/rastersoft/gsl1680   ，how to launch the driver ?

gslx680_ts_acpi.ko and silead_ts.fw，only one point，can't Multi-touch；OK，I don't ask for more，Then basic Double finger zoom，I want this.

barni2000, 2016-03-19:
If only touch it at one point, then need track flag when you are generating firmware for gslx680_ts_acpi.ko.

holzingerm, 2016-03-19:
as superuser (linux normally sudo or su)

`modeprobe gslx680_ts_acpi`

if the module is installed correctly than it would load
during our discussion on android-x86 i would say you shall open the terminal app
type in su to get super user permissions and type in the command (modeprobe....) 

tanks for the tip with flag to get multitouch working will try it also on my side

regards
Mario

youling257, 2016-03-19:
barni2000  ，please tell me Terminal Command

barni2000, 2016-03-19:
fwtool

youling257, 2016-03-19:
I can't speak english，I do not proficient linux，you say fwtool  more，Please tell me for each terminal command，i only can install Ubuntu and Androidx86 on my tablet，，barni2000

holzingerm, 2016-03-19:
Linux:
`sudo modprobe gslx680_ts_acpi`

Android: terminal app
`su
modprobe gslx680_ts_acpi`

youling257, 2016-03-19:
holzingerm  ， I know who are you，The basic load drive command, I knew ；now the  problem，Double finger zoom，xposed GravityBox [MM] pie control……

barni2000, 2016-03-19:
Something like this:
./fwtool -c SileadTouch.fw -3 -w 1920 -h 1200 -t 10 -f xflip,yflip,track silead_ts.fw

holzingerm, 2016-03-19:
is the -3 really correct??
i try to help youling257 but with android drivers I need to confert the android drivers like 

`./fwtool -c GSLX680_FW_I81_GSL3676B_8.fw -1 -m 3676 -h 1920 -w 1200 -t 10-f xflip,yflip,track silead_ts.fw`

to get a valid output

barni2000, 2016-03-19:
i just copy paste an comment from above and write to flags the track

youling257, 2016-03-19:
use xinput，only can get probably parameters，Inaccurate, imprecise，This is not enough；you know，The touch screen on Androidx86 is must be more accurate than the mouse，must Precise and accurate parameters，can't get.
I say，can't do this ，http://i.imgur.com/oKPQBWr.png ，because parameters is Inaccurate ，imprecise

i can tell your these，repeat

youling257, 2016-03-19:
my tablet，onda v891w 32GB，single Windows 8.1，copy This file，/Windows/system32/drive/SileadTouch.fw，then 
./fwtool -c SileadTouch.fw -3 -w 1920 -h 1200 -t 10 -f yflip,track silead_ts.fw，This silead_ts.fw
./fwtool -c FW_I89_GSL3676B_19201200_.fw -w 1920 -h 1200 -t 10 -f yflip,track silead_ts.fw ，
Two silead_ts.fw, the same problem, the same effect

holzingerm, 2016-03-19:
using screen resolution on my tablet for fwtool needs also a calibration that is far away from beeinng perfect

today I played around with width and high valuses to get gslx working without calibration
now the accuracy is near perfect also on the edges for swipe in commands
let us play with these value

on my side when the value is to large the touch input is slower than the real finger ==> reduce the value
when the touch input is faster ==> increase the value

on my Trekstor tablet wit 1024x600 screen resolution I need to mad the values to 900x640 to work best

youling257, 2016-03-19:
gslx680_ts_acpi.ko and silead_ts.fw，Basic click、long press、up and down around the slide，can do this，I Have to say，No practical significance . I need Sliding edge， Double finger zoom.
gslx680_ts_acpi.ko and silead_ts.fw ，Not as good as with the mouse.

youling257, 2016-03-19:
holzingerm，please add https://github.com/rastersoft/gsl1680 https://github.com/sigboe/gslX68X to https://sourceforge.net/p/android-x86/kernel/ci/kernel-4.4/tree/ ，test these on Androidx86

holzingerm, 2016-03-19:
sorry but I'm only a little user not an admin of android-x86. :-D

I got some hardware and try to get it work by combining solutions from others like this acpi driver.
only sometimes I try to adopt simple code fraqments for my situation.

youling257, 2016-03-19:
staging	2016-01-27	Mario Holzinger Mario Holzinger	[81c4f1] staging: add GSLx680 I2C touch controller，like this，add gsl1680、gslx68x to Androidx86 kernel 4.4 tree.

holzingerm, 2016-03-19:
let us first create a working firmware file for you 
and after that I will try to write an patch for android-x86 
to push the correct firmware file depending on the device code to the kernel module

holzingerm, 2016-03-19:
so from your comments I see that you get the driver working
also the directions shall work so far, am I right
only the accuracy is bad for booth touch axes

onitake, 2016-03-19:
-3 is for Windows driver firmware. -1 is for firmware extracted from Android drivers using fw_extractor.

silead.ko does not support software finger tracking, so if you have a chip/firmware that doesn't do finger tracking by itself, you can't use silead.ko.

youling257, 2016-03-19:
https://github.com/rastersoft/gsl1680 ，please example，launch the driver

[35 more replies]