onitake/gsl-firmware #10: Chuwi HI8

leandrofina, 2016-04-12:
Hello , I have seen your files on github and would like to try as my touchscreen on the tablet no longer works. so far I have everything imaginable tried and unfortunately without success. the silead touch drivers I can not install with double drivers, there is always an error . 

can you tell me how I install your files best?

I am grateful to you for any help
 
Leandro

onitake, 2016-04-12:
Can you tell me which drivers you're currently using and what you mean with "double drivers" exactly?

To help pinpoint the issue, please create a gist/pastebin/etc. with your dmesg output and link it here.

leandrofina, 2016-04-12:
ok that is now unfortunately a bit incomprehensible to me, since I'm not in English and also possess no great idea of computers. My goal was to clean install again Windows 10. after that I wanted to install with the double driver program, any missing drivers that I have downloaded from the Chuwi homepage. but unfortunately this fails. DoubleDrivers: http://www.majorgeeks.com/files/details/double_driver.html

onitake, 2016-04-12:
Ok. I did not know Double Driver was a software product, I thought you had installed *two* different drivers on your system.

I'm sorry, I don't think I can help you. My two projects ([gsl-firmware](https://github.com/onitake/gsl-firmware) and [gslx680-acpi](https://github.com/onitake/gslx680-acpi)) are purely concerned with *Linux* support for the Silead touch screen controllers.

You can try installing the driver here: https://github.com/onitake/gsl-firmware/tree/master/firmware/chuwi/hi8/KMDF%20HID%20Minidriver%20for%20Touch%20I2C%20Device
But I can't guarantee that it works.

I think you need to copy `SileadTouch.fw` to `c:\windows\system32\drivers\`, then use the Windows Device Manager to install the driver. Just right-click the touch screen device (should have an exclamation mark), then select `Update driver software...`, then specify the location where you've downloaded the files from the link above.

closed 2016-05-29 by onitake