onitake/gsl-firmware #19: fwtool can't find/load Silead.pm

IamHadriel, 2016-06-16:
I'm not sure what I'm doing wrong but when I run `fwtool -c FW_I89_GSL3676B_19201200_.fw -w 1665 -h 1140 -t 10 -f yflip,track silead_ts.fw` perl outputs that it can't find the Silead.pm in any of the INC directories even though afaik it should be using the pm from the Firmware folder, no?

onitake, 2016-06-16:
perl is a bit of a nitpick when it comes to loading modules.
You need to to cd into the folder where Silead.pm is, or set PERL5LIB to the appropriate location.

onitake, 2016-06-16:
Note that Silead.pm is imported as `Firmware::Silead`, so don't cd into `Firmware`, but into `tools`.

IamHadriel, 2016-06-17:
That's what is confusing me. I was in tools when I ran it but when I get home I'll give it another go and provide the exact output perl is giving me

IamHadriel, 2016-06-17:
Well I got it working by changing fwtool to take Silead.pm from the working directory and placed all pertinent files in the same folder and touch works (woo!) However I fear that either I need to flip the x axis or something else because I touch one point in the screen and it's actually selecting another. 

The current issue is similar to when windows updated to the newer driver for the touchscreen which is meant for the newer version (DualOS) of the v891w whereas I have the Single OS version (v2 to be precise)

onitake, 2016-06-17:
If touch point mapping seems random, I think you have the wrong firmware.
Of course, there's no harm in trying xflip first.

onitake, 2017-03-03:
I'll close this issue now, if you have particular questions or issues with your device, please open a new one.
However, I'm a but busy currently and won't have much time to look into it.

closed 2017-03-03 by onitake