onitake/gsl-firmware #20: cube i10w

kahrloz, 2016-07-05:
i got fw_extractor to work off my android-tablet's.ko, however i don't know how to make use of the fw files so that my touchscreen works.

onitake, 2016-07-05:
I responded to your question in [gslx680-acpi#16](https://github.com/onitake/gslx680-acpi/issues/16), but feel free to continue the discussion here.

onitake, 2016-07-06:
In response to https://github.com/onitake/gslx680-acpi/issues/16#issuecomment-230628395 :
Have you tried all the firmware files, and did you enable finger tracking?
What was your fwtool command line?

onitake, 2016-07-11:
I downloaded the Windows driver you posted and converted it.
It's here: https://github.com/onitake/gsl-firmware/tree/master/firmware/cube/i10
The result is possibly not the same as yours.
Can you test that and report your results?

kahrloz, 2016-07-11:
OK

kahrloz, 2016-07-11:
It's works perfectly

closed 2016-07-11 by onitake