onitake/gsl-firmware #21: question

cubelibre, 2016-08-07:
hi . you write a script `tools/untscfg GSL_TS_CFG.h firmware.fw` but how about script `firmware.fw -> GSL_TS_CFG.h` ?

onitake, 2016-08-09:
Umm... Forgive me for asking, but what do you need that for?
I didn't think such a tool was useful, so I didn't implement it.

On top of that, the configuration data (gsl_config_data_id) is skipped while converting, so it's lost in the process. You may be able to restore the GSL_TS_CFG section, however.

That's fairly easy, even:
``` 
print "TS_CFG_DATA GSL_TS_CFG[] =  {" + newline
open "firmware.fw"
while firmware has data:
  print "{"
  read 4 bytes from firmware as 32 bit little endian integer
  print as hex string
  print ","
  read 4 bytes from firmware as 32 bit little endian integer
  print as hex string
  print "}," + newline
print "};"

cubelibre, 2016-08-09:
Excuse me, but I do not understand the perl scrypt :( Need make the dump firmware from kernel and convert it to header for rebuild new one kernel. Where header looks like 
```
static const struct fw_data GSLX680_FW[] = {

{0xf0,0x3},
{0x00,0xf1578d12},
{0x04,0x00000000},
{0x08,0x00000000},
{0x0c,0x00000000},
.........
```
I think I found a firmware in the kernel . Need to check , maybe im wrong .

onitake, 2016-08-10:
That's not a Perl script, just pseudo-code to give you a hint how it's done. It's pretty straight-forward, really. The firmware image is just a binary blob and represented almost as-is in the header file.
The question is, why would you want to use it like this at all?

Also, you can use the fw_extractor tool to extract firmware images directly from the driver. Did you try that?

cubelibre, 2016-08-10:
The problem is perl scrypt "fw extractor" working only with the modules , and don't know the kernel .
Like i say , need make the dump from kernel and convert to header for rebuild new kernel .
uploaded kernel https://ufile.io/5105d The firmware start from offset 701C50 and end 70B2E0 .
Maybe i'm wrong and not understand that GSL driver .

onitake, 2016-08-11:
Ok, I see.

Can you give me a bit more information about what hardware (tablet? CPU architecture?) you have and what you are trying to do with it?

You may be much better off with one of the newer drivers (([gslx680-acpi](https://github.com/onitake/gslx680-acpi) or  [silead.c](https://git.kernel.org/cgit/linux/kernel/git/dtor/input.git/commit/?id=3197704cae5e00d9c78563d2fa4647b576a49fef) ) - but if you really need the legacy Silead driver, I can help you with the conversion, of course.

cubelibre, 2016-08-11:
Name of tablet : HUU H7 LEAP
Hardware 
- 7 Inch Screen, 1024×600 Pixel
- CPU RK3188T Quad Core  1.4GHz
- 1GB RAM + 8GB ROM
- 2MP Back Camera + 0.3MP Front Camera

onboard ( gs_mma7660 , act8931 , rtc_hym8563 , gslX680 , es8323 , rtl8723BU , siv130b , siv121d)
web page : www.myhuu.com.cn
about cpu http://www.rock-chips.com/a/en/products/RK31_Series/2013/0808/314.html

Need change the backlight , add support (ntfs,udf,md4) and maybe another driver and fixes .

arsonoff, 2016-09-15:
Same problem:
Tablet on Allwinner A64. 

There is a module from an Android — gslX680new.ko (containing GSL1680E_86PG_FW.fw and other firmware).

Are available a kernel Linux with the same driver, but without the necessary firmware -- https://github.com/longsleep/linux-pine64/tree/pine64-hacks-1.2/drivers/input/touchscreen/gslx680new

```
#modprobe  gslX680new.ko
[  273.586685] ctp_fetch_sysconfig_para: ctp_power_io is invalid.
[  273.593895] gslx680: no matched TP firmware(gsl1680e_86PG)!
```

How to transfer firmware to the kernel module?

onitake, 2016-09-17:
@arsonoff That's probably this driver here: https://github.com/pcduino/modules/blob/master/touch/gsl/gslX680.c
I don't give support for that, please look at the source code and find out for yourself where to put the firmware. Perhaps it supports loading from /lib/firmware, perhaps you need to recompile the driver. I don't know.

However: If your target is ARM(64), I recommend Robert Dolca's driver instead. It will be included in mainline Linux 4.8, and it loads all device parameters from the DeviceTree. I'm not very familiar with ARM Linux myself, but it should be fairly easy to add DeviceTree blocks if they're missing.

onitake, 2016-09-17:
And to everyone else: I received another request for a conversion tool than can convert all formats back and forth, so I'm rewriting fwtool. But it will take some time, as I'm a bit busy at the moment. Sorry.

onitake, 2022-09-27:
It's unlikely that I'll be working on this further.
If anybody would like to fix this themselves and submit a pull request, please go ahead.

closed 2022-09-27 by onitake