onitake/gsl-firmware #26: firmware mssl1680.fw

Ilia1, 2016-12-28:
How to convert firmware from windows to linux for mssl1680 controller? File format is different or not? It is possible through the hex editor to change the title to mssl1680 and prescribe the magic number? How do you calculate the magic number, and where prescribed?

browerbr, 2017-02-13:
Hi!
I would like to know too. If someone could help...
I have a CCE TF74W (Drivers for win 8.1, including touch here) - SoC Atom

http://suporte.cce.com.br/produto/Tablet/TF74W/Download

And it uses SileadTouch.Inst, ACPI\MSSL1680.

I installed Ubuntu modded from linuxium.com.au but the touch didin't work... I'm very new to linux, someone could help how i can install (or port) this Windows touch drivers to work on Linux?

Thanks alot! If you need some information, ask and I will try to put here!

onitake, 2017-03-03:
Please try to extract the firmware from the Windows or Android drivers as per the README.
I know the instructions are not very user friendly, but it shouldn't be too hard.

If you have Silead.fw, you can use the unscramble tool. For an Android driver, use fw_extract. And if you only have Silead.sys, use the hex dump instructions from the README. You don't need to calculate any magic, just find the hex codes listed, check if the data that follows fits the described format, and copy it into a new file (firmware.fw).

If you want to use the firmware with gslx680-acpi, convert it using fwtool and experiment with the parameters. If you would rather try it with the new driver in Linux 4.8, keep it in firmware.fw format. I haven't tested this part yet, you probably have to look around yourself on where to put the firmware image and how to get it working.

Ilia1, 2017-06-26:
https://github.com/edward-p/mssl1680-firmware

closed 2017-06-26 by Ilia1

onitake, 2017-06-26:
Can you pick the firmware from that repository up and format it into a pull request for gsl-firmware? The instructions are in the readme.

That would be really great. Thank you in advance!