onitake/gsl-firmware #27: "can't load firmware invalid magic" on firmware.fw unscrambled from sileadtouch.fw

vishnuhariam, 2017-01-01:
I got SileadTouch.fw from the manufacturer's website for Windows 10. Anyways, I downloaded your repo and ran unscramble with Perl locally. I unscrambled it to firmware.fw as in docs without any issues (I guess). But, SileadTouch is 103KB file whereas firmware.fw is 40KB. 
Anyways, I moved onto run fwtool on firmware.fw and it says, "Can't load firmware firmware.fw: Invalid magic =   at fwtool line 84"
I also tried adding attributes "-c -m -w -h" with no success.
I am a total noob though. Forgive me if I have done something wrong. But, please help me sort this issue out.

vishnuhariam, 2017-01-01:
I also tried running it on Ubuntu. But the results are same.

closed 2017-01-31 by vishnuhariam

jgminer, 2017-09-14:
For me, the issue was solved by upgrading perl.

onitake, 2017-09-14:
Could you both post which Perl versions you were using?