onitake/gsl-firmware #29: Firmware not named GSL_TS_CFG.h nor SileadTouch.fw

DavidVentura, 2017-01-28:
Hi, I have a chinese 10" tablet with windows 10 that uses this driver but trying the chuwi firmwares didn't work(mouse movement is erratic, but does respond to touch, dragging stuff everywhere, etc). This tablet has a 1280x800 resolution.
```

[   21.289014] silead_ts i2c-MSSL1680:00: Shutdown GPIO request failed
[   21.333134] axp20x-i2c i2c-INT33F4:00: AXP20x variant AXP288 found
[   21.401805] axp20x-i2c i2c-INT33F4:00: AXP20X driver loaded
[   21.472145] silead_ts: probe of i2c-MSSL1680:00 failed with error -16
[   23.897294] input: axp20x-pek as /devices/pci0000:00/808622C1:06/i2c-6/i2c-INT33F4:00/axp20x-pek/input/input6

```

The SoC is a `Intel(R) Atom(TM) x5-Z8300  CPU @ 1.44GHz`

My question is, what can I do ? I have the SileadTouch.{sys,cat,inf} files (attached)

SileadTouch.cat: data
SileadTouch.sys: PE32 executable (native) Intel 80386, for MS Windows
sileadtouch.inf: Windows setup INFormation, ASCII text, with CRLF line terminators

[touch1280x800.zip](https://github.com/onitake/gsl-firmware/files/737195/touch1280x800.zip)

Anything else I can try/give you to extract the firmware?

DavidVentura, 2017-01-28:
Ok, so the firmware was named `GSL_TS_CFG_THREE.h`.
How do I "guess" the resolution of the panel? as you show some weird stuff in the readme (940x750 ?? )

DavidVentura, 2017-01-28:
Sorry for the spam, I saw that this: https://github.com/onitake/gsl-firmware/tree/master/firmware/trekstor/surftab-twin-10.1-ST10432-8 had the same firmware filename, so I set up the same options and it works, although my display is rotated, and so is the input, but using -f swap fixes this.

onitake, 2017-01-30:
Sorry about the confusion.
The touch is a separate piece of hardware from the display, and may have a completely different resolution. There is not straightforward way to determine that resolution. The firmware doesn't give a clue, and I haven't been able to derive that information from the Windows driver.
That only leaves experimentation, hence the complicated instructions.

I'm glad you were able to figure it out, though.

DavidVentura, 2017-01-30:
I could get it working with similar parameters to the one linked (no flip, just had to swap), would you mind adding the "GSL_TS_CFG_THREE.h" filename to the readme?

Unrelated, the touch screen is ... erratic under linux, is there any way to fix this? touching the screen in one place and keeping the finger there shows the cursor moving 8-10px randomly around the finger. Is this driver related? firmware? I'm not using multitouch at all

onitake, 2017-01-30:
I added a short note: "(or similar)"

Naming _THREE explicitly doesn't seem to make sense to me, this looks more like a mess made by the vendor, not firmware file supported by the driver.

If you experience erratic behavior, this can be due to multiple reasons. The Silead controllers are generally pretty bad, and if finger tracking isn't working, it gets much worse. Did you enable software finger tracking in the firmware?

DavidVentura, 2017-01-30:
I had not enabled finger tracking, but I tried it just now and there's no difference.
Here's a very short video of the 'erratic' behavior:
[image]

This doesn't happen under windows

PS: The "(or similar)" is enough, I spent a while looking for exact filenames that's why I suggested it

onitake, 2017-03-03:
Yeah, it's something I've seen myself. I don't know how to solve that problem.
Perhaps a high jitter value helps. If you have to means to hack the driver code, please experiment with this part.

onitake, 2022-09-27:
The original question was answered.
Please open a new issue if you still have problems with this firmware / device and current drivers.

closed 2022-09-27 by onitake