onitake/gsl-firmware #30: silead_ts.fw from SileadTouch.fw not covering the whole screen

vishnuhariam, 2017-01-31:
My Tablet 7inch has a resolution of 1024x600. I checked with `wm` The touchscreen was all messed up after RemixOS installation but, I tried converting SileadTouch to silead_ts.fw with a resolution of 1024x600 first and then tried with different resolutions. 1024x600 was the one that came closest to perfection. But, it still has got issues like the right 2-3 cm screen isnt detected and the touches are registered a cm or less to the left of the area of actual input. I am not an expert. Can anyone tell me what I am doing wrong.

onitake, 2017-01-31:
That means the configured touch area does not cover what the controller sends, or the firmware does not match the panel.
As mentioned elsewhere, the firmware image contains device specific calibration data, so without a perfect match, there will always be problems. Some can be corrected with xinput_calibrator or proper width/height settings, but sometimes, that's not enough.

vishnuhariam, 2017-02-04:
Hey. Am sorry for the delay in reply. I use the following driver for my device - http://www.iball.co.in/File/Product/Drivers/i701%20Windows%2010%20TouchPad%20Driver.rar which happens to be the working driver for Windows 10. But when I manually update the driver in Windows 10 with the .inf it doesnt cover some portions of the screen as seen on RemixOS, but when I run the executable file as per instructions, the driver works fine on Windows 10. 
Can you help me understand what the issue is now since its the driver for my device. And since I am using RemixOS, I cant run xinput_calibrator 
Please advise and thanks a lot for commenting. I have been trying to fix the touchscreen for android on this tablet for more then an year now.

onitake, 2017-03-03:
I'm really sorry, I can't help with Android specifically. You should experiment with the device parameters in fwtool and try to get it working as closely as possible, then do the final calibration Android.
I haven't tested the driver with Android myself, so you really have to find help elsewhere.
Sorry.