onitake/gsl-firmware #35: Firmware for unknown generic Windows tablet

mario92, 2017-03-26:
Hello,

I own a Windows tablet that is unbranded, and came with Windows 8.1

Stupid me has clean installed Windows 10 and after that touch is not working properly - touch points are registered all over the place.

I have identified some components in that tablet:

SCREEN

8 inch
1280×800
IPS
CPU

Intel Atom Z3745, 1.33GHz
RAM

2GB LDDR3
ROM

32GB + optional microSD card
aluminium back side, no HDMI output
BIOS

Some sort of Insyde BIOS, full of advanced features
LINK to pictures:

https://imgur.com/a/3tIYY

Speccy:

http://speccy.piriform.com/results/kcvPqOiOjL57kTVSdiNbJWo

Video of the issue:

https://vid.me/95wU

I have opened up tablet and recognized other components:

- 4000mAH battery, SJY 3565155
- LCD Manufacturer: Samsung V200
- Touch digitizer is ZHG-0100A01, with Silead GSL 3676 chip

It is same like this one:
https://www.aliexpress.com/store/product/Black-New-For-ZHG-0100A01-Tablet-PC-Capacitive-Touch-Screen-Panel-Digitizer-Sensor-Replacement-Parts-Free/2446005_32775630447.html

but unfortunately there is no tablet model name that comes with it.

What can I do?

Thank you very much in advance,

Mario

onitake, 2017-03-27:
I'm surprised that it works _at all_.
These chips require highly device-specific firmware and wont work without it.
Can you make a Gist from the kernel log so I can at least see which driver you're using and which firmware?

mario92, 2017-03-27:
Hello Onitake,

thanks for your reply. I'm quite pessimistic about outcome now, since I hardly can find any information about tablet and its drivers/firmware.

I don't know what specifically you need from kernel log. This tablet runs Windows exclusively (it came like that when I got it) and Insyde BIOS was configured for Windows only.

I have tried several drivers and firmwares and best I got was when y-axis worked but x-axis was messed up quite badly. Since then I tried other combinations, but I will take some time today after work to repro best-case-scenario and will send you that driver and sileadtouch.fw I used.

Thanks for your help!

mario92, 2017-03-28:
Hello Onitake,

I am currently using these drivers:
https://1drv.ms/f/s!Al74eIX0YA0ypKhLAiYv9xxVe9e1dw

Please let me know is there anything else I can provide you with :)

onitake, 2017-03-28:
I'm sorry, I totally missed the point that you're using Windows on this tablet.
I can't help you much with that, I'm afraid.

If you're unable to obtain a suitable driver package for exactly this device, there's not much you can do.
You could try with and without the SileadTouch.fw file installed (make sure it's really removed, Windows tends to cache drivers), to see if it makes a difference.

Your best bet is probably obtaining an identical device and extracting the firmware from there.

Sorry.

closed 2017-10-23 by onitake