onitake/gsl-firmware #37: Firmware for VOYO V3 Pentium version (perhaps for all V3 models)

fazerg, 2017-04-18:
This is mostly again about fwtool unable to locate Silead.pm.
Stuck with this issue on Ubuntu 17,04 Mate,  and cannot sort it out.
When I enter 
`./fwtool -c firmware.fw -m t113 -w 1920 -h 1080 -t 10 silead_ts.fw`
I got the same 

> Can't locate Firmware/Silead.pm in @INC (you may need to install the Firmware::Silead module)

And this happens even if I have added it, below is part of perl -V output:
```
%ENV:
    PERL5LIB="/home/voyo/gsl-firmware/tools/Firmware/Silead.pm"
  @INC:
    /home/voyo/gsl-firmware/tools/Firmware/Silead.pm
    /etc/perl
    /usr/local/lib/x86_64-linux-gnu/perl/5.24.1
    /usr/local/share/perl/5.24.1
    /usr/lib/x86_64-linux-gnu/perl5/5.24
    /usr/share/perl5
    /usr/lib/x86_64-linux-gnu/perl/5.24
    /usr/share/perl/5.24
    /usr/local/lib/site_perl
    /usr/lib/x86_64-linux-gnu/perl-base

```
Please, help me sort out this issue or create  firmware. The model description is here(needs translation from Chinese):[http://www.myvoyo.com/chanpin/pingban/714.shtml]( http://www.myvoyo.com/chanpin/pingban/714.shtml)
Downloads are available from [http://en.myvoyo.com/xiazai/Win%20Pad/VBOOK%5FV3%5Fpentium/down/](http://en.myvoyo.com/xiazai/Win%20Pad/VBOOK%5FV3%5Fpentium/down/) and [http://pan.baidu.com/share/init?shareid=3303446473&uk=2905619295](http://pan.baidu.com/share/init?shareid=3303446473&uk=2905619295) , password  xdbg.
The touch screen driver available only with entire  OS recovery and all drivers  huge package. Hence, I upload .h file here 
[SileadTouch_voyo.zip](https://github.com/onitake/gsl-firmware/files/930716/SileadTouch_voyo.zip)

fazerg, 2017-04-18:
Module compiled but does not work. 
Kernel 4.10.0-19-generic, dmesg output after reboot:
```
root@vbuntu:/home/voyo# dmesg | grep silead
[   12.356118] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[   12.495867] silead_ts i2c-MSSL1680:00: Direct firmware load for silead/mssl1680.fw failed with error -2
[   12.495870] silead_ts i2c-MSSL1680:00: Firmware request error -2
[   12.515932] silead_ts: probe of i2c-MSSL1680:00 failed with error -2
```
after insmod of gslx680_ts_acpi.ko:
```
root@vbuntu:/home/voyo/gslx680-acpi-master# dmesg | grep gslx
[  234.253233] gslx680_ts_acpi: loading out-of-tree module taints kernel.
[  234.253355] gslx680_ts_acpi: module verification failed: signature and/or required key missing - tainting kernel
[  234.272666] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 50, flags 0x0
[  234.272761] gslx680 i2c-MSSL1680:00: Direct firmware load for silead_ts.fw failed with error -2
[  234.272768] gslx680 i2c-MSSL1680:00: gsl_ts_probe: failed to load firmware: -2
[  
234.292540] gslx680: probe of i2c-MSSL1680:00 failed with error -2
```
**What that error -2 does actually mean?**

onitake, 2017-04-19:
Woops, looks like a recent Perl version broke this.
I've patched fwtool so it always searches Silead.pm in the same directory.

Error -2 means this:
```
$ grep '\b2\b' /usr/include/asm-generic/errno-base.h 
#define ENOENT           2      /* No such file or directory */
```
It's thrown when the controller couldn't be detected on the I2C bus, usually because the driver didn't succeed waking it up from sleep.

This has started appearing in recent kernels because of changes in the ACPI subsystem and because the DSDT entry for the Silead controller isn't how it's supposed to be.
There isn't really much you can do while the kernel devs figure out a solution. You could try with an older kernel though.

fazerg, 2017-04-19:
Thanks,  not a good news.  Kernels, older than 4.9, are not working well on the existing hardware (Intel Apollo Lake platform with N4200 processor). The system is  sluggish and drains battery pretty quick.
The only hope then is waiting for kernel devs...

fazerg, 2017-04-23:
It wooorks!!!!
That was my fault actually, despite it's clearly told in Readme.md where to put firmware, I put it into /lib/firmware/silead/ instead of just /lib/firmware/.
But anyway, it does not work perfectly well. The issues are:
1. In browsers x-axis is flipped, on desktop is all right.
2. Most left side, especially top left on desktop is a bit dodgy. Touches recognized in wrong quite random places about 30-40 pixels away from where they should be. The rest of the screen is OK, and there are no inverted axis on the desktop.
There is obviously nothing to do with calibration but something should be done with firmware or driver.  I've got  headers file for firmware from the manufacturer but have no idea what all those bunches of numbers mean and where to make necessary corrections. Can anyone give me some advice where to dig?
Also, with the mainstream driver and uncompressed  firmware screen is responding on entire area but both axis inverted and all touches displayed only in the left top quarter, i.e. if I touch right lower corner, response of cursor is  almost exactly in the center.  Calibration does not help.

onitake, 2017-04-23:
The official kernel driver requires the firmware in /lib/firmware/silead, gslx680-acpi expects it in /lib/firmware.

About 1: This is very weird. What exactly do you mean with "in browsers"? Does it scroll in the wrong direction? That is a problem with the scroll emulation or the  browser addon. I don't think I can help with that.
About 2: The official driver currently doesn't have a good way of setting the panel parameters. The behaviour you're seeing is caused by the default values it uses when it doesn't know about the panel. I don't know what the current status is, but in the future there will be the possibility of adding DMI overrides for specific devices.

fazerg, 2017-04-23:
No, it scrolls well but all buttons respond on opposite side on the x-axis. I've checked in Chromium only as it was too late last night when I sorted things out.
Will play now a bit with different versions of firmware too, as the one obtained from sys file is different comparing to one  made from  headers file. 
I do not really care at the moment about the mainstream driver, just told  about it for a case it may bring some ideas.

fazerg, 2017-04-26:
Nothing helps. Y coordinates do not match the screen and cannot adjust it.
About 1/3 of the upper screen part shows only X-coordinate changes. xinput test shows 0 for Y all the time in this area. 
xinput_calibrator offers following setting:
```
Section "InputClass"
	Identifier	"calibration"
	MatchProduct	"Silead GSLx680 Touchscreen"
	Option	"MinX"	"-1206"
	Option	"MaxX"	"65512"
	Option	"MinY"	"-8819"
	Option	"MaxY"	"61732"
	Option	"SwapXY"	"0" # unless it was already set to 1
	Option	"InvertX"	"0"  # unless it was already set
	Option	"InvertY"	"0"  # unless it was already set
EndSection
```
However, seems that the 99-calibration.conf rule is not loaded neither after **udevadm control --reload-rules && udevadm trigger**, nor after reboot. 
Setting 
xinput set-prop 10 'libinput Calibration Matrix' 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
to some values for Y scale, alternates behavior but  all changes  make only negative effect, i.e. vertical sensitive area changes a lot depending on value I set for Y scale. What I need as it seems to me is to shift "0" position for  Y to some negative value, most likely suggested by xinput_calibrator. But I cannot get it working, i.e. force Y zero value to higher point at the screen. Please give me some advice,  how to fix it..

Also, what I have noticed,  firmware.fw files have a lot of divverence, if compare generated from header file and  Windows sys file.
At the same moment, silead_ts.fw files generated from both sources have difference in just one byte in the header.  Unfortunately, I do not understand perl for trying to find the source of the issue.  Both firmware.fw files attached. 

[firmware.zip](https://github.com/onitake/gsl-firmware/files/958332/firmware.zip)

onitake, 2017-05-07:
I don't think you can use normal udev rules for that.
The device sits on an I2C bus and doesn't have a regular Vendor or Product ID.

You could match the ACPI ID (MSSL1680), but I have no idea how to do this.

Perhaps it's possible to pass more information from the driver to udev, but I have no idea how to do this either.

I'll ask the developers of silead.ko, they're more familiar with the kernel. Perhaps they know a way.

onitake, 2017-05-07:
You could try this, perhaps it works: http://hunterhu.com/blogs/setup-xorg-touchpanel-input-dev-path-with-udev-rules

fazerg, 2017-05-10:
I've sorted it out by adding calibration string to ~/.xinputrc. 
It works only after login of course, but that's OK for me.

onitake, 2017-05-10:
That's good to hear.
Are you still having the x axis issue in the browser?

fazerg, 2017-05-10:
That sorted after re-compilation, not sure now what helped after many tries.
The only issue in browsers found so far is Qupzilla where instead of scrolling I got selection.
The major issue for now is that touchscreen is working as single touch.
Also, more precise calibration needed but I can't understand yet how convert xinput-calibrator  results that do not work with xorg.conf to  numbers for matrix in xinput set-prop.

onitake, 2017-05-15:
Did you check with `xinput test` if the multitouch events are reaching X?

Does `xinput_calibrator --output-type xinput` help?

fazerg, 2017-05-15:
No, they don't. I have response like:
```
button release 1 a[0]=25397 a[1]=19357 
motion a[0]=40326 a[1]=31571 
button press   1 a[0]=40326 a[1]=31571 
motion a[0]=40496 a[1]=31428 
motion a[0]=41517 a[1]=31667 
motion a[0]=41517 a[1]=31571 
motion a[0]=41585 a[1]=31333 
motion a[0]=41585 a[1]=31333 
button release 1 a[0]=41585 a[1]=31333
```
The second says that this in not supported:
`ERROR: XorgPrint Calibrator does not support the supplied --output-type`

vbooka1, 2017-06-16:
Could not make touchscreen work on VOYO VBook A1, with all 4.11.* kernels on openSUSE 42
Upstream kernel: 

```
vbook:/local/sileadtouch/onitake-gslx680-acpi # modprobe silead
vbook:/local/sileadtouch/onitake-gslx680-acpi # dmesg|tail
[33452.976963] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[33453.294532] i2c_designware i2c_designware.3: i2c_dw_handle_tx_abort: lost arbitration
[33454.312683] i2c_designware i2c_designware.3: controller timed out
[33454.312709] silead_ts i2c-MSSL1680:00: Firmware load error -110
[33454.328827] silead_ts: probe of i2c-MSSL1680:00 failed with error -110
```

gslx680_ts_acpi:
```
vbook:/local/sileadtouch/onitake-gslx680-acpi # insmod gslx680_ts_acpi.ko
vbook:/local/sileadtouch/onitake-gslx680-acpi # dmesg|tail
[33483.932869] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 50, flags 0x0
[33483.933095] input: Silead GSLx680 Touchscreen as /devices/pci0000:00/0000:00:16.3/i2c_designware.3/i2c-10/i2c-MSSL1680:00/input/input17
[33485.000691] i2c_designware i2c_designware.3: controller timed out
[33485.000710] gslx680 i2c-MSSL1680:00: gsl_ts_reset_chip: gsl_ts_write 1 fail!
[33485.000712] gslx680 i2c-MSSL1680:00: gsl_ts_probe: chip reset failed
[33485.052822] gslx680: probe of i2c-MSSL1680:00 failed with error -110
```

Attached the original Windows firmware and binary file converted with gsl-firmware/tools/

[firmware.zip](https://github.com/onitake/gsl-firmware/files/1081950/firmware.zip)

fazerg, 2017-06-17:
vbooka1, have you placed your firmware file  in proper place, e.g. /lib/firmware/ in Ubuntu? 
BTW, mainstream module will not work with this firmware, you need uncompressed one  (about 41 kB in size) you got at the first stage and put it in /lib/firmware/silead/.

onitake, 2017-06-17:
I don't think this is related to firmware at all.
As you can see, there seems to be a communication error in the I²C controller, as reported by the Designware I²C driver (which is used for most Intel platforms).

I remember a discussion on the Linux kernel mailing lists about bugs in that driver, but those were supposedly fixed a while ago.

-110 means ETIMEDOUT, which indicates that either the communication partner (the Silead chip) wasn't responding, or there are still bugs with the I²C controller driver.
Most likely, the Silead chip is not powered on correctly. This has been one of the most long-standing issues with Silead on ACPI platforms. Each manufacturer messes up differently, so it's really hard to find a reliable solution.

I suggest downgrading the kernel (perhaps that helps) and reporting the error on [linux-input](http://vger.kernel.org/vger-lists.html#linux-input) and [linux-i2c](http://vger.kernel.org/vger-lists.html#linux-i2c). The developers of silead_ts are on those lists and may be able to help you better than myself.

vbooka1, 2017-06-17:
@fazerg sure! 
```
01:07:56 /local/sileadtouch> ls -l /lib/firmware/silead/
total 0
lrwxrwxrwx 1 root root 25 Jun 10 06:29 mssl1680.fw -> /lib/firmware/mssl1680.fw
lrwxrwxrwx 1 root root 26 Jun 10 06:49 silead_ts.fw -> /lib/firmware/silead_ts.fw
01:08:01 /local/sileadtouch> ls -l /lib/firmware/silead_ts.fw /lib/firmware/mssl1680.fw
-rw-r--r-- 1 voyo users 21936 Jun  2 06:14 /lib/firmware/mssl1680.fw
-rw-r--r-- 1 voyo users 21936 Jun 10 06:48 /lib/firmware/silead_ts.fw
01:08:14 /local/sileadtouch> md5sum /lib/firmware/silead_ts.fw /lib/firmware/mssl1680.fw
c0ac8d2bad04d46662ab337b5ed13f19  /lib/firmware/silead_ts.fw
c0ac8d2bad04d46662ab337b5ed13f19  /lib/firmware/mssl1680.fw
```

@onitake i had the same issue with stock 4.4.62 openSUSE kernel. Yes, it definitely is a timeout, because `modprobe` takes 3 seconds every time.

```
# time modprobe silead

real    0m3.794s
user    0m0.048s
sys     0m0.034s
```

fazerg, 2017-06-18:
There is a little difference in hardware between A1 and V3 'Pentium' version.
I have touchscreen working on 4.10 and 4.11 kernels on Ubuntu.
Just to be sure that the issue is with OpenSUSE version of kernel:
1. Try one/any of attached firmware files with mainstream module. Rename it  to mssl1680.fw and place to /lib/firmware/silead/ or replace existing simlinked in /lib/firmware.  Reboot. Mainstream module should work but inverted and responding on just about 1/4 of screen area.
2. If not, try bootable stick with Ubuntu. You can make one that will boot with help of rEFInd. There was also a version of BIOS that can boot GRUB2. You may find it on 4PDA VOYO A1 thread. 
[firmware.zip](https://github.com/onitake/gsl-firmware/files/1082922/firmware.zip)

vbooka1, 2017-06-18:
@fazerg 
1. Both firmware files work! And exactly as you said - the input is inverted and works on the 1/4 of the screen. How did you got this firmware? I want to report this to linux-input and linux-i2c maillists.
2. It's me, who uploaded this BIOS... :)

huangtao00, 2017-09-16:
i just install ubuntu 17.04 in my laptop of voyo v3 pentium. everything works perfectly except the touchscreen. can anybody tell me how to fix the touchscreen.  how can i use this firmware updated in this repo?  waiting for your reply!

onitake, 2017-10-21:
Since the new kernel driver is all out and about, can you guys prepare a kernel patch similar to https://github.com/onitake/gsl-firmware/pull/47  for both the A1 and the V3?

This will help everybody who has one of these books, as it will make them work out-of-the-box (as long as the firmware files are installed).

Basically, what you have to do is prepare a patch for silead_dmi.c that matches your hardware and includes all the touchscreen settings. Give your firmware.fw a unique name, like the example I linked in https://github.com/onitake/gsl-firmware/pull/47#issuecomment-336563406 .
Unload and reload silead.ko and see if it loads the correct firmware (dmesg will print a warning otherwise).

When you get it working, Cc @jwrdegoede with the patch and submit a PR to me with the correct firmware files. These should go into `firmware/linux/silead`.

vbooka1, 2017-11-08:
I'm afraid the `silead_dmi.c` patch could not help as every single Voyo laptop is built with the different hardware lol. 
I've finally made the touchscreen work on my laptop. Happily noticed that someone shared the Voyo VBook A1 firmware here: `https://github.com/onitake/gsl-firmware/tree/master/firmware/voyo/at13` and immediately gave it a try. However it did not work  well. The touch screen covered the whole screen instead of 1/4 top-left corner now, but X and Y axis were inverted. So I converted the `SileadTouch_voyo_T113_skylak_3692_landscape_activepen_64it_20160907_zhaowei.h` with the correct flags and resolution (1920x1080 misplaced the cursor every touch and 1980x1536 seems to be the correct value) and now the touchscreen works correctly!

```
~/gsl-firmware/tools # ./untscfg SileadTouch_voyo_T113_skylak_3692_landscape_activepen_64it_20160907_zhaowei.h firmware.fw
~/gsl-firmware/tools # ./fwtool -t 10 -f xflip,yflip -3 -w 1980 -h 1536 -c firmware.fw silead_ts.fw 
~/gsl-firmware/tools # cp silead_ts.fw /lib/firmware/
~/gsl-firmware/tools # cd ../../onitake-gslx680-acpi/
~/onitake-gslx680-acpi/ # make
~/onitake-gslx680-acpi/ # insmod gslx680_ts_acpi.ko
```

My Linux kernel version is 4.13.9 if that matters.

vbooka1, 2017-11-08:
This may be an answer: a broken firmware? Though YJHY_AT31_3692_L_Pen_x64_ZW_170401.h works well in Windows 10 on my specific laptop.

```
~/gsl-firmware/tools # diff -u SileadTouch_voyo_T113_skylak_3692_landscape_activepen_64it_20160907_zhaowei.h YJHY_AT31_3692_L_Pen_x64_ZW_170401.h 
--- SileadTouch_voyo_T113_skylak_3692_landscape_activepen_64it_20160907_zhaowei.h       2017-11-09 03:44:24.262399277 +0700
+++ YJHY_AT31_3692_L_Pen_x64_ZW_170401.h        2017-06-10 06:47:48.219718161 +0700
@@ -1,16 +1,16 @@
 unsigned int gsl_config_data_id[] =
 {
-       0x886502,  
+       0x8d0937,  
        0x200,
        0,0,
        0,
        0,0,0,
-       0,0,0,0,0,0,0,0x10f1285f,
+       0,0,0,0,0,0,0,0x10ef785f,
 
 
        0x4012010e,0xa,0x180020,0x18001f,0x4b00780,0,0x5100,0x8e00,
        0,0x320014,0,0,0,0,0,0x20,
-       0x8,0x10000,0x10000,0x106e0011,0x1030000b,0,0,0,
+       0x8,0x4000,0x1000,0x106e0011,0x1030000b,0,0,0,
        0x12492488,0x100,0,0,0,0,0,0,
        0,0,0,0,0,0,0,0,
        0,0,0x801000,0x160100,0xa0010,0x10004,0,0,
@@ -130,7 +130,7 @@
 {0x78,0x00000000},
 {0x7c,0x181f1820},
 {0xf0,0x3},
-{0x00,0xc33d367f},
+{0x00,0x644a300d},
 {0x04,0x00000000},
 {0x08,0x00000000},
 {0x0c,0x00000000},
@@ -141,7 +141,7 @@
 {0x20,0x00000000},
 {0x24,0x00005100},
 {0x28,0x00008e00},
-{0x2c,0x0000ff9c},
+{0x2c,0x0000fe70},
 {0x30,0x00000000},
 {0x34,0x00000000},
 {0x38,0x00000000},
@@ -149,7 +149,7 @@
 {0x40,0x00000000},
 {0x44,0x00000000},
 {0x48,0x00000000},
-{0x4c,0x00000000},
+{0x4c,0x5f010020},
 {0x50,0x00000000},
 {0x54,0x00000000},
 {0x58,0x00000000},
@@ -183,7 +183,7 @@
 {0x44,0x04020a00},
 {0x48,0x0014012c},
 {0x4c,0x9a000000},
-{0x50,0x00000026},
+{0x50,0x00000013},
 {0x54,0x00010203},
 {0x58,0x04050607},
 {0x5c,0x08090a0b},
@@ -192,13 +192,13 @@
 {0x68,0x14151617},
 {0x6c,0x18191a1b},
 {0x70,0x1c1d1e1f},
-{0x74,0x001c000a},
+{0x74,0x0028000a},
 {0x78,0x80808080},
 {0x7c,0xcba981f4},
 {0xf0,0x5},
 {0x00,0x00000005},
 {0x04,0x0000000a},
-{0x08,0x000000dc},
+{0x08,0x0000012c},
 {0x0c,0x66806666},
 {0x10,0x0000000a},
 {0x14,0x00000000},
@@ -258,7 +258,7 @@
 {0x68,0x00000000},
 {0x6c,0x00000000},
 {0x70,0x00000000},
-{0x74,0x0000012c},
+{0x74,0x00000123},
 {0x78,0x000a003c},
 {0x7c,0x00000000},
 {0xf0,0x7},
@@ -2409,13 +2409,13 @@
 {0x6c,0x00000000},
 {0x70,0x00000000},
 {0x74,0x00000000},
-{0x78,0x00060407},
-{0x7c,0x00060406},
+{0x78,0x00060306},
+{0x7c,0x00060305},
 {0xf0,0x4f},
-{0x00,0x070d0102},
-{0x04,0x070d0102},
-{0x08,0x0e140407},
-{0x0c,0x0e140407},
+{0x00,0x070d0001},
+{0x04,0x070d0001},
+{0x08,0x0e140306},
+{0x0c,0x0e140306},
 {0x10,0x00000000},
 {0x14,0x00000000},
 {0x18,0x00000000},
@@ -2425,9 +2425,9 @@
 {0x28,0x00000000},
 {0x2c,0x00000000},
 {0x30,0x00000000},
-{0x34,0x000221f3},
-{0x38,0x000d0708},
-{0x3c,0x000d0bb8},
+{0x34,0x000221fb},
+{0x38,0x00000a8c},
+{0x3c,0x0000157c},
 {0x40,0x0000015e},
 {0x44,0xf8010017},
 {0x48,0x0000000a},
```

onitake, 2017-11-08:
My point is that getting an individual device to work is no long-term solution.
I stopped supporting gslx680_ts_acpi because it requires manual firmware installation and is still a hassle.

silead.ko tries to tackle this by applying a set of options and a custom firmware name based on DMI identification. This should be possible for many (most?) devices, even if they are sold under the same brand name.

If different Voyo devices need different settings, then different DMI matches need to be added. Is there a particular reason why this approach would not work here? Have you compared DMI blocks for different Voyo tablets?

huangtao00, 2017-12-01:
Actually I read all the discussions you guys talked here. Now  I finally known how to do this now. really tough things for me. thx for all of your hard work. I will try to solve the touch screen thing for my ezpad 6 plus tablet since i have sold voyo v3 one month ago.

vrobolab, 2018-07-18:
The driver stopped working somewhen around 4.16, I do not know the exact version because I rarely enable the touchscreen.
`dmesg` with current 4.17.4 kernel:

```
[  294.479160] gslx680 i2c-MSSL1680:00: gsl_ts_probe: got a device named MSSL1680:00 at address 0x40, IRQ 50, flags 0x0
[  294.479353] input: Silead GSLx680 Touchscreen as /devices/pci0000:00/0000:00:16.3/i2c_designware.3/i2c-10/i2c-MSSL1680:00/input/input17
[  295.157275] sdhci-pci 0000:00:1b.0: SDHCI controller found [8086:5aca] (rev b)
```

`xinput --list`
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ ImExPS/2 Generic Explorer Mouse           id=11   [slave  pointer  (2)]
⎜   ↳ Silead GSLx680 Touchscreen                id=12   [slave  pointer  (2)]
⎜   ↳ 2.4G Mouse                                id=13   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
 . . . 
```

— the touchscreen got detected successfully, but system does not react to touches, and `xev` does not see any touches too.

Any ideas?

onitake, 2018-09-17:
@vrobolab Please don't report driver issues here.

Bug reports for gslx680-acpi should go to that driver repository, and bug reports for silead_ts should go to the linux-input kernel mailing list.

Thanks.

closed 2018-09-17 by onitake

githublex2020, 2020-03-21:
**@vbooka1** Please can you help me to install touchscreen on my Voyo vbook V3? i have kali linux kde version the last one. Please thanks in advice for that