onitake/gsl-firmware #43: Self-generated silead_ts.fw for Wintron 7.0 doesn't work

sixtyfive, 2017-07-11:
When I copy this repository's `firmware/trekstor/surftab7new/silead_ts.fw` to `/lib/firmware` and `modprobe` the driver, the touchscreen works just fine, except for the screen orientation being 90 degrees off. So I used `fwtool` with the parameters found in `firmware/trekstor/surftab7new/README.md`, but an added `,xflip` to generate a new file. With it, the driver loads fine, too, but the touchscreen doesn't react to any input. Also, if I use *exactly* the same `fwtool` command as given in the README (without xflip added), a different `silead_ts.fw` results from that as the one that has been committed to your repo. Is it possible that a bug/regression was introduced into the `fwtool` at some point?

onitake, 2017-07-12:
Not really, no.

fwtool processes the firmware data, separates it into pages and reorders these pages. Vendor firmware is assembled in a (seemingly) random order, which is not preserved when converting with fwtool. From my tests, I concluded that it doesn't make a difference.

On the other hand, the posted command line is *supposed* to generate the same firmware as before.
I really can't explain why this is not the case. Perhaps you should ask the submitter ( @holzingerm ) about it.

By the way: `xflip` mirrors the X axis. If you want rotation, you should use `swap`.

onitake, 2017-10-24:
The official kernel driver includes settings and for the wintron 7.0 (the version with the 6 at the end). Please install the firmware from https://github.com/onitake/gsl-firmware/tree/master/firmware/linux and see it if works. I'm slowly phasing out gslx680_acpi, but haven't had time to write documentation for the silead kernel driver yet.

If the settings are incorrect, please look at: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/drivers/platform/x86/silead_dmi.c
You can add your own settings there, then recompile the silead.ko module.

closed 2017-10-24 by onitake