onitake/gsl-firmware #50: Touch & hold only last less than 2 seconds

sobakazu, 2017-10-25:
Hello,

I success to configure silead_ts.fw for my RCA W101 v2 with this script :

./fwtool -c SileadTouch.fw -3 -m 1680 -w 890 -h 1660 -t 10 -f yflip,swap,track silead_ts.fw

Its work perfectly on alignment, but the problem is when I touch and hold, the input only last less than 2 seconds and then gone.

How I fix it?

Thanks.

onitake, 2017-10-27:
I remember having this problem, but didn't find a good solution.

Perhaps you can get better results with silead.ko?
You'd have to add support for your tablet first, though. See here an example how: #47

sobakazu, 2017-10-28:
I did some experiments using silead_ts.fw from another device (eg. eastrising er-tpc050-1) and the input holds as long as I hold the touchscreen. But I can't find the right configuration because eastrising touch resolution is 800x480, and RCA is 1280x800.

Based on experiments above, is it possible to manipulate directly into silead_ts.fw file using hex editor to achieve touch (hold) behavior like eastrising firmware?

I haven't tried silead.ko, since I'm running on Phoenix OS (based on Android x86), and I'm noob on Linux programming. 😃

onitake, 2017-10-30:
You can modify silead.fw by hand if you like, but there isn't much point.
All the settings you can make are built into `fwtool`, and the firmware itself is a mystery to everyone.

The only software that can generate a completely new firmware is Silead Touch Configurator, but that's OEM-only. I haven't seen it in the wild yet.

sobakazu, 2017-11-01:
If I'm gonna using silead.ko, here is my understanding :

I have to add my device parameters into this code : https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/platform/x86/silead_dmi.c?id=c3a73ed8a82b666ac01466d3badd7824eae89c44

and then `makefile` using this : https://github.com/onitake/gslx680-acpi/blob/master/Makefile to produce silead.ko by replacing gslx680_ts_acpi into silead and silead_dmi.o

and then replace silead.ko from my device using the new one.

and the firmware should put in /lib/firmware/silead.

Please CMIIW.

Thanks.

onitake, 2017-11-01:
No, you shouldn't do that.

For drivers in the kernel tree, you should always use the kernel build system directly.
Do not copy anything out and try to build it out-of-tree.
This way, you can use all the bells and whistles of the build system and don't have to worry about taining the running kernel.

I think there's even a way to build only a subset of all modules, instead of having to wait for a complete kernel build.

onitake, 2017-11-01:
Though, if you don't run `make clean` in between, you could also just build all modules with `make modules` once, and subsequent builds would only rebuild the modified driver.

Don't forget to run `sudo make modules_install` after each build.

sobakazu, 2017-11-11:
Finally, I success to maintain hold gesture on my RCA.

I tried to analyze several different silead_ts.fw with hex editor and failed to understand the patterns.
Since I haven't Linux environment to rebuild silead.ko module, I never try it.
After disassembled, I found out that my device using Silead MSL 3670 chipset.

I try to use silead_ts.fw from another devices.

gsl3670-cube-iwork8-air.fw or gsl1680-pipo-w2s.fw works on my RCA W101v2, but problem is the pointer shaking as it to sensitive to read touch coordinate.

I tried to use gsl1680-a082-q8-a70.fw, the pointer looks steady, but alignment not right.

Is there a way to fix shaking pointer problem?

Thanks.

onitake, 2017-11-11:
You did see the separation into memory pages and the identification block, right?
As for the rest, I don't have a clue either.

The shaky pointer is something that cannot be solved without properly calibrated firmware. It may be possible to apply jitter interpolation at the driver level, but that will make the pointer less precise and slower. I'll try to raise the issue on the linux-input mailing list, perhaps this is something to be taken into consideration.

Have you succeeded building a patched silead.ko module yet?

onitake, 2017-11-14:
I just received a confirmation from the linux-input maintainer: You can use any input subsystem flags with silead.ko.

See here: http://www.spinics.net/lists/linux-input/msg53986.html

Scorpion2185, 2018-09-02:
@sobakazu i need some help. Can you help me with the basic steps? You can see my problem here in my issue https://github.com/onitake/gsl-firmware/issues/86. I don' t know how use the firmware my device has been tested.

closed 2018-09-17 by onitake