onitake/gsl-firmware #51: Driver for Teclast 16 Power

l-costantini, 2017-10-30:
Hello,
I need some advice on how to proceed about the FW for Silead on a Teclast 16
Power.

This is a Z8750 (Intel Atom) based device.
As far as I see the controller is recognized as Silead on my Ubuntu 17.10
(with smaller area and inverted X axis if using firmware included in 17.10  )

I got the Android 6.0  image (system.img) for the device and into it I found
the /etc/firmware folder with some fw files. Among them

    4   -rw-r--r--.  1 root root       3514 dic 31  2008 maxtouch.cfg
  112 -rw-r--r--.  1 root root   113856 dic 31  2008 maxtouch.fw
   40  -rw-r--r--.  1 root root     39272 dic 31  2008 silead.fw

But no .ko files (I guess as it is a x86 Android, not pure Google/ARM one)

As you see there are two FW files.

I found no instruction this case (but I could be wrong)... How to proceed?

Best Regards,
Leonardo

onitake, 2017-11-14:
I don't know about maxtouch.fw, but silead.fw looks promising.

You should be able to use it with the Linux silead driver, but you will have to add a DMI match and correct hardware parameters to [silead_dmi.c](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/drivers/platform/x86/silead_dmi.c), then recompile the kernel module locally. These aren't documented anywhere, you'll have to find them out yourself. See the readme for a rough guide on how to proceed.

I don't recommend using my [gslx680-acpi](https://github.com/onitake/gslx680-acpi) driver any more, as I stopped maintaining it. But if you feel adventurous, you could use it to determine the hardware parameters. Note that it requires converted firmware.

Woobyxd, 2020-03-17:
Hey @onitake I also own this tablet, im trying to get touchscreen working on Bliss OS x86 also on Ubuntu, touch is working partially, its inverted and works only in few places with included silead_ts.fw in Bliss OS. Can you help a little? 
Thanks a lot.

onitake, 2020-03-19:
Sorry, I can't give you more help than what I wrote above.
To get everything out of this touchscreen, you need to patch your kernel so it configures the driver accordingly. The firmware alone is not enough. Check out the readme for more info.

Woobyxd, 2020-03-19:
i tried with the x3 plus fw and silead_ts and its quite good, the touch works much better i can operate, but its like moved 2 cm to the left, means that when I want to touch home button in the middle  on the navbar i have to touch more from the left side, to fix that i need to edit silead_ts or firmware file, can you advice?

onitake, 2020-03-19:
Simple offset problem should be fixable without much fiddling around with the driver or firmware.
You can use xinput_calibrator or similar to set up a working operating range for Xorg.

Woobyxd, 2020-03-19:
Can you guide me a little with the xinput_calibrator?

onitake, 2020-03-19:
Well, it's not super difficult: https://wiki.archlinux.org/index.php/Touchscreen#Calibration

Quote:
> Install `xinput_calibrator`. Then, run `xinput_calibrator` and follow the instructions. 

:slightly_smiling_face:

Woobyxd, 2020-03-19:
Is there any other way, i dont have any linux system installed, only ubuntu terminal on windows and android

onitake, 2020-03-19:
Um... You wrote:
> Hey @onitake I also own this tablet, im trying to get touchscreen working on Bliss OS x86 also on Ubuntu

I don't know about Bliss OS, but you wrote "Ubuntu", so I assumed you're trying to make the touchscreen work on Linux. I'm sorry, I can't help with Android or Windows issues.

Woobyxd, 2020-03-19:
Anyway, im downloading now ubuntu, gonna install it. Last time i tried ubuntu touch didnt work at all.
I have stock android 6.0.1 with working touchscreen (maxtouch.fw and silead.fw) also  working touch with Windows 10 (SileadTouch.sys). I wonder if i can do anything.