onitake/gsl-firmware #53: Multilaser M8W support

ngustavo, 2017-11-09:
Hello, I hope somenone can help me.
I'm trying to install the firmware and driver for Multilaser M8W on Arch Linux. There are 2 packages on the AUR but after installing them, calibrator still can't find it. What should I do?

https://www.driveridentifier.com//scan/desktop.php/desktop/591562F73281497EA59AD5D3AAA94F33?scanid=591562F73281497EA59AD5D3AAA94F33&friendly_pc_model=multilaser-m8w-type1-familytype1-tbd-by-oem-skutype1-sku0

onitake, 2017-11-14:
You will very likely not succeed with these.

Silead touchscreen controllers are terribly misdesigned and require firmware and hardware parameters that are precisely matched for each device. The Linux kernel has a generic driver built in, but it requires manual tuning if there are no matching settings for your device.

I recommend obtaining a firmware file first. If you still have a Windows driver package or Android modules, you can extract them from these. See the README for more information.

Then, you need to obtain the Linux kernel sources and add a matching parameter block to [drivers/platform/x86/silead_dmi.c](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/silead_dmi.c). Since the parameters are different for each device and not stored anywhere, you need to find them by trial and error. You can rely on my readme for a rough guide.

I wouldn't recommend using [gslx680-acpi](https://github.com/onitake/gslx680-acpi) any more as I stopped working on it a while ago.

ngustavo, 2017-11-15:
Ok, thank you. I installed Arch so I don't have it but I could try downloading the .exe

gabriel-messas, 2023-05-17:
@ngustavo I got you. I am working on it 😎.

Here is the firmware: [SileadTouch.zip](https://github.com/onitake/gsl-firmware/files/11493685/SileadTouch.zip)

gabriel-messas, 2023-05-17:
And here is the driver: [silead_ts.zip](https://github.com/onitake/gsl-firmware/files/11493933/silead_ts.zip)

ngustavo, 2023-06-07:
Thanks but I don't have the device with me anymore ;)