onitake/gsl-firmware #54: Port driver embedded in android kernel

wiltonlazary, 2017-11-15:
Dear onitake,
Please, If you can help me with informations.
I found firmware from a customer tablet inside android kernel image, its give me four files and one of then can function with the "modprobe silead", but its have one inch of dead zone at right region and jump to the border. 

How can i pass these parameters to silead module?
 input:b0018v0000p0000e0000-e0,1,3,14,k8B,9E,AC,D9,ra2F,30,32,35,36,39,mlsfw 

ANDROID:
>cat  /sys/class/input/event7/device/uevent    
    PRODUCT=18/0/0/0
    NAME="GSL_TP"
    PROP=2
    EV=10000b
    KEY=2000000 1000 40000800 0 0 0 0
    ABS=2658000 0
    MODALIAS=input:b0018v0000p0000e0000-e0,1,3,14,k8B,9E,AC,D9,ra2F,30,32,35,36,39,mlsfw

UBUNTU:
>cat /sys/bus/i2c/devices/i2c-MSSL1680:00/input/input16/uevent  
PRODUCT=18/0/0/0 
NAME="silead_ts" 
PHYS="input/ts" 
PROP=2 
EV=b 
KEY=400 0 0 0 0 0 
ABS=260800000000003 
MODALIAS=input:b0018v0000p0000e0000-e0,1,3,k14A,ra0,1,2F,35,36,39,mlsfw

#--------------------------------------#
#firmware extraction from android kernel

cat kenelimage \
    | hexdump -e '1/1  "0x%8.8_ax    "' -e '1/1  "%8._ad    "' -e '8/1 "%02X ""\n"""' \
    | grep -i -E "F0 00 00 00 02 00 00 00|7C 00 00 00 .. .. .. .." \
    | grep "F0 00 00 00 02 00 00 00" -B1 \
    ; \
echo -- \
    ; \
cat kenelimage \
    | hexdump -e '1/1  "0x%8.8_ax    "' -e '1/1  "%8._ad    "' -e '8/1 "%02X ""\n"""' \
    | grep -i -E "F0 00 00 00 02 00 00 00|7C 00 00 00 .. .. .. .." \
    | tail -n2
 
#--------------------------------------#

Thanks so much.

onitake, 2017-11-15:
The dead zone can be one of two things:
It's either the wrong firmware or the max_x parameter is wrong for your device.

There isn't much that can be done about the incorrect firmware, except trying to find a different one that works better.

As for the max_x parameter, you will have to tweak `drivers/platform/x86/silead_dmi.c` in the kernel source, then recompile `silead.ko`. I'm not sure if there is any way to pass an override for this parameter to the module. What does `modinfo silead` say?

wiltonlazary, 2017-11-15:
>dmesg give me this:

[  499.788275] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
[  499.788359] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy 
regulator
[  499.789130] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[  501.143335] input: silead_ts as /devices/pci0000:00/808622C1:05/i2c-5/i2c-MSSL1680:00/input/input13

wiltonlazary, 2017-11-15:
may be that android driver code behave different from linux silead driver

wiltonlazary, 2017-11-15:
i found a second firmware that behaves a little different, its responds to touch 5mm out of screen on the left and less 5mm on the right, and when i touch the up/right corner mouse jumps to the opposite
 screen position, seems that touch sensor is dislocated -5mm

onitake, 2017-11-15:
The problem with the Android driver (which was developed or commissioned by Silead) is that it has all parameters hardcoded. Changing them is not possible without recompiling the driver.

This is why I'm strongly suggesting that you switch to the official silead.ko Linux driver; it is customisable and generic. As soon as parameters for a new device are added, it will work for all users of that device. That is not the case with the Android driver.

wiltonlazary, 2017-11-16:
i'm using silead.ko linux driver, but i have only android kernel to extract firmware, vendors don't help me to put this in good use, i will need to learn a little more to make its function properly, if its possible.
If a learn some thing useful i will give it back to the community, i'm a developer with knowledge in several languages including c and c++, the lack of information from vendors is what make me suffer.

Thanks alot for your help.

onitake, 2017-11-16:
Can you get a hold of the Android driver sources, perhaps?

In the original driver code by Silead, the firmware is contained in a header file (similar to GSL_TS_CFG.h as used by the Windows driver), and compiled into the kernel module.

`fw_extractor` skims the compiled binary for constant data blocks that look like they were generated from this header file.

closed 2018-01-14 by wiltonlazary