onitake/gsl-firmware #55: please add teclast x3 plus firmware

DoJungJin, 2017-11-22:
I tested it working.

download firmware and silead_ts.fw files through below link.
http://download.djjproject.com/teclast_x3_plus/touch_firmware

device : 1680
width : 1980
height : 1500
options : track

I find touch panel size through experiment (about 120 times..)
that value is estimated value, but good working!

merge please. maybe this file can help others.

thank you.

onitake, 2017-12-11:
@ponces just added the requested firmware in #58 .
I'm closing this issue.

closed 2017-12-11 by onitake

ponces, 2017-12-11:
This was only possible with the findings of @DoJungJin !

onitake, 2017-12-11:
Thank you @DoJungJin too, of course!

If you want to help some more, please make the firmware work with the "official" kernel driver (silead.ko).
PR #56 should give you an idea on how to do this.
Thank you!

ponces, 2017-12-13:
What should be that file? firmware.fw or silead_ts.fw?
What should be its name? "gsl1680-teclast-x3-plus.fw"?
Where should I change the silead_dmi.c?

Thank you!

onitake, 2017-12-13:
`silead_ts.fw` is only for gslx680-acpi. `firmware.fw` should be the the "vanilla" firmware.
silead.ko needs a unique name so the files don't get mixed up.
Usually that includes the chip, manufacturer and device, so your suggestion (gsl1680-teclast-x3-plus.fw) is perfectly fine.

As soon as you have made and tested the modifications, please post them here for a short review.
Then send them to the linux-input mailing list: http://vger.kernel.org/vger-lists.html#linux-input

onitake, 2017-12-13:
Sorry, I misread your question: silead_dmi.c is in `drivers/platform/x86/silead_dmi.c` of the Linux kernel sources. You need to recompile the silead.ko kernel module to test your changes.

ponces, 2018-01-19:
It's working!
I could only make it to work using a combination of these three lines:
DMI_MATCH(DMI_SYS_VENDOR, "TECLAST"),
DMI_MATCH(DMI_PRODUCT_NAME, "X3 Plus"),
DMI_MATCH(DMI_BOARD_NAME, "X3 Plus")

Thank you for your help!

onitake, 2018-01-21:
Thanks @DoJungJin and @ponces - I merged all of your work.

onitake, 2018-01-22:
@ponces Did you submit your kernel patch to [linux-input](http://vger.kernel.org/vger-lists.html#linux-input)?
If not, please do. Thank you!

ponces, 2018-01-23:
Yes, just now - https://patchwork.kernel.org/patch/10179961/

Thank you for your help!