onitake/gsl-firmware #60: Can you help me to get a sileadtouch.tw?Thanks a lot for you.

lockey666, 2017-12-25:
The first,thanks a lot for you to read the mail.I am so sorry to take up your valuable time.But I think only you can answer my question.I bought a  x86 pad but I don't know it's brand.When i send it, it is normal. Then I backup its drivers and reinstall WIN10.But I forgot to backup sileadtouch.tw .When I open the new WIN10 system and install the drivers that I backup,I find the touch screen is wrong.It's X-Y swap.If I point the left up corner,the left low corner have reaction.If I point the left low corner,the right low corner have reaction.If I point the rigth low corner,the right up corner have reaction. I want to get the suitable siletouch.fw. Can you help me? I give you my driver.Look forward to your reply. Thanks a lot for you.
                                                                 
[KMDF HID Minidriver for Touch I2C Device_15.34.52.998_11-27-2015_10.0_x64.zip](https://github.com/onitake/gsl-firmware/files/1585940/KMDF.HID.Minidriver.for.Touch.I2C.Device_15.34.52.998_11-27-2015_10.0_x64.zip)
                                                           A Chinese Friend.

onitake, 2017-12-26:
I'm afraid that it will be very difficult to find the exact driver (or firmware) you need.

Truth is, many manufacturers release multiple versions of their tablets with different hardware configurations. That is not a problem for most system components, as the drivers are mostly generic and work for any compatible device. For Silead touch screen controllers, this is not the case. You need an exact firmware and/or driver for each specific device.

I do not know how to modify the Windows driver to change parameters, such as axis swapping or dimensions.
You could test if any of the drivers from [firmware/](firmware) work, perhaps you are lucky. Otherwise, I don't think I can help you.
You could also find out DMI information about your device (with http://gnuwin32.sourceforge.net/packages/dmidecode.htm for example, or by looking at the hardware info in Device Manager) and see if you can track down the manufacturer.

lockey666, 2017-12-27:
I am surprised to receive your reply. Thank a lot for your replay and sugesstions. I've succeeded in finding the firmware that I need.  If you don't mind, I want to share the process with you. I looked up a lot of information about silead.I found that the company used a tool called "Silead Ctouch" to write and debug the sileadtouch.fw. But it seems that the tool can't be downloaded from the Internet.Therefore，I thought it is difficult to edit sileadtouch.fw.Then I split the tablet computer to search for the number of the  touch screen.From my point of view, for those Flat Computer without Brand , the best solution to find Drivers and Firmware is referring to Touch Screen number.So I searched the number from SkyDrive and downloaded them.I test every drivers and I found a driver is suitable for my tablet.I was very lucky.In addition,I think it is a way to help them who searching for their windows sileadtouch drivers(sys) and firmware（fw or h）of silead touch screen.(For example: https://github.com/onitake/gsl-firmware/issues/35 )
    Yours sincerely.Thanks a lot for your help again.
                                                                                                                                 A Chinese Friend

closed 2018-01-04 by lockey666

onitake, 2018-01-04:
That is good to hear!
Glad you found the driver you needed.