onitake/gsl-firmware #67: please add firmware for chuwi surbook mini

nuttakit, 2018-02-22:
I have SileadTouch.sys file for the device. how can I extract firmware for linux from it?

onitake, 2018-05-14:
I just updated the readme with better instructions and added a tool for SileadTouch.sys firmware extraction.
Please give them a try!

singkiat, 2018-09-06:
Hi it seems like there are some issues with the surbook mini firmware:
1) Multi touch is broken with the current uploaded silead_tw.fs (forgot to add track option), can be fixed by using fwtool -c firmware.fw -m 1680 -w 2040 -h 1524 -t 10 -f track,yflip silead_ts.fw to reconvert the firmware

2) There is a dead zone on the left side of the screen that cannot be reached (I am using Phoenix OS). Based on what youling57 posted in the Phoenix OS forums, in Chinese, this is because there is a xmin offet. Currently fwtool -w sets the x size (xmax), can the fwtool be changed to support a xmin offset?

onitake, 2018-09-07:
There is no point in changing fwtool, because gslx680_acpi doesn't support xmin/ymin.

I recommend switching to the silead_ts that is available in mainline Linux - it supports standard input subsystem properties and has xmin/ymin support afaik.