onitake/gsl-firmware #68: Help--no multitouch

maximumsomething, 2018-02-24:
On my Teclast x3 plus tablet, I've installed the firmware and the gslx680 driver, and touch with a single finger works, but multitouch gestures don't. `xinput test` with multiple fingers never shows more than one "button press" at a time. I was wondering if this is an issue with known workarounds.

G-Michel, 2018-03-28:
have you tried the multitouch into chromium browser because all the software don't support multitouch into linux 
the only desktop enviroment that supports correctly touchscreeen is gnome 3

maximumsomething, 2018-04-18:
I have solved the issue by using Linux's silead driver instead of the gslx680 driver, (the firmware must be named /lib/firmware/silead/mssl1680.fw) and calibrating it with the calibration matrix: `2.087 0 0 0 2.7 0 0 0 1`

e.g. `xinput set-prop "silead_ts" --type=float "Coordinate Transformation Matrix" 2.087 0 0 0 2.7 0 0 0 1`

maximumsomething, 2018-05-06:
The multitouch problem was solved by switching drivers, so even if it is a firmware issue it is a driver issue as well.
I will try and extract the firmware from my device and see if it works differently from the firmware in this repository.

maximumsomething, 2018-05-06:
The firmware that I extracted from the SileadTouch.sys from my machine is exactly the same as the firmware in this repository. Maybe there is some special calibration done in the Windows driver software? Or, maybe the wrong firmware was extracted? (I used the instructions [linked from the Readme](https://github.com/onitake/gsl-firmware/blob/master/firmware/trekstor/surftab-twin-10.1-ST10432-8/README.md#command-to-find-the-offsets-used-for-extraction).

Scorpion2185, 2018-09-05:
@maximumsomething i need some help. Can you help me with the basic steps? You can see my problem here in my issue #86. I don' t know how use the firmware my device has been tested.