onitake/gsl-firmware #69: Change touchscreen resolution

zusturk, 2018-02-25:
Hi. Recently downloaded and installed gsl3676-chuwi-vi8.fw firmware from gsl-firmware/firmware/linux/silead/
And my touch resolution is awful - about 1/4 part of screen. How can I fix it? Maybe the reason is - desktop orientation? I'm working with 90 degrees clockwise desktop.

neckoed, 2018-02-25:
I don't even know what can I do - xinput_calibrator just ignores me: when I touch the screen - cursor just moves in the corner of the screen. 
Maybe siledad.ko firmware works bad, but how can I fix it? Is it possible to change configuration of the firmware for siledad.ko with fwtool to gslx680-acpi driver format and then extract it to work with siledad.ko, not with gslx680-acpi driver? Or how can I turn off siledad.ko and work just with gslx680-acpi module?

onitake, 2018-03-01:
The configuration for the tablets is contained in the kernel driver - if you want to modify it, you have to look at [drivers/platform/x86/silead_dmi.c](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/silead_dmi.c) and recompile the module.

But first, you should verify if the driver is loading the correct firmware. Do you see a line like
```
firmware: direct-loading firmware  gsl3676-chuwi-vi8.fw
```
in your kernel log?

If yes, and you're going to modify the dmi match table, it would be great if you could find a way to accurately identify your tablet, so a better match can be added to the kernel.
Apparently, many manufacturers change hardware configurations without changing the product (and its ID) itself.

neckoed, 2018-03-02:
No, I don't have this string. I had to rename "gsl3676-chuwi-vi8.fw" to "mssl1680.fw" and placed it inside "silead" folder. And there is nothing in kern.log about any firmware load. What should I do?

onitake, 2018-03-03:
In that case it's pretty clear it doesn't work: You've got the wrong firmware and/or panel settings.
The driver defaults to mssl1680.fw if no matching stanza in silead_dmi.c is found, and thanks to the horrible design of the Silead controllers, this will be wrong in almost all cases.

You can try experimenting with the settings, perhaps you can find some that will work. To do that, find out what DMI strings your device matches and add a new stanza to silead_dmi.c with tuned paramters.
If you can't get it working like this, you need to obtain a new firmware image from the Windows or Android driver for your device and add that. Browse the other bug reports, there's some hints on how to do that.

neckoed, 2018-03-05:
OK, I get it to work by updating kernel to 4.15. Now everything is fine, except swapped xy. Tried to calibrate with xinput_calibrate, but nothing changed.
Xorg.log tells me, that calibrate file is loaded, but nothing changed at all. 
And, I don't know why, there are two silead_ts in xinput -list - one in pointer and  one in keyboard. Maybe that's the reason?

neckoed, 2018-03-05:
OK, I get it. Touchscreen works, but how can I make it work in the 90° screen correctly?

neckoed, 2018-03-05:
Did it.

closed 2018-03-05 by neckoed

Yes21, 2019-02-03:
> OK, I get it. Touchscreen works, but how can I make it work in the 90° screen correctly?

But you didn't explain how you did :(