onitake/gsl-firmware #72: strange touchscreen behavior teclast x3 plus + questions

G-Michel, 2018-03-28:
hi i have installed the teclast x3 plus firware and globally it's working fine 
there is just one problem, if you don't use the touchscreen for 2 seconds the next input is not detected.
 a sort of sleep mode or something else do you know what is the issue ?

next question i have tried the last linux kernem 4.16 from yesterday and the touchscreen is not detected but the changelog tells that the support of teclast x3 plus touchscreen has been implemented is it normal ?

maximumsomething, 2018-04-18:
This strange behavior seems to be caused by the gslx680 driver--use the builtin silead driver as described [here](https://github.com/onitake/gsl-firmware/issues/68)

onitake, 2018-05-14:
I just updated the readme with better instructions for adding support to silead_ts.
Perhaps they may help you.