onitake/gsl-firmware #75: "Jumpy" touchscreen

maximumsomething, 2018-04-21:
When using Linux, the display isn't uniformly covered by the touchscreen. In other words, there are "gaps" on the screen, which can never be touched. This is most apparent when slowly scrolling: when certian places on the screen are scrolled over, the scrolled content will jump forward abot 25 pixels. This is on a Teclast X3 Plus.

onitake, 2018-09-17:
That most likely means you've got the wrong firmware for your device.
Which one did you use? And which driver?

maximumsomething, 2018-09-25:
All drivers had this same issue. I used the firmware in firmware/teclast/x3_plus.

maximumsomething, 2018-10-16:
@ponces You added the firmware for this device. Did you have this same issue?

ponces, 2018-10-17:
Can't reproduce your issue...

abcsds, 2019-05-07:
I have the same issue when working on trekstor primebook c11b. Both on Windows and Linux, with original drivers, and drivers downloaded from the official webpage. I just thought it was bad quality sensors. I'm using an active stylus to precisely measure the screen properties. One thing to notice is that the "jumpyness" is different on the x than on the y axis. The latter being more refined.

maximumsomething, 2019-05-08:
@abcsds Interesting that you also experience it in Windows. Maybe it's a similar issue?

BTW don't buy this tablet. Not a week after the warranty expired the battery started swelling, cracking the screen, and Teclast never responded.