onitake/gsl-firmware #83: Mirrored X axis on the other half of the screen?

polyzium, 2018-07-04:
I've just installed the converted firmware from the Windows driver that my manufacturer provides.
However, the other half of the screen mirrors the X axis (the first half is alright except there are deadzones)

onitake, 2018-07-05:
That means you're not using the correct firmware for your panel.
Did you only find that single firmware in the driver?
How did you extract it?

polyzium, 2018-07-06:
I use the GSLx680 ACPI driver
The extraction command was `tools/fwtool -c SileadTouch/SileadTouch.fw -3 -m 1680 -w 2200 -h 1450 -t 10 silead_ts.fw` then `sudo cp silead_ts.fw /lib/firmware`
The 2200x1400 resolution is just me trying to manually calibrate the TS, I don't know the exact values, but the mirrored screen is not resolution-dependent.
I don't have the tablet on my hand right now, so I'll upload the driver I used (for reference) later.

polyzium, 2018-07-07:
OK so the driver was broken, I downloaded another one and it works as intended, now all I need to do is to calibrate.

closed 2018-07-07 by polyzium