onitake/gsl-firmware #86: Mediacom tablet

Scorpion2185, 2018-08-29:
Mediacom Winpad 7.0 W700corrisponde al SurfTab® wintron 7.0
Mediacom WinPad 8.0 W801 corrisponde al SurfTab® wintron 8.0
Mediacom WinPad 10.1 W101 corrisponde al SurfTab® wintron 10.1

onitake, 2018-08-29:
Which driver are you using?
gslx680_acpi .ko or silead.ko?

Scorpion2185, 2018-08-29:
What should i use? 
Now i don' t have any of them yet. 
I should copy the` silead_ts.fw` and `firmware.fwf` files from https://github.com/onitake/gsl-firmware/tree/master/firmware/trekstor/surftab7new into this repository https://github.com/onitake/gslx680-acpi.
Then run `make` ?

onitake, 2018-08-30:
No, that won't help.
Linux drivers always load their firmware from /lib/firmware.

I stopped working on gslx680-acpi, but it may still be useful for testing. Consult the readme in that project for instructions, maybe.

If you need a more permanent solution, you should add your tablet's DMI ids to drivers/platform/x86/silead_dmi.c in the Linux kernel sources. If it's actually the same firmware as for another device, you can reference that file.
See https://github.com/onitake/gsl-firmware#silead_ts - it describes how to do this. Unfortunately, these tables are compiled into the kernel, not the driver, so you need to recompile the kernel whenever you change them. I think there are also some module options in silead.ko, maybe run `modinfo silead` to see if it supports passing parameters on load.

onitake, 2018-08-30:
You can't see it anywhere, unfortunately.
This is a parameter you need to find out by trial and error. :(

And about the touch points - that's very well possible. I think 10 finger touch was never properly tested.

Scorpion2185, 2018-08-30:
My device has been tested so can someone tell me the touchscreen size or better share the kernel?
The `gsl1680-Mediacom-W700.fw` file should be the `silead_ts.fw` file the from https://github.com/onitake/gsl-firmware/tree/master/firmware/trekstor/surftab7new?
There is some guide to how recompile the kernel in this case?
And the .ko files? Am i missing something?

Scorpion2185, 2018-09-02:
@onitake So the `silead_ts.fw` firmware for my deivce is already fnished. How can I use it?<br>
This section taken from https://github.com/onitake/gsl-firmware#silead_ts states:

> silead_ts
> 
> First, decide on a unique name for the firmware. The recommended
> format is: `gsl<chip name>-<manufacturer>-<device>.fw` If you don't know
> the chip name, use gsl1680. Manufacturer and device name should be
> shortened to make the name readable, but still sufficiently unique.
> 
> Fetch the current Linux source code and open
> `drivers/platform/x86/silead_dmi.c` in a text editor.
> 
> Supposing you named your firmware `gsl1680-mycompany-mytablet.fw` and
> the touchscreen has a resolution of 128x128 points, add a new section
> to the file:

    static const struct property_entry mycompany_mytablet_props[] = {
    	PROPERTY_ENTRY_U32("touchscreen-size-x", 128),
    	PROPERTY_ENTRY_U32("touchscreen-size-y", 128),
    	PROPERTY_ENTRY_STRING("firmware-name", "gsl1680-mycompany_mytablet.fw"),
    	PROPERTY_ENTRY_U32("silead,max-fingers", 10),
    	{ }
    };
    
    static const struct silead_ts_dmi_data mycompany_mytablet_data = {
    	.acpi_name	= "MSSL1680:00",
    	.properties	= mycompany_mytablet_props,
    };
> In some rare cases, the `acpi_name` may be different.
> 
> Further below, there is a list named `silead_ts_dmi_table`:
> 
`static const struct dmi_system_id silead_ts_dmi_table[] = {`
> 
> Add a new entry to this list:
> 
	{
		/* Teclast X3 Plus */
		.driver_data = (void *)&mycompany_mytablet_data,
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "MyCompany"),
			DMI_MATCH(DMI_PRODUCT_NAME, "MyTablet"),
			DMI_MATCH(DMI_BOARD_NAME, "Generic Board"),
		},
	},
> 
> The exact values and suitable matching tags depend on the particular
> device. You can find them with the help of `dmidecode`. Look for the
> section called "System Information".
> 
> [See here for an example patch][3].
> 
> After adding the override tables, you have to recompile your kernel,
> as the DMI overrides are outside the context of the driver. Install
> and reboot. You may also try reloading the kernel using `kexec`, but
> this can lead to an unstable system and is not recommended.

I suppose that the `gsl1680-mycompany-mytablet.fw` file is the `silead_ts.fw` firmware of before.<br>
How can i recompile the kernel? I' m trying [this](https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel) guide.<br>

I used `apt-get source linux-headers-4.15.0-33-generic` and got the `linux-4.15` folder of 800 MB. That has the path `drivers/platform/x86/silead_dmi.c` in which i already made the `drivers/platform/x86/silead_dmi.c` file.<br> 

Following the guide i used `apt-get source linux-image-$(uname -r)` and got the `linux-signed-4.15.0` folder of 118 kB.<br>
Now using `fakeroot debian/rules editconfigs` i get an error:

`dh editconfigs
dh: Unknown sequence editconfigs (choose from: binary binary-arch binary-indep build build-arch build-indep clean install install-arch install-indep)
debian/rules:35: recipe for target 'editconfigs' failed
make: *** [editconfigs] Error 2`

How can i fix this error and recompile the kernel?<br>
And the folder obtained from `apt-get source linux-headers-4.15.0-33-generic` which has the path `drivers/platform/x86/silead_dmi.c`?<br>
And the firmware?

onitake, 2018-09-05:
You don't _have_ to use the package management tools to recompile the kernel, but it will certainly make installation and uninstallation easier.

I have never heard of `editconfigs`, normally you would use `make oldconfig` (will prompt you for driver flags that have changed since release), `make config` (will prompt you for _all_ flags, not recommended) or `make menuconfig` (will display a nice text menu, use that if oldconfig doesn't help).
Prepending each command with `fakeroot` may or may not be necessary.

I'm pretty sure the silead driver is disabled in the Debian/Ubuntu kernels, so you'd have to hunt it down and enable it as a module.

Scorpion2185, 2018-09-05:
@onitake  meanwhile i recompiled and installed the kernel.
I put the `gsl1680-mediacom-w700.fw` in `linux-source-4.15.0/firmware` (and modified again the `drivers/platform/x86/silead_dmi.c`).
```
dmesg | grep silead
[...]
Direct firmware load for silead/mssl1680.fw failed with error -2
```
So i placed in `/lib/fimrware/silead` the `gsl1680-mediacom-w700.fw` but with the name `mssl1680.fw`.
Now i get:
```
dmesg | grep silead
[   38.359916] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
[   38.359948] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
[   38.361416] silead_ts i2c-MSSL1680:00: Silead chip ID: 0xB4820000
[   39.584318] silead_ts i2c-MSSL1680:00: Firmware load error -110
[   39.584579] silead_ts: probe of i2c-MSSL1680:00 failed with error -110
[ 2307.760993] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
[ 2307.761034] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
[ 2308.769843] silead_ts i2c-MSSL1680:00: Chip ID read error -110
[ 2308.770496] silead_ts: probe of i2c-MSSL1680:00 failed with error -110
```
The touchscreen isn' t working at all.

onitake, 2018-09-05:
That almost certainly means you got the wrong firmware. And not just the wrong calibration data, but firmware for the wrong chip.
Also, if your DMI quirk is working, it will ask for the firmware name you put there.
mssl1680.fw is a fallback if no specific section matches.
Please recheck your device match strings, they're likely incorrect...

Scorpion2185, 2018-09-05:
```
#include <linux/acpi.h>
#include <linux/device.h>
#include <linux/dmi.h>
#include <linux/i2c.h>
#include <linux/notifier.h>
#include <linux/property.h>
#include <linux/string.h>

struct silead_ts_dmi_data {
	const char *acpi_name;
	const struct property_entry *properties;
};

static const struct property_entry mediacom_w700_props[] = {
	PROPERTY_ENTRY_U32("touchscreen-size-x", 884),
	PROPERTY_ENTRY_U32("touchscreen-size-y", 632),
	PROPERTY_ENTRY_STRING("firmware-name",
			      "gsl1680-mediacom-w700.fw"),
	PROPERTY_ENTRY_U32("silead,max-fingers", 10),
	PROPERTY_ENTRY_BOOL("silead,home-button"),
	{ }
};

static const struct silead_ts_dmi_data mediacom_w700_data = {
	.acpi_name	= "MSSL1680:00",
	.properties	= mediacom_w700_props,
};
```
Is the beginning of `drivers/platform/x86/silead_dmi.c`.
What' s wrong?

onitake, 2018-09-05:
That's just the device parameters, you still need a DMI match for your particular device.
Otherwise the kernel won't recognise it and activate the property structure you created.

onitake, 2018-09-05:
Oh, and one more thing: silead_ts.fw is firmware for my older gslx680_acpi driver. It won't work with silead.ko. Unfortunately, there is no heuristics in the driver that would reject an incorrect file, and it would happily try to load it, rendering the chip unusable until the next power cycle.
silead_ts.fw has a more streamlined format and some sanity checks, but the developers of silead.ko wanted to stick with the original vendor firmware instead. Tough luck...

Scorpion2185, 2018-09-05:
So what i should do? Extract the firmware from the windows drivers? And then where i should put it?

Scorpion2185, 2018-09-05:
So i should use the `firmware.fw`? 
Then where i should put it?

onitake, 2018-09-05:
First, make sure you have firmware in the format that is stored in the Windows driver. Don't convert it into silead_ts.fw format. Copy that to /lib/firmware/silead/<your_custom_firmware_name>.fw
Second, add a suitable DMI_MATCH section (as you copy-pasted above) to silead_dmi.c. You need to find out what strings to put there from the output of dmidecode.

onitake, 2018-09-05:
There's no harm in trying - if you know the firmware is the same for both devices, you don't even have to copy the file. Just reference the same filename in your device property section.

So, for example, if you want to try gsl1686-surftab-wintron70-st70416-6.fw on your device, your properties should look like this:
```c
static const struct property_entry mediacom_w700_props[] = {
	PROPERTY_ENTRY_U32("touchscreen-size-x", 884),
	PROPERTY_ENTRY_U32("touchscreen-size-y", 632),
	PROPERTY_ENTRY_STRING("firmware-name",
			      "gsl1686-surftab-wintron70-st70416-6.fw"),
	PROPERTY_ENTRY_U32("silead,max-fingers", 10),
	PROPERTY_ENTRY_BOOL("silead,home-button"),
	{ }
};
```

As for the DMI_MATCH, look at this part:
```c
{
	/* Teclast X3 Plus */
	.driver_data = (void *)&mycompany_mytablet_data,
	.matches = {
		DMI_MATCH(DMI_SYS_VENDOR, "MyCompany"),
		DMI_MATCH(DMI_PRODUCT_NAME, "MyTablet"),
		DMI_MATCH(DMI_BOARD_NAME, "Generic Board"),
	},
},
```
There's a section in silead_dmi.c where all these are listed.
Add a new stanza like the one above and change the parameters so they match your board.
Use the `dmidecode` utility to find out what strings are suitable to accurately identify it.

onitake, 2018-09-05:
Unfortunately, I don't know which values from dmidecode correspond to the keys I listed, you'll have to look this up somewhere else... or use trial-and-error.

Scorpion2185, 2018-09-05:
I don' t need to copy the firmware? Just modify the silead_dmi.c ??

Scorpion2185, 2018-09-05:
My device has not drivers if i need them i must use those of surftab, maybe also the  DMI_MATCH,?

onitake, 2018-09-05:
silead_dmi.c is part of the kernel, not the driver. So, technically, yes, you don't need to modify the driver. But you need to modify the kernel so it loads the driver with the correct parameters.

The DMI_MATCH will be different for sure. But the rest may be the same or similar.

As for the values from dmidecode, I think you need to look at the "Base Board Information" section. There should be a vendor and product name there. I don't know about the board name though. Perhaps the version field

Scorpion2185, 2018-09-05:
If i only modified the  silead_dmi.c telling it to look for gsl1686-surftab-wintron70-st70416-6.fw . Then how the system will see that firmware?

onitake, 2018-09-05:
Oh. Good point!
That one's easy - just follow the instructions here: https://github.com/onitake/gsl-firmware/tree/master/firmware/linux
You actually only need to copy the file you're interested in - but there's no harm in copying the whole folder. Just make sure that gsl1686-surftab-wintron70-st70416-6.fw ends up in /lib/firmware/silead/

Scorpion2185, 2018-09-05:
I did it using the https://github.com/onitake/gsl-firmware/blob/master/firmware/trekstor/surftab7old/firmware.fw but the touch is pretty bad (i can' t use it).
Should i use the https://github.com/onitake/gsl-firmware/blob/master/firmware/linux/silead/gsl1686-surftab-wintron70-st70416-6.fw file will be better right?
And i need to made the proper silead_dmi.c? 
And when you use the touch don' t you should have a on-screen keyboard?

Scorpion2185, 2018-09-05:
```
static const struct silead_ts_dmi_data mediacom_w700_data = {
	.acpi_name	= "MSSL1680:00",
```
This is way is looking for `mssl1680.fw`?

Scorpion2185, 2018-09-05:
After:
```
struct silead_ts_dmi_data {
	const char *acpi_name;
	const struct property_entry *properties;
};
```
I added:
```
static const struct property_entry mediacom_w700_props[] = {
	PROPERTY_ENTRY_U32("touchscreen-size-x", 884),
	PROPERTY_ENTRY_U32("touchscreen-size-y", 632),
	PROPERTY_ENTRY_STRING("firmware-name",
			      "gsl1686-surftab-wintron70-st70416-6.fw"),
	PROPERTY_ENTRY_U32("silead,max-fingers", 10),
	PROPERTY_ENTRY_BOOL("silead,home-button"),
	{ }
};
```
Right?

Scorpion2185, 2018-09-05:
Then add a new DMI_MATCH section:
```
static const struct dmi_system_id silead_ts_dmi_table[] = {
        {
		/* Mediacom WinPad 7.0 W700 */
		.driver_data = (void *)&surftab_wintron70_st70416_6_data,
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "INSYDE Corp."),
			DMI_MATCH(DMI_PRODUCT_NAME, "ST70416-6"),
			/* Exact match, different versions need different fw */
			DMI_MATCH(DMI_BIOS_VERSION, "MC.G.WI71C.MGBMRBA03"),
		},
	},
```
`.driver_data = (void *)&surftab_wintron70_st70416_6_data ` is ok? i should left those of the surftab?

`DMI_MATCH(DMI_SYS_VENDOR, "INSYDE Corp.")` i used  INSYDE Corp. instead of INSYDE becasue of the dmidecode.

`DMI_MATCH(DMI_PRODUCT_NAME, "ST70416-6"),` is important? iì m not sure which field is.

`DMI_MATCH(DMI_BIOS_VERSION, "MC.G.WI71C.MGBMRBA03"` becasue of dmidecode.

onitake, 2018-09-06:
All the fields are important - if only one of them doesn't match, the DMI match will fail.
I assume this is "Product Name" in the "Base Board Information" section. Note that you may not _need_ to use this field, just make sure that your `matches` block accurately identifies your device, and only that.

onitake, 2018-09-06:
You can even reference the properties block of the Surftab, if you think all the parameters are the same. In that case, just define the DMI match, and that should be it.

Scorpion2185, 2018-09-06:
```
static const struct silead_ts_dmi_data mediacom_w700_data = {        
	.acpi_name	= "MSSL1680:00",                               
	.properties	= mediacom_w700_props,                       
};

```
The section above must be added?
And the DMI_MATCH that i found in the `dmidecode` output were in:
```
Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: INSYDE Corp.
	Version: MC.G.WI71C.MGBMRBA03
[...]
Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: MEDIACOM
	Product Name: WinPad 7 W10 - WPW700
```
The "Base Board Information" section wasn' t much of help:
```
Handle 0x0002, DMI type 2, 17 bytes
Base Board Information
	Manufacturer: Type2 - Board Manufacturer
	Product Name: Type2 - Board Product Name
	Version: Type2 - Board Version
	Serial Number: Type2 - Board Serial Number
	Asset Tag: Type2 - Board Asset Tag
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Type2 - Board Chassis Location
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0
```

Scorpion2185, 2018-09-06:
I want to reuse the driver of the surftab mine don' t exixt anywhere.
driver_data??? in which section?
Can you made the ?silead_dmi.c? ?

Scorpion2185, 2018-09-06:
[silead_dmi.c.zip](https://github.com/onitake/gsl-firmware/files/2356420/silead_dmi.c.zip)

Scorpion2185, 2018-09-06:
i set:
```
static const struct dmi_system_id silead_ts_dmi_table[] = {
        {
		/* Mediacom WinPad 7.0 W700 */
		.driver_data = (void *)&surftab_wintron70_st70416_6_data,
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "INSYDE Corp."),
			DMI_MATCH(DMI_PRODUCT_NAME, "WinPad 7 W10 - WPW700"),
			DMI_MATCH(DMI_BIOS_VERSION, "MC.G.WI71C.MGBMRBA03"),
		},

```
And still the touch in unusable and the system doesn' t load the proper fimrware:
```
 dmesg | grep silead
[   38.316176] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
[   38.316214] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
[   38.319954] silead_ts i2c-MSSL1680:00: Silead chip ID: 0xB4820000
[   39.792375] input: silead_ts as /devices/platform/80860F41:03/i2c-3/i2c-MSSL1680:00/input/input7
```
What can be?

Scorpion2185, 2018-09-06:
Now i ' m trying:
```
 {
		/* Mediacom WinPad 7.0 W700 */
		.driver_data = (void *)&surftab_wintron70_st70416_6_data,
		.matches = {
			DMI_MATCH(DMI_SYS_VENDOR, "INSYDE Corp. - Type2 - Board Manufacturer"),
			DMI_MATCH(DMI_PRODUCT_NAME, "WinPad 7 W10 - WPW700 - Type2 - Board Product Name"),
			DMI_MATCH(DMI_BIOS_VERSION, "MC.G.WI71C.MGBMRBA03 - Type2 - Board Version"),
		},
```
As you suggested i used the System Information and Base Board Information, any error?

Scorpion2185, 2018-09-06:
Tried same situation.
What i should try now?
I really don 't know.

onitake, 2018-09-07:
Can you try 
```
{
	/* Mediacom WinPad 7.0 W700 */
	.driver_data = (void *)&surftab_wintron70_st70416_6_data,
	.matches = {
		DMI_MATCH(DMI_SYS_VENDOR, "MEDIACOM"),
		DMI_MATCH(DMI_PRODUCT_NAME, "WinPad 7 W10 - WPW700"),
	},
},
```
(no DMI_BIOS_VERSION)

Scorpion2185, 2018-09-07:
Finally is working!!!
But how i can use the function hold as mouse 2 as  it should be?
And there is a screen-keyboard that is designed for this hardware?

Scorpion2185, 2018-09-08:
BUT using dmesg | grep silead it seems that is using the backup firmware which is the same but with the different name.
How can i check?

Scorpion2185, 2018-09-09:
I set ` gsettings set org.gnome.desktop.a11y.mouse secondary-click-enabled "true"? but i working only with the physical mouse.

onitake, 2018-09-09:
I don't know about hold-click, haven't tried it myself.

As for the onscreen keyboard, I tried a few and found most of them terrible to use. Gnome3 lacks a lot of keys, KDE doesn't have any keyboard at all, enlightenment is good but doesn't integrate with other desktops well. The best one I found was the Ubuntu Unity keyboard, but since they switched to Gnome3, I don't know if it still exists.
But that was a long time ago, perhaps things have changed now.

Scorpion2185, 2018-09-09:
Onboard  (that is the ubuntu default keyboard so maybe is the one you ' ve said) is good but open when editing text is not working and Lubuntu doesn' t support multit-touch. Unity and gnome yes so i will install ubuntu-desktop with unity. With the 18.04 gnome is default but you can install unity.
What about speakers and camera? And why ` dmesg | grep silead' is finding backup firmware?

Scorpion2185, 2018-09-09:
Also grub menu (the one with recovery etc.) is a black screen but the countdown.

I mean that it loads the mssl 1680.fw (same firmware as the one of silead_dmi.c but with that name).
```
dmesg | grep silead
[   38.308220] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
[   38.308257] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
[   38.308511] silead_ts i2c-MSSL1680:00: Silead chip ID: 0xB4820000
[   39.744384] input: silead_ts as /devices/platform/80860F41:03/i2c-3/i2c-MSSL1680:00/input/input6
```

Scorpion2185, 2018-09-12:
I tried gnome (default ubuntu 18.04 desktop) is nice i can rotate the screen and it opens automatically a keyboard (If i `Slide up from the bottom screen edge` or when searching apps.) Only that when the display is landscape the keyboard is half out of the monitor. 

As soon i touch the screen the cursor disappears and i moving it i create the "square for select things". I can use it for select and double click but no right click (touch hold and release). Is very hard to use the right click i did some times.
It works almost fine on the launcher.

Scorpion2185, 2018-09-12:
In lubuntu the cursor remains but after trying right click (and failing it) i can' t move anymore the items.
There is some problem with the click.
Btw with 1 Gb of ram i will use Lubuntu,

Scorpion2185, 2018-09-12:
I tried gnome (default ubuntu 18.04 desktop) is nice i can rotate the screen and it opens automatically a keyboard (if i `Slide up from the bottom screen edge` or when searching apps.) Only that when the display is landscape the keyboard is half out of the monitor. 

First i have to touch the item then being careful i can use right click. So in the launcher i got problems, because first i have to select an item.
Installing `florence` gnome should integrate it https://www.makeuseof.com/tag/replace-gnome-onscreen-keyboard-florence-virtual-keyboard-linux/.
But with 1 Gb of ram i must use Lubuntu,

Scorpion2185, 2018-09-12:
I tried gnome (default ubuntu 18.04 desktop) is nice i can rotate the screen and it opens automatically a keyboard (if i `Slide up from the bottom screen edge` or when searching apps.) Only that when the display is landscape the keyboard is half out of the monitor (and miss some keys like F1 etc.).

First i have to touch the item then being careful i can use right click. 
So in the launcher i got problems, because first i have to select an item.
Installing `florence` gnome should integrate it https://www.makeuseof.com/tag/replace-gnome-onscreen-keyboard-florence-virtual-keyboard-linux/.
But with 1 Gb of ram i must use Lubuntu.

Scorpion2185, 2018-09-12:
I tried gnome (default ubuntu 18.04 desktop) is nice i can rotate the screen and it opens automatically a keyboard (if i `Slide up from the bottom screen edge` or when searching apps.) Only that when the display is landscape the keyboard is half out of the monitor (and miss some keys like F1 etc.).

First i have to select the item then being careful i can use right click. 
So in the launcher i got problems, because first i have to select an item.
If i install `florence` gnome should integrate it https://www.makeuseof.com/tag/replace-gnome-onscreen-keyboard-florence-virtual-keyboard-linux/.
But with 1 Gb of ram i must use Lubuntu.
Also in Lubunut

Scorpion2185, 2018-09-12:
But after trying right click i can't drag selected items and when i move the cursor it always open the selection square.

Scorpion2185, 2018-09-12:
But after trying right click i can't drag selected items anymore.

Scorpion2185, 2018-09-12:
But after that i tried the right click i can't drag selected items anymore.

Scorpion2185, 2018-09-12:
Again in Lubuntu hard times to use right click ( so i can drag what i want),  i think that there is something missing.

Scorpion2185, 2018-09-12:
Again in Lubuntu hard times to use right click ( so i can drag what i want),  i think that there is something missing. 
In ubuntu (deafult) the selected item become lighter and then i can easily right click it.

Scorpion2185, 2018-10-15:
@onitake  now my device is being tested (only bad right click) so you can add it in the repo? And in the official kernel `silead_dmi.c`?
Thanks for this work, if you don' t have any suggestion for the right click i' ll close the issue.
In the other devices is the same?

Scorpion2185, 2018-10-31:
Problem fixed:
Don' t forget to install moustweaks so you can use onboard mouse keys, though with the touch they work very bad so actually you won't use it.
But you can use the right click perfectly:
With the touch you must tap quickly and shortly 3 times (in the last hold) and the right click is perfect. 
But you must have edited with `dconf-editor` the field `secondary-click-enabled`, in `org/gnome/desktop/a11y/mouse`.

Here drivers for intel atom and more: https://linuxiumcomau.blogspot.com/2017/06/customizing-ubuntu-isos-documentation.html
For my device i need the kernel 4.18.15 to have the audio.

closed 2018-10-31 by Scorpion2185