onitake/gsl-firmware #87: Insignia Flex NS-P10w8100/Where do I start?

Hom3Lite, 2018-09-05:
Hi All! I've been trying to get Android x86 working on my tablet. Everything that I've tested seems to work just fine, except for the touch screen. It appears to be a Silead device with the HID of MSSL1680. On the Windows side of this dual-boot setup, the drivers associated with it are SileadTouch.sys and mshidkmdf.sys. As I understand it, I need to run the script "scanwindrv" to extract the necessary driver. How exactly do I do this? I'll be honest with you, this is a bit over my head, but I'm desperately trying to learn. I didn't see any notes on what language the script was written in, so I'm not exactly sure how to execute it. Is there some sort of step by step guide that I can follow to get this all working?

onitake, 2018-09-05:
There is, please look at the readme. I agree, it's a bit technical, but the process is very tedious and technical, especially if you want to add support for your device to the mainline kernel.

What you need to extract isn't the driver, it's the firmware for your device.
Sadly, every Silead controller needs different firmware, because calibration data and other device-specific parameters are stored in there. So, you need to extract it from a vendor driver first.

scanwindrv is written in Perl, and you should be able to execute it right from the command line - as long as you have a perl interpreter installed. The readme also tells you the command you need to run:
```bash
tools/scanwindrv SileadTouch.sys
```

Davidhdz28, 2025-01-22:
@Hom3Lite Hi Hom3Lite, I have a tablet same modelo NS-P10W8100 with windows 10 home,
You will have all the drivers and an image of your operating system. I formatted mine and now I have problems with 3 sensors (touch, proximity or celerometer and rear camera).
I have tried several drivers that I have found on the internet but I have not had success.
I would appreciate it very much.