onitake/gsl-firmware #91: Can you make for Chuwi Hi 8 Air

zondaime13, 2018-10-25:
Hi ya. i’m a big fan of your project. May you make firmware for Chuwi Hi 8 Air new release 2018, please because I want to run it on Kali Linux. Thank you. 
I have a windows driver for it and I will send you as soon as possible!

onitake, 2018-11-07:
Did you find the Windows driver?
If yes, please submit it via a PR, including a readme.
We'll worry about conversion later.
Thanks!

zondaime13, 2018-11-08:
Thank you so much for the reply. And yes! I found the Chuwi Hi8 Air Driver
from the forum of Chuwi:
This is the Windows Driver Link:
http://www.mediafire.com/file/ven2gq50mx2wxem/HI8+Air-driver.rar

Thank you so much. Please make my dream come true that I can using Kali
Linux in Chuwi Air Driver Tablet! Thank you!

onitake, 2018-11-09:
Can you prepare a readme similar to [this one](https://raw.githubusercontent.com/onitake/gsl-firmware/master/firmware/chuwi/hibook/README.md) and post it into [a gist](https://gist.github.com/)?
Thanks!

zondaime13, 2018-11-25:
I'm sorry for the late reply but I don't have much information about that
device..This is all I have about the Chuwi-Air Tablet:
Chuwi/HiBook
---------------------------------------------

| Item                      | Description |
|---------------------------|-------------|
| Manufacturer              | Chuwi |
| Device                    | Hi-Air |
| Website                   |
https://www.chuwi.com/product/items/Chuwi-Hi8-Air.html |
| Vendor driver (Windows)   | https://forum.chuwi.com/thread-5785-1-1.html |
| Extracted firmware        | Unknow |
| Firmware for gslx680-acpi | Unknow |
| Display resolution        | 1920x1200 |
| Touch panel resolution    | Unknow |
| Touch controller          | GSL3680 |
| Multitouch support        | Yes |
| Finger tracking           | Yes |
| Mirrored horizontally     | Unknow |
| Mirrored vertically       | Unknow |
| Rotating                    | Yes
| Axes swapped              | No |

Yes21, 2019-01-13:
Hi, I just got a Chuwi Hi 8 Air.
I did try to use the Hi 8 driver, but it doesn't compile :
`cc1: error: code model kernel does not support PIC mode`

I've installed Xubuntu 18.04 on the tablet.

I can't find the Chuwi Hi 8 Air firmware in this repo ...

Thanks

Yes21, 2019-01-13:
I could extract the firmware with `tools/untscfg`, and then format it with `tools/fwtool`.
But I can't compile. Always the same error : 
`cc1: error: code model kernel does not support PIC mode`

onitake, 2019-01-14:
@Yes21 If possible, please use the silead driver that ships with the Linux kernel.
It requires some manual work if your device isn't supported yet. The README describes what you must do.
Also, you should not use fwtool with the in-kernel driver. To extract firmware from the Windows driver, please use `scanwindrv`.

Yes21, 2019-01-14:
@onitake Thanks for your answer.
I've read the README. I will use `scanwindrv` with .sys driver.
If I use the kernel silead driver, I need to recompile the kernel, but do I need to run `make` to build the .ko ?

Yes21, 2019-01-14:
I've still tried to make `make` work but impossible !!
The `stdarg.h'` is in `/usr/lib/gcc/x86_64-linux-gnu/7/include`. I did include it to the `$PATH`, but didn't work. 
Did you ever succeed in compiling the .ko module on ubuntu 18.04 ??

Now I give up and will try to use the in-kernel driver method.
But I'm afraid to have to recompile the kernel on my Hi8 Air !!

onitake, 2019-01-14:
You shouldn't have to do any of that, as long as you have the kernel header package and its dependencies installed. What does this command print?
```
apt policy linux-headers-amd64
```
A special compiler is required for building the kernel and kernel modules, but it should be installed along linux-headers automatically.

Yes21, 2019-01-14:
```
# apt policy linux-headers-amd64
linux-headers-amd64:
  Installé : (aucun)
  Candidat : (aucun)
 Table de version :

# apt policy linux-headers-4.15.0-43-generic
linux-headers-4.15.0-43-generic:
  Installé : 4.15.0-43.46
  Candidat : 4.15.0-43.46
 Table de version :
 *** 4.15.0-43.46 500
        500 http://fr.archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages
        500 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages
        100 /var/lib/dpkg/status

```

Yes21, 2019-01-14:
```
# apt policy linux-kbuild-4.15
N: Impossible de trouver le paquet linux-kbuild-4.15
N: Impossible de trouver de paquet correspondant à l'expression rationnelle « linux-kbuild-4.15 »
N: Impossible de trouver de paquet correspondant à l'expression rationnelle « linux-kbuild-4.15 »
```

onitake, 2019-01-14:
Huh.
I assumed the kernel build packages are similar between Debian and Ubuntu.
Let me check what you need on Ubuntu.

onitake, 2019-01-14:
See https://bugs.launchpad.net/ubuntu/+bug/830550 for a similar issue and workaround, but I really think this should not be necessary.
Did you install `build-essential`?

Yes21, 2019-01-14:
> Did you install `build-essential`?

Yes
Here is the output of `make` :

```
# make
make -C /lib/modules/4.15.0-43-generic/build ARCH=x86_64 CROSS_COMPILE= M=/home/yves/Téléchargements/gslx680-acpi modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-43-generic »
  CC [M]  /home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.o
In file included from ./include/linux/list.h:5:0,
                 from ./include/linux/module.h:9,
                 from /home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.c:19:
./include/linux/types.h:17:9: error: unknown type name ‘__kernel_ino_t’
 typedef __kernel_ino_t  ino_t;
         ^~~~~~~~~~~~~~
./include/linux/types.h:18:9: error: unknown type name ‘__kernel_mode_t’
 typedef __kernel_mode_t  mode_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:21:9: error: unknown type name ‘__kernel_off_t’
 typedef __kernel_off_t  off_t;
         ^~~~~~~~~~~~~~
./include/linux/types.h:22:9: error: unknown type name ‘__kernel_pid_t’
 typedef __kernel_pid_t  pid_t;
         ^~~~~~~~~~~~~~
./include/linux/types.h:23:9: error: unknown type name ‘__kernel_daddr_t’
 typedef __kernel_daddr_t daddr_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:25:9: error: unknown type name ‘__kernel_suseconds_t’
 typedef __kernel_suseconds_t suseconds_t;
         ^~~~~~~~~~~~~~~~~~~~
./include/linux/types.h:26:9: error: unknown type name ‘__kernel_timer_t’
 typedef __kernel_timer_t timer_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:27:9: error: unknown type name ‘__kernel_clockid_t’
 typedef __kernel_clockid_t clockid_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:32:9: error: unknown type name ‘__kernel_uid32_t’
 typedef __kernel_uid32_t uid_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:33:9: error: unknown type name ‘__kernel_gid32_t’
 typedef __kernel_gid32_t gid_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:34:9: error: unknown type name ‘__kernel_uid16_t’
 typedef __kernel_uid16_t        uid16_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:35:9: error: unknown type name ‘__kernel_gid16_t’
 typedef __kernel_gid16_t        gid16_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:41:9: error: unknown type name ‘__kernel_old_uid_t’
 typedef __kernel_old_uid_t old_uid_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:42:9: error: unknown type name ‘__kernel_old_gid_t’
 typedef __kernel_old_gid_t old_gid_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:46:9: error: unknown type name ‘__kernel_loff_t’
 typedef __kernel_loff_t  loff_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:55:9: error: unknown type name ‘__kernel_size_t’
 typedef __kernel_size_t  size_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:60:9: error: unknown type name ‘__kernel_ssize_t’
 typedef __kernel_ssize_t ssize_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:65:9: error: unknown type name ‘__kernel_ptrdiff_t’
 typedef __kernel_ptrdiff_t ptrdiff_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:70:9: error: unknown type name ‘__kernel_time_t’
 typedef __kernel_time_t  time_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:75:9: error: unknown type name ‘__kernel_clock_t’
 typedef __kernel_clock_t clock_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:80:9: error: unknown type name ‘__kernel_caddr_t’
 typedef __kernel_caddr_t caddr_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:199:2: error: unknown type name ‘__kernel_daddr_t’
  __kernel_daddr_t f_tfree;
  ^~~~~~~~~~~~~~~~
./include/linux/types.h:200:2: error: unknown type name ‘__kernel_ino_t’
  __kernel_ino_t  f_tinode;
  ^~~~~~~~~~~~~~
In file included from ./include/linux/list.h:9:0,
                 from ./include/linux/module.h:9,
                 from /home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.c:19:
./include/linux/kernel.h:6:10: fatal error: stdarg.h: Aucun fichier ou dossier de ce type
 #include <stdarg.h>
          ^~~~~~~~~~
compilation terminated.
scripts/Makefile.build:339: recipe for target '/home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.o' failed
make[2]: *** [/home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.o] Error 1
Makefile:1551: recipe for target '_module_/home/yves/Téléchargements/gslx680-acpi' failed
make[1]: *** [_module_/home/yves/Téléchargements/gslx680-acpi] Error 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-43-generic »
Makefile:18: recipe for target 'modules' failed
make: *** [modules] Error 2
```

Yes21, 2019-01-14:
> See https://bugs.launchpad.net/ubuntu/+bug/830550 for a similar issue and workaround, but I really think this should not be necessary.

Thanks for the link. I did this :
```
$ cd /lib/modules/$(uname -r)/build/include
$ sudo ln -s $(gcc -print-file-name=include)/stdarg.h
```
And the result was :
```
# make
make -C /lib/modules/4.15.0-43-generic/build ARCH=x86_64 CROSS_COMPILE= M=/home/yves/Téléchargements/gslx680-acpi modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.15.0-43-generic »
  CC [M]  /home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.o
In file included from ./include/linux/list.h:5:0,
                 from ./include/linux/module.h:9,
                 from /home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.c:19:
./include/linux/types.h:17:9: error: unknown type name ‘__kernel_ino_t’
 typedef __kernel_ino_t  ino_t;
         ^~~~~~~~~~~~~~
./include/linux/types.h:18:9: error: unknown type name ‘__kernel_mode_t’
 typedef __kernel_mode_t  mode_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:21:9: error: unknown type name ‘__kernel_off_t’
 typedef __kernel_off_t  off_t;
         ^~~~~~~~~~~~~~
./include/linux/types.h:22:9: error: unknown type name ‘__kernel_pid_t’
 typedef __kernel_pid_t  pid_t;
         ^~~~~~~~~~~~~~
./include/linux/types.h:23:9: error: unknown type name ‘__kernel_daddr_t’
 typedef __kernel_daddr_t daddr_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:25:9: error: unknown type name ‘__kernel_suseconds_t’
 typedef __kernel_suseconds_t suseconds_t;
         ^~~~~~~~~~~~~~~~~~~~
./include/linux/types.h:26:9: error: unknown type name ‘__kernel_timer_t’
 typedef __kernel_timer_t timer_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:27:9: error: unknown type name ‘__kernel_clockid_t’
 typedef __kernel_clockid_t clockid_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:32:9: error: unknown type name ‘__kernel_uid32_t’
 typedef __kernel_uid32_t uid_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:33:9: error: unknown type name ‘__kernel_gid32_t’
 typedef __kernel_gid32_t gid_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:34:9: error: unknown type name ‘__kernel_uid16_t’
 typedef __kernel_uid16_t        uid16_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:35:9: error: unknown type name ‘__kernel_gid16_t’
 typedef __kernel_gid16_t        gid16_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:41:9: error: unknown type name ‘__kernel_old_uid_t’
 typedef __kernel_old_uid_t old_uid_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:42:9: error: unknown type name ‘__kernel_old_gid_t’
 typedef __kernel_old_gid_t old_gid_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:46:9: error: unknown type name ‘__kernel_loff_t’
 typedef __kernel_loff_t  loff_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:55:9: error: unknown type name ‘__kernel_size_t’
 typedef __kernel_size_t  size_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:60:9: error: unknown type name ‘__kernel_ssize_t’
 typedef __kernel_ssize_t ssize_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:65:9: error: unknown type name ‘__kernel_ptrdiff_t’
 typedef __kernel_ptrdiff_t ptrdiff_t;
         ^~~~~~~~~~~~~~~~~~
./include/linux/types.h:70:9: error: unknown type name ‘__kernel_time_t’
 typedef __kernel_time_t  time_t;
         ^~~~~~~~~~~~~~~
./include/linux/types.h:75:9: error: unknown type name ‘__kernel_clock_t’
 typedef __kernel_clock_t clock_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:80:9: error: unknown type name ‘__kernel_caddr_t’
 typedef __kernel_caddr_t caddr_t;
         ^~~~~~~~~~~~~~~~
./include/linux/types.h:199:2: error: unknown type name ‘__kernel_daddr_t’
  __kernel_daddr_t f_tfree;
  ^~~~~~~~~~~~~~~~
./include/linux/types.h:200:2: error: unknown type name ‘__kernel_ino_t’
  __kernel_ino_t  f_tinode;
  ^~~~~~~~~~~~~~
In file included from ./include/linux/kernel.h:7:0,
                 from ./include/linux/list.h:9,
                 from ./include/linux/module.h:9,
                 from /home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.c:19:
./include/linux/linkage.h:8:10: fatal error: asm/linkage.h: Aucun fichier ou dossier de ce type
 #include <asm/linkage.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
scripts/Makefile.build:339: recipe for target '/home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.o' failed
make[2]: *** [/home/yves/Téléchargements/gslx680-acpi/gslx680_ts_acpi.o] Error 1
Makefile:1551: recipe for target '_module_/home/yves/Téléchargements/gslx680-acpi' failed
make[1]: *** [_module_/home/yves/Téléchargements/gslx680-acpi] Error 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.15.0-43-generic »
Makefile:18: recipe for target 'modules' failed
make: *** [modules] Error 2
```
Now it can't find `linkage.h` !!

Yes21, 2019-01-14:
Perhaps a solution here ?
https://stackoverflow.com/questions/48812053/gcc-kernel-module-compilation-errors

Yes21, 2019-01-14:
> Then I will try to modify the ` drivers/platform/x86/silead_dmi.c` file.

I can't find this file !
Where is the `drivers/platform/x86/` directory ?

onitake, 2019-01-14:
`drivers/platform/x86/` is inside the kernel sources.
However, they changed its name to `touchscreen_dmi.c` in kernel 4.18 I think.

Note that you need the *full* kernel sources to compile the kernel, not just `linux-headers`.

About the compilation errors you get... It seems like you're either compiling with the wrong gcc or it's not picking up the kernel headers correctly. The Makefile has been tested with many different kernel versions since 4.0 (and probably even 3.6), so I don't think there's anything wrong with it.

Yes21, 2019-01-14:
> `drivers/platform/x86/` is inside the kernel sources.
> However, they changed its name to `touchscreen_dmi.c` in kernel 4.18 I think.
> 
> Note that you need the _full_ kernel sources to compile the kernel, not just `linux-headers`.

I've now installed full `linux-source`.
Can you help me to locate `drivers/platform/x86/` from `/`

Yes21, 2019-01-14:
> About the compilation errors you get... It seems like you're either compiling with the wrong gcc

I've installed it with `apt-get install gcc`. What could be wrong ?

> or it's not picking up the kernel headers correctly. 

Perhaps, but why ?

onitake, 2019-01-14:
> I've installed it with `apt-get install gcc`. What could be wrong ?
On Debian at least, that's not the right compiler. You need `linux-kbuild`. But I have no idea how this is handled on Ubuntu. Sorry.
> Perhaps, but why ?
Normally, you don't have to specify any kernel version or source path. It will be picked up from the running kernel automatically.

> I've now installed full `linux-source`.
> Can you help me to locate `drivers/platform/x86/` from `/`
The linux-source package installs the compressed kernel source to `/usr/src`. You should unpack it somewhere in your home.
But there is a better method: https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel
This will give you a nice .deb, which is easier to upgrade/uninstall later.

Yes21, 2019-01-15:
> The linux-source package installs the compressed kernel source to `/usr/src`. You should unpack it somewhere in your home.
> But there is a better method: https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel
> This will give you a nice .deb, which is easier to upgrade/uninstall later.

Thanks for the link. I will try it ...

Yes21, 2019-01-21:
@StrongTabletMan Thanks for the information.
I will try with Xubuntu and kernel 5.0-rc2.
Did you succeed in calibrating the touchscreen ?
 It looks not so difficult with xinput_calibrator ...

StrongTabletMan, 2019-01-21:
@Yes21 Based on the reactions I got from the touchscreen, I think I either extracted the firmware in a wrong way from the *.rar archive provided here or the tool had problems extracting it. Maybe it's also just that the proper configuration settings that would usually be in the kernel files are missing.

At the moment, I don't really know how to continue working on this, I lack the knowledge.

I might try to restore the factory system with Android and Windows 10 and copy over all the driver I find there that are directly on my device and then maybe try each of the firmwares the extraction tools from this repository might create for me.

StrongTabletMan, 2019-01-21:
**Short update:** 
Like I assumed, the messed up touchscreen registration was due to extracting the wrong firmware file.

Using `scanwindrv` on the file _SileadTouch.sys_ from [this](http://www.mediafire.com/file/ven2gq50mx2wxem/HI8+Air-driver.rar) archive posted on the Chuwi forums and in here extracts a corrupt firmware file. The correct way is to use `untscfg` on _T892_64b_3676_L_DZ_OGS80B32_171215_Lq.h_ in the same folder. This will produce a slightly bigger firmware file that can be renamed to _mssl1680.fw_ and placed into `/lib/firmware/silead` for testing purposes. The touchscreen will not be calibrated, but the touch should register properly.

By the way: while the slight jittering of the mouse pointer does not happen in Windows 10, but it does happen in the touch BIOS of the Chuwi Hi8 Air as well as in Linux, so I think this could be solved by adjusting certain software parameters. I think @onitake already mentioned this in another issue.

I'll now try to find the proper values to calibrate the touchscreen. 

If anyone else wants to try this, I found a post of Hans de Goede on how to find the coordinates: 
https://bugzilla.redhat.com/show_bug.cgi?id=1526312#c3

I uploaded the extracted firmware I just tested:
https://www.mediafire.com/file/es7nuehhdqkylk2/mssl1680.fw/file or
https://github.com/onitake/gsl-firmware/files/2780463/mssl1680.zip

StrongTabletMan, 2019-01-21:
The maximum coordinates I received via `evemu-record` seem to match the ones for the Chuwi Hi8 Pro in https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/touchscreen_dmi.c?h=v5.0-rc2#n44

I created a picture to show the values in the corners: 
https://i.imgur.com/EegRqJu.png

(The actual lowest value for x and y that showed up for my tests was actually 1 for both, not 0.)

onitake, 2019-01-22:
The scanwindrv script only applies a simple heuristic to find firmware in the Windows driver. It's should be able to extract a complete firmware image, but it won't be able to separate them if they are side-by-side. Also, if the vendor has added a separate firmware, the Windows driver will use that instead.

Glad that you could find one that works!
Could you prepare a pull request so I can add it to the repo?

Concerning jitter correction: The Linux input driver framework supports many additional options that you can pass to the driver via DMI or DeviceTree. See https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/input/touchscreen/touchscreen.txt for an overview.

StrongTabletMan, 2019-01-22:
I added entries for the tablet and successfully compiled a kernel, but silead_ts still asks for a mssl1680.fw file. I don't know what to do really, I did everything as described.

StrongTabletMan, 2019-01-23:
I want to report back that the whole touchscreen suddenly started working after a reboot. I have the feeling that my additions worked and maybe something blocked the driver loading the proper firmware. I'll try to investigate this further, but so far it looks promising.

Yes21, 2019-02-02:
@StrongTabletMan 
Hi, and thanks for your help.

I have now Xubuntu 18.10 on my Hi8 Air.
I've downloaded this file https://www.mediafire.com/file/es7nuehhdqkylk2/mssl1680.fw/file, and did copy it into /lib/firmware/silead.
And now what are the next steps ?

> Kernel I modified and compiled successfully: https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.0-rc2/

What should be modified before compiling this kernel ?

> My changes to touchscreen_dmi.c

What to do with this file ?

Thanks for your help.

Yes21, 2019-02-02:
I've another question :

Do one knows how to decrease the resolution of the screen from 1920x1200 to 1280x800. It works with windows, but not with linux.

Sorry to post this question here ...

Yes21, 2019-02-03:
I answer to me !
I could install the v5.0-rc2 kernel.
After that the touchscreen is working, but calibration is totally bad (what I was waiting for !).
I tried to use xinput_calibrator to do the calibration, bur it did nothing.

I guess I have to copy `touchscreen_dmi.c` somewhere in the linux-headers, and then compile a new kernel ?
I will try it ...

Yes21, 2019-02-03:
> The touchscreen will respond with the firmware being placed into the corresponding folder, but there is currently no [configuration](https://gist.github.com/StrongTabletMan/6977a5f9893ce860020875a42c297c30/revisions) for the Chuwi Hi8 Air in `touchscreen_dmi.c`.
> 
> You have to add those blocks to your `touchscreen_dmi.c` file and compile the whole kernel after that.

Ok, Thanks. I will do it.

Did you change something in menuconfig before compiling ?

Yes21, 2019-02-04:
Here are all the steps I did to compile the v5.0-rc2 kernel on the Chuwi Hi8 Air :

`git clone https://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack v5.0-rc2`
cd v5.0-rc2/
wget https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.0-rc2/0001-base-packaging.patch
patch -p0 < 0001-base-packaging.patch

Yes21, 2019-02-04:
> I only downloaded the kernel sources, applied the Ubuntu patches and then followed the instructions here: https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel#Building_the_kernel
> 
> For me it threw out a few errors during the initial compilation steps. If it does the same for you, it's because some packages are missing that are needed to successfully compile the kernel. I can't remember which ones exactly, but you should get error messages telling you what to search for.

@StrongTabletMan 
I'm following the instructions from the link you did send.
I see we have to add lines to "sources.list"

> The above command requires your system to have the correct deb-src lines in /etc/apt/sources.list. For example, on Xenial you should have:
    deb-src http://archive.ubuntu.com/ubuntu xenial main
    deb-src http://archive.ubuntu.com/ubuntu xenial-updates main

If I'm trying with "cosmic" instead of "xenial", it does not work.
Which lines did you add in your "sources.list" ??

Yes21, 2019-02-05:
I couldn't go to the end of the compiling process after 5 hours !!
My 25 Gb "/" partition is now full !!
It's really the HELL to compile the kernel on a small tablet ...
I don't know what to try now ?

Any help is welcome.

StrongTabletMan, 2019-02-06:
Don't compile the custom kernel on the Tablet itself, that's suicide. The folder grows from around 2.9 GB to 23.1 GB and the tablet's hardware is slow enough to make the process take many hours.

Take the most powerful system you own, set up VirtualBox (or any other solution you prefer), assign it as many resources as possible (I gave mine 4 GB RAM, all 4 cores on my quad-core CPU and set the disk to 100 GB with dynamic expansion), download Xubuntu 18.10 64-bit and install it, then start it and edit your sources.list ([mine](https://gist.github.com/StrongTabletMan/05db3926339e9259f4971a306e41cfed) (DE mirrors since I'm from Germany)).

Then get the sources from the git repository and follow the mentioned steps. [This](https://gist.github.com/StrongTabletMan/acff269d9c94b0d9587f395f1c26b4b0) is roughly what I did.

StrongTabletMan, 2019-02-06:
Like some of the other contributors with pull requests here, I also submitted a [patch](https://patchwork.kernel.org/patch/10794785/) to add the Chuwi Hi8 Air data for the silead driver into the Linux kernel, so hopefully there won't be the necessity to compile a custom kernel in many situations someday.

onitake, 2019-02-07:
To be honest, I wouldn't do this in a VM (and particularly not VBox...)
If you really don't want to pollute your host box, just launch a container with a base Debian or Ubuntu system and mount a host path so you get better I/O performance.
Make sure the container image matches what you installed on the tablet, so you get the right compilers etc.

FWIW, I did compile a kernel on a Baytrail tablet once. Just kept it running overnight. But I moved to a faster machine after the first time...

Yes21, 2019-02-08:
@StrongTabletMan 
Many many Thanks for your help !! That's very nice to you ...

So I did create a vm with Xubuntu 18.10-64.
Then git clone the v5.0-rc2 sources and apply patches.
Then modify my sources.list and the touchscreen_dmi.c file.

But I can't go further :

StrongTabletMan, 2019-02-09:
My first thought would be that you didn't apply the Ubuntu patches correctly, because you seem to be missing files and folders that are created by those patches.

Make sure you properly download all the patches, put them into the main folder (the one with `arch`, `block`, `certs`, `crypto`, etc. folders) and then apply them with `patch -p1 <0001-base-packaging.patch`, then `patch -p1 <0002-UBUNTU-SAUCE-add-vmlinux.strip-to-BOOT_TARGETS1-on-p.patch`, then the next patch and so on.

After applying the first patch, you should already see the `debian` folder and other files and folders there. If not, something probably went wrong.

You can also just download the patches with a browser from https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.0-rc2/ etc.

I'm probably as new to compiling a kernel as you are, but I hope it works for you!

Yes21, 2019-02-10:
@StrongTabletMan Thanks again for your help.
I've finally succeed in compiling the kernel. My error was that I used the patch command with -p0 option instead of -p1.
Then I've copied and installed the deb files on the tablet.
But after many reboots it didn't change anything and the touchscreen is still not calibrated.

Perhaps should I rename the firmware `gsl3676-chuwi-hi8-air.fw` instead of `mssl1680.fw` ?

Yes21, 2019-02-10:
> Perhaps should I rename the firmware `gsl3676-chuwi-hi8-air.fw` instead of `mssl1680.fw` ?

No it doesn't change nothing ...

StrongTabletMan, 2019-02-10:
Make sure that 
- the additions are in your `touchscreen_dmi.c` file,
- you installed all five *.deb packages on the tablet and
- you booted with the new kernel you installed (`uname -a` should show this).

Yes21, 2019-02-14:
> Perhaps should I rename the firmware `gsl3676-chuwi-hi8-air.fw` instead of `mssl1680.fw` ?

No it doesn't change nothing ...

> Make sure that
> 
>     * the additions are in your `touchscreen_dmi.c` file,
> 
>     * you installed all five *.deb packages on the tablet and
> 
>     * you booted with the new kernel you installed (`uname -a` should show this).

I did double check all these things !
I will try to redo all from zero ...

Please, could you share you five .deb files ?

StrongTabletMan, 2019-03-24:
The changes are now in the current mainline kernel v5.1-rc1: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/touchscreen_dmi.c?h=v5.1-rc1#n44

The firmware file that goes into `/lib/firmware/silead/` can be found here: https://github.com/onitake/gsl-firmware/blob/b0492c1137d91b8025b3b3227786704d624e9da4/firmware/linux/silead/gsl3676-chuwi-hi8-air.fw

I updated to that kernel version on Xubuntu 18.10 and it works, but the jitter makes it very hard to double-click and therefore navigate properly.

Kellwaxx11, 2024-02-11:
Hi @StrongTabletMan 
I know its a bit long time since the your last comment, but I hope you can help
I have booted Ubuntu 22.04 on Chuwi Hi8 Air
but am struggling to get the touch screen to work
I downloaded this file
https://github.com/onitake/gsl-firmware/blob/master/firmware/linux/silead/gsl3676-chuwi-hi8-air.fw
renames it to mssl1680.fw and copied to /lib/firmware/silead

now the touch registers but it is a total mess
seems to be rotated 90 degrees, and kind of scaled down to quarter of the actual screen 

surprisingly, adding this file got the accelerometer to work, now Ubuntu rotates.

any advice please what to do to get the touch to work?

also on another topic maybe you have experience in, anyway to get the OTG & Charging to work
i havea hub with 3 USB ports + 1 charging port ,that works perfectly on Windows (using the hub and charging simultaneously), but not on ubuntu.

It worked once on ubuntu randomly, but stopped after i was trying to fix the touch and rebooted to many times, and I couldnt get it to work anymore,even with reinstalling Ubuntu.
could it be any boot parameters to be added to grub to make it work.

appreciate any advice.

jwrdegoede, 2024-02-11:
> I know its a bit long time since your last comment, but I hope you can help
> I have booted Ubuntu 22.04 on Chuwi Hi8 Air
> but am struggling to get the touch screen to work
> I downloaded this file
> https://github.com/onitake/gsl-firmware/blob/master/firmware/linux/silead/gsl3676-chuwi-hi8-air.fw
> renames it to mssl1680.fw and copied to /lib/firmware/silead

You should not need to rename it. The kernel should actually try to load /lib/firmware/silead/gsl3676-chuwi-hi8-air.fw and then also automatically apply the right size and axis swapping.

Please run the following on the tablet as normal user: `grep . /sys/class/dmi/id/* 2> /dev/null` and then copy and paste the output here. Maybe your Hi8 Air has slightly different BIOS DMI strings causing the kernel to not recognize it.

Which model do you exactly have ? The current kernel config is for the CWI543 you can find the CWI5xx number on the back of the tablet.

Another option which may cause this to not work is if Ubuntu has not enabled the touchscreen_dmi driver in the kernel. To check this I need you to retreive the Ubuntu kernel config from the running system. To do this see: https://www.baeldung.com/linux/kernel-config

Also note this all assumes that you are using the silead touchscreen driver which comes with the mainline kernel and not building a driver from somewhere else.

Kellwaxx11, 2024-02-12:
> You should not need to rename it. The kernel should actually try to load /lib/firmware/silead/gsl3676-chuwi-hi8-air.fw and then also automatically apply the right size and axis swapping.
>
> Please run [...]

Thank you for your reply
My device model is : CWI543
the ouptut of the command `grep . /sys/class/dmi/id/* 2> /dev/null` is:
```
/sys/class/dmi/id/bios_date:01/16/2018
/sys/class/dmi/id/bios_release:5.11
/sys/class/dmi/id/bios_vendor:American Megatrends Inc.
/sys/class/dmi/id/bios_version:CHUWEI.DT892P.bpBJREA02
/sys/class/dmi/id/board_asset_tag:Default string
/sys/class/dmi/id/board_name:Cherry Trail CR
/sys/class/dmi/id/board_vendor:Default string
/sys/class/dmi/id/board_version:Default string
/sys/class/dmi/id/chassis_asset_tag:Default string
/sys/class/dmi/id/chassis_type:30
/sys/class/dmi/id/chassis_vendor:Default string
/sys/class/dmi/id/chassis_version:Default string
/sys/class/dmi/id/modalias:dmi:bvnAmericanMegatrendsInc.:bvrCHUWEI.DT892P.bpBJREA02:bd01/16/2018:br5.11:svnCHUWIInnovat
onAndTechnology(ShenZhen)co.,Ltd:pnHi8Air:pvrDefaultstring:rvnDefaultstring:rnCherryTrailCR:rvrDefaultstring:cvnDefault
tring:ct30:cvrDefaultstring:skuDefaultstring:
/sys/class/dmi/id/product_family:Default string
/sys/class/dmi/id/product_name:Hi8 Air
/sys/class/dmi/id/product_sku:Default string
/sys/class/dmi/id/product_version:Default string
/sys/class/dmi/id/sys_vendor:CHUWI Innovation And Technology(ShenZhen)co.,Ltd
/sys/class/dmi/id/uevent:MODALIAS=dmi:bvnAmericanMegatrendsInc.:bvrCHUWEI.DT892P.bpBJREA02:bd01/16/2018:br5.11:svnCHUWI
nnovationAndTechnology(ShenZhen)co.,Ltd:pnHi8Air:pvrDefaultstring:rvnDefaultstring:rnCherryTrailCR:rvrDefaultstring:cvn
efaultstring:ct30:cvrDefaultstring:skuDefaultstring:
```

if i put the file "gsl3676-chuwi-hi8-air.fw" as it is under "/lib/firmware/silead", nothing happens and I still get this msg while booting
`silead_ts i2c-MSSL1680:01: Firmware request error -2` which is why i renamed the file initially.

and attached is the kernel config
[kernel-config-ubuntu20.04.config.txt](https://github.com/onitake/gsl-firmware/files/14236859/kernel-config-ubuntu20.04.config.txt)

many thanks again for helping

Kellwaxx11, 2024-02-12:
> @Kellwaxx11 thank you for gathering all the necessary info to debug this.
> 
> So the DMI strings match, but the problem is that your touchscreen has an ACPI name of `MSSL1680:01` where as the kernel's current match for the Chuwi Hi8 Air expects `MSSL1680:00`.
> 
> I have prepared a kernel-patch to fix this: https://lore.kernel.org/platform-driver-x86/20240212120608.30469-1-hdegoede@redhat.com/
> 
> (click on the "raw" link to download it)
> 
> To test this you will need to do a build of the Ubuntu kernel with the patch added, see Ubuntu's docs. I assume the process for this will generate 1 or more .deb files for you to install. You will want to run this process on a much faster machine and then copy the resulting debs to the tablet.
> 
> Alternatively you could install Fedora and then I can provide a pre-build kernel with the patch for you to test. Note if you install I advice to use the Fedora 39 x86_64 workstation livecd dd-ed to an USB stick, you may run into this issue while trying to boot this though:
> 
> https://discussion.fedoraproject.org/t/install-media-dont-boot-in-uefi-mode-on-certain-motherboards/71376
> 
> If you hit that issue I advice workaround C. You don't want to install F36 first and then upgrade on this slow hw (workaround B does not work on this hw).

wow, thanks a lot

I will try to patch and build Ubuntu kernel, i have never done it before so i doubt my success.

If you could provide Fedora kernel as well please, I never tried Fedora before, so why not give it a try.

later after i get the touch to work hopefully, I hope you dont mind me asking your help regarding the OTG and Charge issue
I have this exact hub and there is a patch in this link
https://linux-sunxi.org/USB_OTG_Charging_Hub
because getting it to work is crucial for the purpose I want linux on this tablet, to install Octoprint and connect to my 3d printer

closed 2024-02-12 by jwrdegoede