onitake/gsl-firmware #96: Driver for ezPad 6 plus linux

jeremyle-planorama, 2018-11-07:
I am looking for a driver for the touchscreen ezpad 6 plus (JUMPER). Anybody has it ?

onitake, 2018-11-07:
Do you have Windows or Android drivers for it?
If yes, we should at least be able to extract the firmware.

jeremyle-planorama, 2018-11-08:
Trying to find one ... Do you believe ezpad 6 plus is just an extension of the ezpad 6? meaning the firmware should be the same?

onitake, 2018-11-08:
There's only one way to find out, you know. :smirk:
 
If you can get it to work with the same fw, you should definitely submit a patch or a request for a patch to [linux-hid](http://vger.kernel.org/vger-lists.html#linux-input).

jeremyle-planorama, 2018-11-14:
Will try on my ezpad 6 plus tonight with firmware of ezpad 6. If it works, I will tell you :)

jeremyle-planorama, 2018-11-17:
@onitake re
I have all driver for windows. can you help me extract the firmware from it to get driver for linux?
driver : https://drive.google.com/file/d/17-wBDVz8Hskf1I-Ccae5HHq6s6BW5N3k/view