onitake/gsl-firmware #103: Firmware request for EastRising Display: TFTM050A2-3-3661

PetervdNieu, 2019-01-11:
Hi,

Right now I'm using the ER-TFTM050A2-3-3661 Display:
[https://www.buydisplay.com/default/5-inch-tft-lcd-display-capacitive-touchscreen-ra8875-controller-800x480](url) 

The display is used on the Intel Cycone V dev. board with the Linux Silead.ko driver. I tried to use the firmware from this repository:
[https://github.com/onitake/gsl-firmware/tree/master/firmware/eastrising/er-tpc050-1](url)
but none is working.

So, I tried the firmware from this repository:
[https://github.com/onitake/gsl-firmware/tree/master/firmware/linux](url)
and I found some firmware that was working with my display.

The one working best is the **gsl1680-polaroid-mid2407pxe03.fw** firmware, but the y-axis is flipped and the resolution is something like: 850x640.

Is it possible to get firmware for the ER-TFTM050A2-3-3661 Display that is working with the Linux Silead.ko driver?

Many Thanks!

Peter

onitake, 2019-01-11:
If this firmware is close enough to your device, you should be able to add support by adding a parameter section to silead_dmi.c (or touchscreen_dmi.c in newer kernels), as per https://github.com/onitake/gsl-firmware#silead_ts .

Note that this requires you to recompile the kernel, because the device matching code cannot be built as a kernel module.

PetervdNieu, 2019-01-11:
Thanks for your fast answer!

I'm using a custom Linux for ARM processors which doesn't have these files.

I did add some parameters to the device tree:
[image]

This flipped the y-as for me but the screen resolution is worse now...

I added a printk to the Silead driver to see the x- and y-coordinates and this shows that the x-axis is still something like 850 and the y-axis now goes from -150 to 480.

It looks like the `touchscreen-size-x` and `touchscreen-size-y` don't effect the touchscreen.

onitake, 2019-01-11:
Looks like you're ahead of me on this - I haven't tried Silead on ARM with DT tuning myself.
I assume you know of https://github.com/torvalds/linux/blob/master/Documentation/devicetree/bindings/input/touchscreen/silead_gsl1680.txt then.

If you can't get a proper response window on the touchscreen, it's probably because you have unsuitable firmware. The firmware image contains calibration data for each device, and that obviously also specifies the resolution. The value you pass to the driver is only applied to the kernel's representation of the touchscreen and not sent to the controller.

PetervdNieu, 2019-01-15:
I'm stilling using the same firmware, but now I scaled the x- and y-axes in my software.
It is not perfect, but it is accurate enough for now.

Thanks

closed 2019-01-15 by PetervdNieu