onitake/gsl-firmware #105: Wrong diver loaded silead_ts jumper ezpad 6 (mssl1680.fw)

BenjaminRenz, 2019-01-14:
I'm trying to get the driver to load my firmware for the b-version of the ezpad 6 pro, but it always defaults to mssl1680.fw. 
I already tried:
modifying the sourced file in the linux kernel, compiling the module and manually copying it to 
/lib/modules/4.18.0-13-generic/kernel/drivers/input/touchscreen/silead.ko
it also is using this driver, (modprobe -v shows this location) but it still loads the default firmware nevertheless. 

`[  266.164048] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000`
`[  266.305777] silead_ts i2c-MSSL1680:00: Direct firmware load for silead/mssl1680.fw failed with error -2`

 I get partial touch when renaming my firmware to the default name.

I have added the following to the silead_dmi.c file:
`{
                /* Jumper EZpad 6 Pro b*/
                .driver_data = (void *)&jumper_ezpad_6_pro_b_data,
                .matches = {
                        DMI_MATCH(DMI_PRODUCT_NAME, "EZpad"),
                        DMI_MATCH(DMI_BIOS_VERSION, "5.12"),
                        /* Above matches are too generic, add bios-date match */
                        DMI_MATCH(DMI_BIOS_DATE, "04/24/2018"),
                },
        },
`

dmidecode returns the following
Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
        Vendor: Jumper.A116
        Version: 5.12
        Release Date: 04/24/2018
        Address: 0xF0000
        Runtime Size: 64 kB
        ROM Size: 5120 kB
        Characteristics:
                PCI is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                BIOS ROM is socketed
                EDD is supported
                5.25"/1.2 MB floppy services are supported (int 13h)
                3.5"/720 kB floppy services are supported (int 13h)
                3.5"/2.88 MB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 5.12
        Firmware Revision: 176.6

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Jumper
        Product Name: EZpad
        Version:  
        Serial Number:  
        UUID: 03000200-0400-0500-0006-000700080009
        Wake-up Type: Power Switch
        SKU Number: 0001
        Family: WS

Maybe @jwrdegoede could give me a hint on why the dmi match does not work?
Thanks for your help.

ChalesYu, 2019-01-15:
maybe you filled the wrong DMI information

`DMI_SYS_VENDOR` is from `cat /sys/class/dmi/id/sys_vendor` command.
`DMI_PRODUCT_NAME` is from `cat /sys/class/dmi/id/product_name` command.
`DMI_BIOS_VERSION` is from `cat /sys/class/dmi/id/bios_version` command.
`DMI_BIOS_DATE` is from `cat /sys/class/dmi/id/bios_date` command.

BenjaminRenz, 2019-01-15:
Those are
`DMI_SYS_VENDOR       = Jumper`
`DMI_PRODUCT_NAME = EZpad`
`DMI_BIOS_VERSION     = 5.12`
`DMI_BIOS_DATE           = 04/24/2018`
, so they seem to match what I have set.
This time I also recompiled the kernel 4.20.2 and made sure that it uses the correct module, but still want's to load silead/mssl1680.fw

ChalesYu, 2019-01-15:
check your kernel config file ,make sure it have `CONFIG_TOUCHSCREEN_DMI=y`
if it still want `silead/mssl1680.fw`, I don't know why .

BenjaminRenz, 2019-01-16:
> 
> 
> check your kernel config file ,make sure it have `CONFIG_TOUCHSCREEN_DMI=y`
> if it still want `silead/mssl1680.fw`, I don't know why .

Thanks, that solved the issue, I thought this option would be enabled by default, but it wasn't.
I would guess I will post an update for the readme here because the process of compiling the kernel was quite unintuitive.

onitake, 2019-01-17:
@BenjaminRenz Thank you, that is most appreciated.

closed 2019-01-18 by BenjaminRenz