onitake/gsl-firmware #114: Problem with touchscreen calibration from onda v891w

rlsl, 2019-03-06:
I installed Debian Gnu/Linux Buster, that will go stable in near time, and installer the gslx680_ts_acpi and copy the firmware this repository "silead_ts.fw" for onda v891w but the calibration no way function, the touchscreen don't touch in the same place that I touch, only in some places on the screen. Can you help me?

onitake, 2019-03-06:
I stopped supporting gslx680_ts_acpi a long time ago.
Please switch to the in-kernel silead driver if you can.

Looking at https://github.com/onitake/gsl-firmware/tree/master/firmware/linux/silead , there's support for a "v891w v1" device - not sure if this is the same as yours. If it is, recent kernels should support it out of the box once you copy the firmware to the right place.

Otherwise, you will need different firmware anyway. Look at the README and other issues in the repo for instructions for more info.

rlsl, 2019-03-06:
Ok. Thank You

StrongTabletMan, 2019-03-24:
@onitake 

Last time I checked, there was no silead driver built in the kernel currently shipped with Debian Buster.

rlsl, 2019-03-24:
I know but chance the touch BIOS configuration of silead to goodix, but no way, he show this error in the Debian Buster 64bits.

- Command: dmesg | grep i2c-GDIX
[   10.099510] Goodix-TS i2c-GDIX1001:00: i2c test failed attempt 1: -121
[   10.125938] Goodix-TS i2c-GDIX1001:00: i2c test failed attempt 2: -121
[   10.153673] Goodix-TS i2c-GDIX1001:00: I2C communication failure: -121
[   10.172955] Goodix-TS: probe of i2c-GDIX1001:00 failed with error -121

StrongTabletMan, 2019-03-24:
I think many of these devices come with some kind of 'default' template where the manufacturers simply choose the BIOS options that fit the built-in hardware. If your device uses a Silead touchscreen controller, switching options in the BIOS probably won't do anything or break things.

Since the old gslx680_ts_acpi driver at least worked in some way, I guess it's a Silead and not a Goodix touchscreen controller.

onitake, 2019-03-24:
@StrongTabletMan Yes, I know.
If you can, please file a bug report against the linux kernel package.

The maintainers should set the following config options:
```
CONFIG_TOUCHSCREEN_SILEAD=m
CONFIG_TOUCHSCREEN_DMI=y
``
so the module and the DMI overrides get built and installed.

rlsl, 2019-03-25:
I report in March 11, 2019 with the title "Error for module touchscreen Goodix", no answer, and November 12, 2018 with the title "Module silead (tooch)", no answer, all for debian-kernel@lists.debian.org. I tested ubuntu but it gets very unstable and slow, the ubuntu have the silead module in the kernel. I have 3 tablet with silead touch and it's worked with the "onitake/gslx680-acpi" and firmware "onitake/gsl-firmware" but as the kernel for this type of processor is being developed I'm almost always upgrading it, using the stable version of Debian with the backports kernel, but the last update broke this.

My tablets are: Chuwi HI8, Onda V891W and Teclast X98 PLUS II.

Obs.: Thank you onitake I see your send in the "Bug#925424: linux-image-amd64: Build the Silead touchscreen controller kernel driver" in the my e-mail from submit@bugs.debian.org

rlsl, 2019-03-25:
> I think many of these devices come with some kind of 'default' template where the manufacturers simply choose the BIOS options that fit the built-in hardware. If your device uses a Silead [...]

Thank you and I'll do it.

onitake, 2019-03-25:
I think it's important to always report a bug on the Debian bug tracker *first*, then send mail to the mailing lists for feedback. There's currently a transition to new kernel packages going on, I hope the bug will be retargeted once this is complete.

It's unfortunate that this bug wasn't considered important for Debian buster, so I guess we'll have to stick with no support by default for much longer. :disappointed: 

There's also the question about what to do with the firmware: For proper out-of-the-box support, firmware should be packaged, but without a clear redistribution license, it would never be accepted into Debian (or linux-firmware, for that matter). I'm not sure if @jwrdegoede has done any work on that front.

rlsl, 2019-08-15:
I would like to thank everyone for the tips and did as discussed here, installed Debian Buster, recompiled the kernel, I'm using version 5.2.8, with silead module active and my "teclast x98 plus II" worked perfectly, with the specific firmware of the link "https://github.com/onitake/gsl-firmware/tree/master/firmware/linux/silead", but in the "onda v891w ch" did not work, with the same kernel of the teclast and the specific firmware it from the link above, with these options file names "gsl3680-onda-v891w.fw gsl3680-onda-v891w-v1.fw mssl1680.fw" all copies from the original because silead ask me for the firmware file name "silead/mssl1680.fw".
Does anyone have any ideas to help?

Syslog Report:
kernel: [    8.700116] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
kernel: [    8.700186] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
kernel: [    8.700555] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
kernel: [   11.499157] input: silead_ts as /devices/pci0000:00/808622C1:05/i2c-5/i2c-MSSL1680:00/input/input16
kernel: [   11.499410] evbug: Connected device: input16 (silead_ts at input/ts)

[config-5.2.8.txt](https://github.com/onitake/gsl-firmware/files/3503972/config-5.2.8.txt)

antoniobaiao, 2023-03-16:
Hi,
I have the same problem. Touchscreen is not well calibrated on Onda v820w. I'm using the file present here https://github.com/onitake/gsl-firmware/tree/master/firmware/linux/silead.

I already search about how to calibrate. Every place talks about to use xinput_calibrator. But it doesn't produce a correct result.

Any Idea how can I solve this?

Thanks :-)