onitake/gsl-firmware #117: I have tried to understand what it takes to add touch to Insignia Flex 2017

CoolGames, 2019-03-31:
The Insignia Flex firmware seems to be what I have except mine is much newer.
I can't begin to understand how or why I need to recompile the kernel if
"silead_ts ships with the Linux kernel since version 4.8 and is the recommended driver."
I have the files attached zip from the Windows 10 1809 
found at /Windows/System32/DriverStore/FileRepository/sileadtouch.inf_x86_1ceb09f9397207c1/
that work in Windows 10 but not sure why the files in this repository don't.
[flex11_6_2017.zip](https://github.com/onitake/gsl-firmware/files/3027175/flex11_6_2017.zip)

`$ uname -a
Linux coolflex11 4.15.0-46-lowlatency #49-Ubuntu SMP PREEMPT Wed Feb 6 10:23:17 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

$ dmesg | grep silead
[    6.726389] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
[    6.726459] silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
[    6.731862] silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
[    6.733179] silead_ts i2c-MSSL1680:00: Registers clear error -11
[    6.857150] silead_ts: probe of i2c-MSSL1680:00 failed with error -11

`
I used the MSSL1680 to place renamed firmware in zip in /lib/firmware/silead

Was that all that was needed or do I need to build something.
This is a REAL slow CPU using the recently released ATOM 32 bit UEFI and 64 Bit Ubuntu from [Blog at linuxium](http://linuxiumcomau.blogspot.com/2019/03/ubuntu-announced-new-point-releases-for.html)
running on a 128 GB micro SD with the Boot directory on Windows root drive.
It took two years to find it done March 4,2019.
The audio needed the [HDMI blacklisted](https://forums.linuxmint.com/viewtopic.php?t=280430) in November 2018.

Please help get the touch working , too.

eanbowman, 2019-08-16:
I tried that firmware and it's not as good as the Insignia Flex firmware.fw 

At least with that I have touch registering but only in the bottom right of the screen (moving the cursor to the top left randomly).

Seems like that firmware is just missing a setting or something.

eanbowman, 2019-08-16:
I downloaded your file above and ran scanwindrv on it to produce the firmware_00.fw and put that in the firmware folder with the right name and it works! Kind-of!

It only maps to half the screen and the vertical axis is flipped but it works!
https://www.youtube.com/watch?v=erkn7wfRQAs

onitake, 2019-08-18:
You should be able to correct the flipped axis with a DMI override.
Missing input on one side indicates unsuitable firmware, however.

eanbowman, 2019-08-19:
@onitake Thanks for the reply!

I didn't find it clear how to do a DMI override but I'll read the README again I guess.

"Missing input on one side" Well, it maps to 1/4 of the screen. So maybe they used a few different models of touchscreen for the Insignia Flex 11? Hmm...

onitake, 2019-08-19:
Read this section: https://github.com/onitake/gsl-firmware#silead_ts

You don't need a different driver, but you need a firmware blob that maps exactly to your screen. You also need to pass parameters to the kernel driver that configure the touchscreen's dimensions, screen flipping and other properties.

eanbowman, 2019-08-19:
Sorry, I used the word driver when I meant firmware. Yeah no other firmware blob has produced any results but this one.

So you're saying I'm going to need to recompile my kernel on the tablet machine then? I've done that once or twice but I was hoping to avoid it. Okay.

onitake, 2019-08-19:
I'm afraid that's the only way to actually make the driver recognise the hardware properly.
Originally, there was an effort by an Intel engineer to build proper ACPI-based configuration for Silead touchscreen controllers, but by that point, it was already too late. The market was already flooded by cheap Baytrail tablets that only work right with the preinstalled Windows or Android drivers. Some vendors offer their drivers as a separate download, and that's the only way to even get the touchscreen working on any OS, or obtain firmware for Linux.

That driver doesn't look right to me. I think it's only for the I²C controller in the SoC, not the touchscreen.

If you've already lost the original Windows installation, you might have some luck getting the drivers or a Windows image from the store where you bought the tablet.

Perhaps identifying the exact model might help, have you already run `dmidecode` to get the system identifiers?

eanbowman, 2019-08-19:
Like I said, it's the same tablet as the OP has. NS-P11W6100. I don't have it in front of me. Maybe W7100? Yeah the seller wiped it somehow so I'm not sure if they have an install or if they simply told Windows to revert to factory defaults. It was an eBay purchase in a generic box.

eanbowman, 2019-08-20:
[dmidecode.txt](https://github.com/onitake/gsl-firmware/files/3518213/dmidecode.txt)

onitake, 2019-08-20:
This is what you should match:
> System Information
> 	Manufacturer: Insignia
>	Product Name: NS-P11W7100
and possibly:
> 	SKU Number: 5255800

With the `firmware.fw` in https://github.com/onitake/gsl-firmware/tree/master/firmware/insignia/flex11 , can you create the necessary DMI override for the kernel and see if it helps?
Perhaps you simply need the right parameters to make it work.

And about the Windows driver: It's highly unlikely (but not impossible) that the manufacturer has uploaded the driver for this particular device to the Windows driver store. At least, that wasn't the case for _any_ of the tablets I've seen so far. If you remove the preinstalled OS and the vendor doesn't offer a complete driver package on their website, you're out of luck. Maybe someone else with the same tablet can help.

eanbowman, 2019-08-21:
Do I have to re-compile the kernel or... ?

I checked how to do that for Ubuntu (running Xubuntu) and it says I might not need to but only change the headers. But the file in the instructions on the README.md does not exist in the headers.

I guess it's an unsupported situation if I recompile the kernel in Ubuntu but that's fine for just this tablet.

I'm just not sure what to do, really. It's been a while since I did it. Mostly with Gentoo.

onitake, 2019-08-21:
I'm afraid that you do have to recompile the kernel.

The part that does driver pre-initialisation is not part of any kernel modules or easily replaceable code. You really have to recompile the kernel and reload it (via rebooting or kexec), so the hardware gets re-matched and all drivers get loaded with the correct parameters.

Once you get your touchscreen to work, you can submit your working firmware to this repository (into firmware/linux/silead) and and the kernel patch to the linux-input mailing list. They will review it and merge it into the mainline Linux kernel, which means that other users of the same tablet will automatically get support for it at some point.

CoolGames, 2019-08-22:
Latest upgrade to Ubuntu 19.04 kernel and take these steps

`Linux xxxxxxx 5.0.0-23-generic #24-Ubuntu SMP Mon Jul 29 15:36:44 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
`

`sudo cp /lib/firmware/silead/silead_ts.fw /lib/firmware/silead/mssl1680.fw
`

Check for activation

```
dmesg | grep silead
[    9.081715] input: silead_ts as /devices/pci0000:00/808622C1:01/i2c-1/i2c-MSSL1680:00/input/input10
```

Trial and error calibrate with [xinput_calibrator
](http://linux-sunxi.org/Touchscreen)

yields something like
```
# xinput set-prop 16 305 1.0, 0.0, 0.0,    0.0, -1.0, 1.0,    0.0, 0.0, 1.0
xinput set-prop "silead_ts" "libinput Calibration Matrix" 2.2, 0.0, -0.04,    0.0, -2.7, 1.0,    0.0, 0.0, 1.0

```

eanbowman, 2019-08-22:
@CoolGames 
```
lilithe@signet:~$ uname -sr
Linux 5.0.0-25-generic
lilithe@signet:~$ sudo cp /lib/firmware/silead/silead_ts.fw /lib/firmware/silead/mssl1680.fw 
[sudo] password for lilithe: 
cp: cannot stat '/lib/firmware/silead/silead_ts.fw': No such file or directory
lilithe@signet:~$ 
```

eanbowman, 2019-08-31:
@onitake I have managed to compile the kernel and install the resulting deb files for Ubuntu Bionic including my changes.

https://gitlab.eanbowman.com/ean/ubuntu-bionic/merge_requests/1/diffs

The link above shows what I changed. It still works the same, with the taps registering for the whole screen, but they only map to the top left quarter and Y is flipped.

Am I missing something?

onitake, 2019-09-01:
Thats's promising!
It looks like you finally got the right firmware.

You still need to tweak the parameters though. Determine what range the X and Y coordinates have and put the correct values in. You can tune the values later with xinput_calibrator. Note that you can also specify minimum values.

As for the flipped Y axis - it looks like you should remove the touchscreen-inverted-y flag?

eanbowman, 2019-09-03:
```Calibrating standard Xorg driver "silead_ts"
	current calibration values: min_x=0, max_x=65535 and min_y=0, max_y=65535
	If these values are estimated wrong, either supply it manually with the --precalib option, or run the 'get_precalib.sh' script to automatically get it (through HAL).
	--> Making the calibration permanent <--
  copy the snippet below into '/etc/X11/xorg.conf.d/99-calibration.conf' (/usr/share/X11/xorg.conf.d/ in some distro's)
Section "InputClass"
	Identifier	"calibration"
	MatchProduct	"silead_ts"
	Option	"MinX"	"506"
	Option	"MaxX"	"31135"
	Option	"MinY"	"24555"
	Option	"MaxY"	"202"
	Option	"SwapXY"	"0" # unless it was already set to 1
	Option	"InvertX"	"0"  # unless it was already set
	Option	"InvertY"	"0"  # unless it was already set
EndSection
```
Hmm, maybe my values need to be more like this?

onitake, 2019-09-03:
If possible, put roughly correct values into you kernel patch. That will save others from needing to recalibrate later. Note that the value range doesn't necessarily to correspond with what Xinput gives you.

Try `evtest` to determine the full range of the digitiser.

eanbowman, 2019-09-03:
[evtest.txt](https://github.com/onitake/gsl-firmware/files/3569865/evtest.txt)
I did evtest and touched the bottom left, top left, top right, and bottom right corners.

eanbowman, 2019-09-03:
It works! It now maps to the whole screen. 

THIS is what worked for me: https://github.com/notro/fbtft/issues/445#issuecomment-354714529

Should I still bother submitting a kernel patch? I don't think it did anything...

eanbowman, 2019-09-03:
[image]

onitake, 2019-09-03:
Can you verify if the kernel is loading your "custom" firmware?
You should see a line like

> firmware: direct-loading firmware silead/gsl3670-insignia_flex11.fw

somewhere in dmesg output.

I think we should definitely prepare a patch so others with the same tablet get good support out of the box.
Since you already figured out the correct parameters, it shouldn't be too hard to transfer them to the DMI config, once we figure out if and why the override doesn't work.

From my experience, it's usually just the wrong DMI parameters being compared.

onitake, 2019-09-03:
Oh, and the second most common case of mismatches are trailing spaces or other unprintables.

eanbowman, 2019-09-04:
```
lilithe@signet:~$ sudo dmesg | grep Insignia
[sudo] password for lilithe: 
[    0.000000] DMI: Insignia NS-P11W7100/NS-P11W7100, BIOS 5.11 07/05/2016
lilithe@signet:~$ 
```

I don't think so. I'll try fixing the DMI detection. Did you see my diff above?

https://gitlab.eanbowman.com/ean/ubuntu-bionic/merge_requests/1/diffs

onitake, 2019-09-04:
Yes, I saw it.
But I don't see anything about the firmware in your dmesg output. Can you check again?

eanbowman, 2019-09-04:
[dmesg.txt](https://github.com/onitake/gsl-firmware/files/3574619/dmesg.txt)
Yeah, I can't find it either. Really strange since it's working.
```
[sudo] password for lilithe: 
[    0.138875] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    5.626202] iwlwifi 0000:01:00.0: loaded firmware version 29.1044073957.0 op_mode iwlmvm
```

onitake, 2019-09-05:
Ah... I think I know what's going on.
`request_firmware` only prints a warning when the firmware can't be found. Otherwise, you get debug messages. Can you run `dmesg -n 8` to also get debug output?

eanbowman, 2019-09-05:
[dmesgdebug.txt](https://github.com/onitake/gsl-firmware/files/3579477/dmesgdebug.txt)

AlexRudyak, 2023-05-01:
This post looks very promising, been 4 years, Is there a simpler guide I might follow ? or at least list the steps required ?

lilithebowman, 2023-05-03:
@AlexRudyak https://www.lilithebowman.com/blog/2019/09/07/how-to-get-the-touchscreen-working-in-xubuntu-linux-on-an-insignia-flex-11-6-tablet/ wrote it in a blog format a long time ago.

AlexRudyak, 2023-05-03:
> @AlexRudyak https://www.lilithebowman.com/blog/2019/09/07/how-to-get-the-touchscreen-working-in-xubuntu-linux-on-an-insignia-flex-11-6-tablet/ wrote it in a blog format a long time ago.

Your guide is by far the most extensive ive seen and I thank you very much for the hard work you put into it,

Some details are missing for me, for example the exact operating system used, xubuntu 16? 

I've found some forums that even said to recompile a custom kernel which I did but the compilation failed, unfortunately I'm still left with a non working copy (unfortunately for me of course xD)