onitake/gsl-firmware #118: ABOUILLY dualboot 13" tablet

Eterniitys, 2019-04-05:
Hi, I'm trying to make the touchscreen work on archlinux. The linux system partition doesn't seem to contain the gsl*.ko or any related fw file. I copied the linux firmwares from this repo and get the following logs:
'''
Apr 03 10:59:44 vibraninf kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Apr 03 10:59:44 vibraninf kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Apr 03 10:59:44 vibraninf kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Apr 03 10:59:44 vibraninf kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Apr 03 10:59:44 vibraninf kernel: silead_ts i2c-MSSL1680:00: Direct firmware load for silead/mssl1680.fw failed with error -2
Apr 03 10:59:44 vibraninf kernel: silead_ts i2c-MSSL1680:00: Firmware request error -2
'''
Do you have any idea on how to fix this?
Thanks in advance.

onitake, 2019-04-05:
The driver that ships with Linux is called `silead.ko` and needs special care to work with yet unsupported tablets.
Please read the README for instructions.

cafehaine, 2019-04-05:
Hi, I'm a friend of @Eterniitys.
We tried to find the firmware in the Android system partition that was preinstalled, however it doesn't seem to exist (as he said in his post).
The touchscreen does work under android.
We will try to get the seller to send us the windows driver to extract it form there, but somehow android doesn't seem to need it.

onitake, 2019-04-05:
Sorry, I misunderstood...
Perhaps the vendor compiled the driver directly into the kernel, or it's only in the initrd...

Did you check which modules are loaded in Android?