onitake/gsl-firmware #120: Jumper ezpad 6 s pro touchscreen Drivers

Paolo1985, 2019-04-29:
Hi, i've some problems with the touchscreen after a format and reinstalling of Windows 10 on my jumper ezpad 6 s pro ( bought last month ). I'm looking for the correct Drivers for Windows..I've seen that somone has updated recently the fw for Linux .. i Will be very greatfull if someone could update also the correct Drivers, working correctly, of the touch for Windows!!! ( The driver updated now on jumper official website now doesn't work correctly..) Thanks a lot🙏🙏🙏

Paolo1985, 2019-04-29:
The tablet Is the model of jumper ezpad 6 s pro with the 128 gb SSD, bought here:
https://m.banggood.com/it/Jumper-Ezpad-6S-Pro-Intel-Apollo-Lake-N3450-Quad-Core-6G-RAM-64GB-ROM64GB-SSD-11_6-Inch-Windows-10-Tablet-p-1191700.html?rmmds=search

Paolo1985, 2019-04-29:
@BenjaminRenz could you help me with Drivers? And also...how can I know of in my ezpad 6 s pro there Is the model b of touchscreen or the older?

BenjaminRenz, 2019-04-30:
See:
https://techtablets.com/forum/topic/drivers-for-ezpad-6-pro/page/3/

You can find out if it's the a or b model by trying the two drivers and see which one works (post on page 4) also the Accellerometer is 180° inverted.

Paolo1985, 2019-04-30:
@BenjaminRenz  thank you so much for the driver..you have saved my tablet!!!!
So, I want to help all other people that cuold have the same problems that me with a short summary story:
-I musted format my tablet and restoring windows becouse I had some problems with product key and so on..
•	after the restoring, the touch and the accelerometer doesn't work correcrtly
•	reading this and other forums, I've seen that Jumper has product the same tablet with 2 models of touchscreen and accelerometer, but in their website there is updated only one type of drivers, probably the older..
•	contacting jumper, they not response and contacting banggood assistence they told me that there was no possibility to send drivers to the customer, only the replacing of the tablet ( in my situation it was not possible becouse I use the tablet for work and I can't work without etc..)
•	at the and of the story, I've seen the post of @Benjamin with the drivers for the 2 different models, I've download them and resolved the problems using all the Jumper official website driver for ezpad 6s pro BUT ONLY FOR THE TOUCHSCREEN AND THE ACCELEROMETER I'VE USED THE DRIVERS FOR MODEL B POSTED BY OUR FRIEND AND YOU CAN FIND IN THE LINK OF THE PREVIOUS COMMENT OF THIS CHAT
That's all!!! I would like to give a very big thanks to Benjamin and his friend that give us the correct drivers and resolving this big rebus! I've just make some backups of drivers in my pc and hd for the fucture ( drivers downloaded and stored by sw like drive booster could not be the correct like in this case, so pay attention if you run that type of sw..after you colud have some problems like me)

onitake, 2019-04-30:
Since this issue is pretty much off-topic for gsl-firmware, I'm going to close it.
You can still post messages, but mind you that this ist not a discussion/support forum.
Thanks.

Other than that, would be willing to prepare a pull request with the Windows touchscreen driver and/or firmware you found? That may be a big help for others having the same problem.

closed 2019-04-30 by onitake

1453hilmi, 2024-01-14:
> Şoför için çok teşekkür ederim.. Tabletimi kaydettin!!! Bu yüzden, kısa bir özet hikaye ile benimle aynı sorunları yaşayan diğer tüm insanlara yardım etmek istiyorum: -Tabletimi biçimlendirmem ve pencereleri geri yüklemem gerekiyor çünkü ürün anahtarı vb. ile ilgili bazı sorunlar yaşadım. • geri yüklemeden sonra, dokunma ve ivmeölçer doğru şekilde çalışmıyor • Bu ve diğer forumları okurken, Jumper'ın 2 model dokunmatik ekran ve ivmeölçer ile aynı tablete sahip olduğunu gördüm, ancak web sitelerinde yalnızca bir tür sürücü güncellendi, muhtemelen daha eski.. • jumper ile iletişime geçtiler, yanıt vermediler ve banggood yardımı ile iletişime geçtiler, bana müşteriye sürücü gönderme imkanı olmadığını, sadece tabletin değiştirildiğini söylediler (benim durumumda mümkün değildi çünkü tableti iş için kullanıyorum ve onsuz çalışamıyorum vb.) • hikayenin ve hikayesinde, 2 farklı model için sürücülerle birlikte gönderisini gördüm, onları indirdim ve ezpad 6s pro için tüm Jumper resmi web sitesi sürücüsünü kullanarak sorunları çözdüm AMA SADECE DOKUNMATIK EKRAN VE IVMEÖLÇER IÇIN ARKADAŞIMIZ TARAFINDAN YAYINLANAN B MODELI SÜRÜCÜLERINI KULLANDIM VE BU SOHBETIN ÖNCEKI YORUMUNUN BAĞLANTISINDA BULABILIRSINIZ Bu kadar!! Benjamin ve arkadaşına bize doğru sürücüleri verdikleri ve bu büyük rebus'u çözdükleri için çok teşekkür etmek istiyorum! Ben sadece pc ve hd için bazı sürücülerin yedeklerini yaptım (sürücü güçlendirici gibi sw tarafından indirilen ve saklanan sürücüler bu durumda olduğu gibi doğru olamaz, bu yüzden bu tür sw çalıştırırsanız, dikkat edin. Colud'dan sonra benim gibi bazı problemleriniz var)

I have the same 6 pro and touch more than one point.

1453hilmi, 2024-01-14:
I have the same 6 pro and touch more than one point.
/
/