onitake/gsl-firmware #123: V80 Plus firmware already present? Doesn't look like it's working fine

Lukypie, 2019-05-12:
Hi, I own an Onda V80 plus, and while doing all the steps in the README.... Searching into the kernel source I found out it should be already supported out the box!

And I also figured out it is also present on this github in /firmware/linux/silead/gsl3676-onda-v80-plus-v3.fw

However, after installing the firmware in /firmware/silead, despite being correctly loaded (kernel module silead btw) the touchscreen is not working as it should. 

Looks like extracting my firmware from the driver improved things, however it looks like it works only in portrait mode and not in landscape, how is it supposed to work with the flipped screen? 

Here is my firmware btw, my model is V80 Plus rev v3 already extracted with the tools in this repo
[firmware-linux.zip](https://github.com/onitake/gsl-firmware/files/3170532/firmware-linux.zip)

onitake, 2019-05-12:
What do you mean with "doesn't work in landscape mode"?
The touch screen doesn't care about screen orientation at all (and wouldn't even know about it!), it simply reports touch coordinates relative to the default orientation origin.
Any transformation of the coordinates would have to be done at a later stage.
I don't know how this is done exactly, but I would assume it's something to be done in Xorg or with an additional utility.

As for the replacement firmware, could you take a look, @jwrdegoede and see if it makes sense to replace it?
Perhaps we have to add a more precise DMI match for the OP's device and the one the other firmware was made for?

@Lukypie Can you post the output of `sudo dmidecode -t system` for your device (you might have to install the toll first)?

Lukypie, 2019-05-12:
Nevermind, I found out here https://github.com/onitake/gsl-firmware/issues/95 here that at the current state there is no way aside from manually changing xinput every time you switch the screen orientation, I thought since GNOME would detect the accelerometer, there would already be a way to do that automatically

Looks like something like this script is doing the job anyway in the meantime...
```
#!/bin/bash
while :
do

set -e

TOUCHSCREEN='silead_ts'
XDISPLAY='DSI1'
TRANSFORM='libinput Calibration Matrix'

X_SCALE=2.17
Y_SCALE=3.22
X_OFFSET=-0.045
Y_OFFSET=-0.004

if [ -z "$1" ]; then
        ROTATION=right
else
        ROTATION=$1
        fi

export DISPLAY=:1
        xrandr --output $XDISPLAY --rotate $ROTATION
        xinput set-prop 'silead_ts' --type=float "$TRANSFORM" 0 $Y_SCALE $Y_OFFSET $X_SCALE 0 $X_OFFSET 0 0 1
        xinput --map-to-output "$TOUCHSCREEN" "$XDISPLAY"
sleep 1
done
```

As for the firmware, the one I put here looks to work decently for me even though it's a bit different

Here is the system info

```
System Information
        Manufacturer: ONDA
        Product Name: V80 PLUS
        Version: Type1 - TBD by OEM
        Serial Number: Type1 - 123456
        UUID: 12345678-c8d0-45bd-a79f-54ea5fbd3d97
        Wake-up Type: Power Switch
        SKU Number: 0001
        Family: Tablet
```

jwrdegoede, 2019-05-13:
Touchscreen coordinations not being rotated is a known bug (regression) in GNOME-3.32, a fix for this is pending:
https://gitlab.gnome.org/GNOME/mutter/issues/581

Lukypie, 2019-05-13:
Guess I got lucky choosing arch as a distro for this tablet pc!

closed 2019-05-13 by Lukypie