onitake/gsl-firmware #125: Issue loading silead device in general?

AndroidNazi, 2019-06-07:
When I put my firmware for the RCA w101v2 on my computer and reboot, my dmesg log shows approximately this:
Silead_ts 12c-M5SL1680:00: 120-MSSL1680:00 supply vddio not found, using dummy regulator
Silead_ts 12c-MISSL 1680:00: 12c-MISSL 1680:00 supply avdd not found, using dummy regulator
Silead_ts 12c-MSSL1680:00: GPIO lookup for consumer power
Silead_ts 12c-MSSL1680:00: using ACPI for GPIO lookup
Silead_ts 12c-MSSL1680:00: using lookup tables for GPIO lookup
Silead_ts 12c-MSSL1680:00: lookup for GPIO power failed
Silead_ts 12c-MSSL1680:00: Silead chip ID: @xFFFF0000
silead_ts 12c-M5SL1680:00 Initialization error, status: 0xFF5A5A5A
How do I go about fixing these issues?

onitake, 2019-06-08:
Can you post your ACPI DSDT please?

AndroidNazi, 2019-06-18:
I'm not sure how. Can you give me a link to a guide?

onitake, 2019-06-19:
https://wiki.archlinux.org/index.php/DSDT#Recompiling_it_yourself
You only need to follow steps 1. and 2.