onitake/gsl-firmware #127: Firmware for Microtech 5-inch CTP

sergey-suloev, 2019-07-15:
Hello, guys,

I got a 5-inch LCD + CTP bundle manufactured by Microchip LCD Co. 

Here is the CTP specification https://microtech-lcd.com/product/5-inch-6-PIN-7201280-dots-tft.html and LCD is MTF0500HDI-22 listed here https://microtech-lcd.com/tft-lcd-module/

I want to use CTP with the default mainline kernel driver, silead.ko, which is requiring firmware file.

I would really appreciate if anyone give me a hint about steps to prepare the firmware file.

Thank you.

onitake, 2019-07-15:
First of all: This is just a panel and a digitiser. The pages you linked do not mention anything about the touchscreen or which controller should be used.

Second: If you have a choice, I recommend you stay away from Silead or Goodix controllers. They are among the most low-end you can get and require device-specific firmware to be shipped with the OS instead of the hardware. Might give you a quicker time-to-market for proprietary hardware, but is probably the worst choice for a generic computer.

Third: If you really must, contact Silead directly and ask them for documentation and their configuration and calibration tools. They will probably require signing an NDA.
The firmware contains calibration data and must be generated for your specific hardware. It cannot be simply taken from a different device.

closed 2019-07-15 by onitake

sergey-suloev, 2019-07-15:
@onitake thanks for answer
The controller is mentioned, it is GSL3670.

onitake, 2019-07-15:
Ah, you're right, sorry.

But my other comment still stands: You need to obtain the Silead configuration tools to generate a driver that contains the firmware. Perhaps Microtech can also provide these tools or a driver/firmware to you.

sergey-suloev, 2019-07-15:
@onitake  but  kernel already has the driver, I only need a firmware file.. why do I need "to generate a driver" ?  do you mean that their tools shall generate a binary driver ?

onitake, 2019-07-15:
Yes, these tools record calibration data from the digitiser, then generate a firmware from that calibration data and at last pack it into a Windows or Android driver. This is what tablet vendors normally deliver with their devices. I don't know if it's possible to generate only the raw firmware, I haven't had a change to look at those tools.

You should take the touchscreen vendor up on that the missing drivers; the screen is useless otherwise.

sergey-suloev, 2019-07-16:
@onitake  I contacted Microchip and they gave me  2 files, *.h and *.c. This is probably the driver. 
Is there a way to extract the firmware from the source code or I must build this driver first ?

onitake, 2019-07-16:
You can probably use https://github.com/onitake/gsl-firmware/blob/master/tools/untscfg to extract it, but you have to change TS_CFG_DATA to GSLX680_FW on line 33.

sergey-suloev, 2019-07-16:
@onitake  thanks, it worked