onitake/gsl-firmware #129: Firmware for Polaroid MIDW255

josepgb, 2019-07-21:
Hi, attached the working firmware for the Polaroid MIDW255, a Z3735F 10,1 inches tablet.
I got it from the SileadTouch.sys I obtained from Windows with double driver. I have it working with Linux Mint 19.1 and kernel 5.2 rc7.

https://mega.nz/#!HMdgVYQY!TtvldufMBW_FxR7GBO7ZX7HIfftmdLLGTwOa2jitHQg

Just installed the kernel, put the attached firmware in /lib/firmware/silead and touch works. Callibration and orientation are being made with this script

#!/bin/bash

set -e

TOUCHSCREEN='silead_ts'
OUTPUT='DSI1'

if [ -z "$1" ]; then
  ROTATION=right
else
  ROTATION=$1
fi

xrandr --output "$OUTPUT" --rotate "$ROTATION"
xinput --map-to-output "$TOUCHSCREEN" "$OUTPUT"

which I run at boot.

One weird issue I have, there are areas where touch does not work. They are vertical frames of almost 1CM. Where you put the finger there, nothing happens. I thougth it was a faulty screen, and I placed the motherboard in another tablet case, with another screen, and nothing is fixed, the same problem, in the same areas. The remaing area of the screen works fine.

What I could be missing?

Tks!

josepgb, 2019-07-22:
I imaged the same Windows tablet from where I obtained the firmware (same model). 
I loaded the image in the Linux tablet and touch screen works fine. No gaps.

Looks like the extracted firmware is not enough.

onitake, 2019-07-23:
This is correct, the firmware is not enough.
Panel information, like the active area and some other settings need to be supplied to the driver by other means.

The common way in the case of silead.ko is by adding a DMI override to the kernel. Please see the section [silead_ts](https://github.com/onitake/gsl-firmware/#silead_ts) in the readme on how to do this.

josepgb, 2019-07-23:
Thanks Gregor. The problem is that, at least on that kernel 5.2 rc7, I cannot find silead_dmi.c nor touchscreen_dmi.c. There is just an empty silead.h.

Any thoughts?

vinodkrishnanr, 2020-11-08:
I have the same problem. My kernel is 5.4.0-52-generic and i can't find silead_dmi.c nor touchscreen_dmi.c

onitake, 2020-11-08:
I'm not sure what you mean.
The file is there in the tree: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/touchscreen_dmi.c?h=v5.2-rc7&id=6fbc7275c7a9ba97877050335f290341a1fd8dbf

As for the Ubuntu kernel sources - I don't know where to find the source tree for that particular version, but it's in the unstable branch at the very least: https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/unstable/tree/drivers/platform/x86/touchscreen_dmi.c