onitake/gsl-firmware #130: Insignia Flex 11" touch all in lower right

eanbowman, 2019-08-16:
Touches on the Insignia Flex will only register if you tap the lower right of the screen. I can't seem to get xinput_calibrator to keep any settings, even if I edit /usr/share/X11/xorg.conf.d/99-calibration.conf

This is Xubuntu 18.04.

Am I missing something? Can we add the installation instructions to the README.md of each firmware?

eanbowman, 2019-08-16:
Now I have no touch after running the fwtool with the command in the insignia flex 11 folder.

```
lilithe@signet:~/Downloads/gsl-firmware/firmware/insignia/flex11$ sudo cp silead_ts.fw /lib/firmware/silead/mssl1680.fw
lilithe@signet:~/Downloads/gsl-firmware/firmware/insignia/flex11$ sudo chmod +r /lib/firmware/silead/mssl1680.fw
lilithe@signet:~/Downloads/gsl-firmware/firmware/insignia/flex11$ sudo modprobe usbtouchscreen
lilithe@signet:~/Downloads/gsl-firmware/firmware/insignia/flex11$ sudo journalctl  | grep silead
Aug 14 22:29:31 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 14 22:29:31 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 14 22:29:32 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Aug 14 22:29:32 signet kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Aug 14 22:29:32 signet kernel: silead_ts i2c-MSSL1680:00: Direct firmware load for silead/mssl1680.fw failed with error -2
Aug 14 22:29:32 signet kernel: silead_ts i2c-MSSL1680:00: Firmware request error -2
Aug 14 22:29:32 signet kernel: silead_ts: probe of i2c-MSSL1680:00 failed with error -2
Aug 14 22:54:14 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 14 22:54:14 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 14 22:54:14 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Aug 14 22:54:14 signet kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Aug 14 22:54:14 signet kernel: silead_ts i2c-MSSL1680:00: Direct firmware load for silead/mssl1680.fw failed with error -2
Aug 14 22:54:14 signet kernel: silead_ts i2c-MSSL1680:00: Firmware request error -2
Aug 14 22:54:14 signet kernel: silead_ts: probe of i2c-MSSL1680:00 failed with error -2
Aug 14 23:04:25 signet sudo[2049]:  lilithe : TTY=pts/0 ; PWD=/media/lilithe/XUBUNTU ; USER=root ; COMMAND=/usr/bin/apt-cache search silead
Aug 15 22:14:23 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 15 22:14:23 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 15 22:14:23 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Aug 15 22:14:23 signet kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Aug 15 22:14:23 signet kernel: silead_ts i2c-MSSL1680:00: Direct firmware load for silead/mssl1680.fw failed with error -2
Aug 15 22:14:23 signet kernel: silead_ts i2c-MSSL1680:00: Firmware request error -2
Aug 15 22:14:23 signet kernel: silead_ts: probe of i2c-MSSL1680:00 failed with error -2
Aug 15 22:25:35 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 15 22:25:35 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 15 22:25:35 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Aug 15 22:25:35 signet kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Aug 15 22:25:35 signet kernel: silead_ts i2c-MSSL1680:00: Direct firmware load for silead/mssl1680.fw failed with error -2
Aug 15 22:25:35 signet kernel: silead_ts i2c-MSSL1680:00: Firmware request error -2
Aug 15 22:25:35 signet kernel: silead_ts: probe of i2c-MSSL1680:00 failed with error -2
Aug 15 23:40:27 signet sudo[9135]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads ; USER=root ; COMMAND=/bin/mkdir /lib/firmware/silead
Aug 15 23:40:43 signet sudo[9163]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads ; USER=root ; COMMAND=/bin/cp /home/lilithe/Downloads/firmware.fw /lib/firmware/silead/
Aug 15 23:41:10 signet sudo[9172]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads ; USER=root ; COMMAND=/bin/cp /home/lilithe/Downloads/firmware.fw /lib/firmware/silead/mssl1680.fw
Aug 15 23:42:34 signet sudo[9217]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads ; USER=root ; COMMAND=/bin/cp /home/lilithe/Downloads/gsl1686-surftab-wintron70-st70416-6.fw /lib/firmware/silead/
Aug 15 23:43:23 signet sudo[9222]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads ; USER=root ; COMMAND=/sbin/modprobe silead
Aug 15 23:44:42 signet sudo[9249]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads ; USER=root ; COMMAND=/bin/cp /home/lilithe/Downloads/gsl1686-surftab-wintron70-st70416-6.fw /lib/firmware/silead/mssl1680.fw
Aug 15 23:45:14 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 15 23:45:14 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 15 23:45:14 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Aug 15 23:45:14 signet kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Aug 15 23:45:15 signet kernel: input: silead_ts as /devices/pci0000:00/808622C1:05/i2c-5/i2c-MSSL1680:00/input/input14
Aug 15 23:49:33 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 15 23:49:33 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 15 23:49:33 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
[... cut ...]
Aug 16 14:53:00 signet kernel: silead_ts i2c-MSSL1680:00: Firmware load error -121
Aug 16 14:53:00 signet kernel: silead_ts: probe of i2c-MSSL1680:00 failed with error -121
Aug 16 14:58:13 signet sudo[1863]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads/gsl-firmware/firmware/insignia/flex11 ; USER=root ; COMMAND=/bin/cp silead_ts.fw /lib/firmware/silead/mssl1680.fw
Aug 16 14:58:25 signet sudo[1866]:  lilithe : TTY=pts/0 ; PWD=/home/lilithe/Downloads/gsl-firmware/firmware/insignia/flex11 ; USER=root ; COMMAND=/bin/chmod +r /lib/firmware/silead/mssl1680.fw
```

eanbowman, 2019-08-16:
Note: Yes I rebooted after copying the resulting file to the ```/lib/tools/silead``` folder

eanbowman, 2019-08-16:
This is the output of fwtool:
```
lilithe@signet:~/Downloads/gsl-firmware/firmware/insignia/flex11$ ../../../tools/fwtool -c firmware.fw -m 1680 -w 1920 -h 1620 -t 10 -f yflip,track silead_ts.fw
Loading firmware.fw...
Got page 0x02
Got page 0x03
Got page 0x04
Got page 0x05
Got page 0x06
Got page 0x07
Got page 0x08
Got page 0x09
Got page 0x1e
Got page 0x1f
Got page 0x00
Got page 0x01
Got page 0x1a
Got page 0x1b
Got page 0x1c
Got page 0x1d
Got page 0x20
Got page 0x21
Got page 0x22
Got page 0x23
Got page 0x24
Got page 0x25
Got page 0x26
Got page 0x27
Got page 0x28
Got page 0x29
Got page 0x2a
Got page 0x2b
Got page 0x2c
Got page 0x2d
Got page 0x2e
Got page 0x2f
Got page 0x30
Got page 0x31
Got page 0x32
Got page 0x33
Got page 0x34
Got page 0x35
Got page 0x36
Got page 0x37
Got page 0x38
Got page 0x39
Got page 0x3a
Got page 0x3b
Got page 0x3c
Got page 0x3d
Got page 0x3e
Got page 0x3f
Got page 0x40
Got page 0x41
Got page 0x42
Got page 0x43
Got page 0x44
Got page 0x45
Got page 0x46
Got page 0x47
Got page 0x48
Got page 0x49
Got page 0x4a
Got page 0x4b
Got page 0x4c
Got page 0x4d
Got page 0x4e
Got page 0x4f
Got page 0x50
Got page 0x51
Got page 0x52
Got page 0x53
Got page 0x54
Got page 0x55
Got page 0x56
Got page 0x57
Got page 0x58
Got page 0x59
Got page 0x5a
Got page 0x5b
Got page 0x5c
Got page 0x5d
Got page 0x5e
Got page 0x5f
Got page 0x60
Got page 0x61
Got page 0x62
Got page 0x63
Got page 0x64
Got page 0x65
Got page 0x66
Got page 0x67
Got page 0x68
Got page 0x69
Got page 0x6a
Got page 0x6b
Got page 0x6c
Got page 0x6d
Got page 0x6e
Got page 0x6f
Got page 0x70
Got page 0x71
Got page 0x72
Got page 0x73
Got page 0x74
Got page 0x75
Got page 0x76
Got page 0x77
Got page 0x78
Got page 0x79
Got page 0x7a
Got page 0x7b
Got page 0x7c
Got page 0x7d
Got page 0x7e
Got page 0x7f
Got page 0x80
Got page 0x81
Got page 0x82
Got page 0x83
Got page 0x84
Got page 0x85
Got page 0x86
Got page 0x87
Got page 0x88
Got page 0x89
Got page 0x8a
Got page 0x8b
Got page 0x8c
Got page 0x8d
Got page 0x8e
Got page 0x8f
Got page 0x90
Got page 0x91
Got page 0x92
Got page 0x93
Got page 0x94
Got page 0x95
Got page 0x96
Got page 0x97
Got page 0x98
Got page 0x99
Got page 0x9a
Got page 0x9b
Got page 0x9c
Got page 0x9d
Got page 0x9e
Got page 0x9f
Got page 0xa0
Got page 0xa1
Got page 0xa2
Got page 0xa3
Got page 0xa4
Got page 0xa5
Got page 0xa6
Setting parameters...
Saving silead_ts.fw...
```
then I copy it into the firmware folder and reboot. I get that Firmware load error -121

eanbowman, 2019-08-16:
```
Aug 16 15:12:03 signet kernel: i2c /dev entries driver
Aug 16 15:12:03 signet kernel: i2c_designware 808622C1:06: I2C bus managed by PUNIT
Aug 16 15:12:07 signet kernel: axp20x-i2c i2c-INT33F4:00: AXP20x variant AXP288 found
Aug 16 15:12:07 signet kernel: axp20x-i2c i2c-INT33F4:00: AXP20X driver loaded
Aug 16 15:12:07 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 16 15:12:07 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 16 15:12:07 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Aug 16 15:12:07 signet kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Aug 16 15:12:07 signet kernel: silead_ts i2c-MSSL1680:00: Firmware load error -121
Aug 16 15:12:07 signet kernel: silead_ts: probe of i2c-MSSL1680:00 failed with error -121
Aug 16 16:15:58 signet kernel: i2c /dev entries driver
Aug 16 16:15:58 signet kernel: i2c_designware 808622C1:06: I2C bus managed by PUNIT
Aug 16 16:16:01 signet kernel: axp20x-i2c i2c-INT33F4:00: AXP20x variant AXP288 found
Aug 16 16:16:01 signet kernel: axp20x-i2c i2c-INT33F4:00: AXP20X driver loaded
Aug 16 16:16:01 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply vddio not found, using dummy regulator
Aug 16 16:16:01 signet kernel: silead_ts i2c-MSSL1680:00: Linked as a consumer to regulator.0
Aug 16 16:16:01 signet kernel: silead_ts i2c-MSSL1680:00: i2c-MSSL1680:00 supply avdd not found, using dummy regulator
Aug 16 16:16:01 signet kernel: silead_ts i2c-MSSL1680:00: Silead chip ID: 0x80360000
Aug 16 16:16:01 signet kernel: silead_ts i2c-MSSL1680:00: Firmware load error -121
Aug 16 16:16:01 signet kernel: i2c_designware 808622C1:02: i2c_dw_handle_tx_abort: lost arbitration
Aug 16 16:16:01 signet kernel: i2c_designware 808622C1:02: i2c_dw_handle_tx_abort: lost arbitration
Aug 16 16:16:01 signet kernel: silead_ts: probe of i2c-MSSL1680:00 failed with error -121
Aug 16 16:16:15 signet upowerd[1260]: did not recognise USB path /sys/devices/pci0000:00/808622C1:06/i2c-6/i2c-INT33F4:00/axp288_charger/power_supply/axp288_charger, please report
```

eanbowman, 2019-08-16:
I ran scanwindrv on SileadTouch.sys to produce the firmware_00.fw and put that in the firmware folder with the right name and it works! Kind-of!

It only maps to half the screen and the vertical axis is flipped but it works!
https://www.youtube.com/watch?v=erkn7wfRQAs

I also reset my BIOS settings to default to make sure that was not interfering.

eanbowman, 2019-08-16:
Attempt to fix orientation of touches:

```
AUTHORS.md      fw_extractor  scanwindrv       sileadtouch.PNF  untscfg
autoconvert     fwtool        Silead_loc.txt   SileadTouch.sys
Firmware        LICENSE       SileadTouch.cat  silead_ts.fw
firmware_00.fw  README.md     sileadtouch.inf  unscramble
lilithe@signet:~/Downloads/flex11_6_2017$ ./fwtool -c firmware_00.fw -m 1680 -w 1920 -h 1620 -t 10 -f yflip,track silead_ts.fw
Loading firmware_00.fw...
Got page 0x02
Got page 0x03
Got page 0x04
Got page 0x05
Got page 0x06
Got page 0x07
Got page 0x08
Got page 0x09
Got page 0x1e
Got page 0x1f
Got page 0x00
Got page 0x01
Got page 0x1a
Got page 0x1b
Got page 0x1c
Got page 0x1d
Got page 0x20
Got page 0x21
Got page 0x22
Got page 0x23
Got page 0x24
Got page 0x25
Got page 0x26
Got page 0x27
Got page 0x28
Got page 0x29
Got page 0x2a
Got page 0x2b
Got page 0x2c
Got page 0x2d
Got page 0x2e
Got page 0x2f
Got page 0x30
Got page 0x31
Got page 0x32
Got page 0x33
Got page 0x34
Got page 0x35
Got page 0x36
Got page 0x37
Got page 0x38
Got page 0x39
Got page 0x3a
Got page 0x3b
Got page 0x3c
Got page 0x3d
Got page 0x3e
Got page 0x3f
Got page 0x40
Got page 0x41
Got page 0x42
Got page 0x43
Got page 0x44
Got page 0x45
Got page 0x46
Got page 0x47
Got page 0x48
Got page 0x49
Got page 0x4a
Got page 0x4b
Got page 0x4c
Got page 0x4d
Got page 0x4e
Got page 0x4f
Got page 0x50
Got page 0x51
Got page 0x52
Got page 0x53
Got page 0x54
Got page 0x55
Got page 0x56
Got page 0x57
Got page 0x58
Got page 0x59
Got page 0x5a
Got page 0x5b
Got page 0x5c
Got page 0x5d
Got page 0x5e
Got page 0x5f
Got page 0x60
Got page 0x61
Got page 0x62
Got page 0x63
Got page 0x64
Got page 0x65
Got page 0x66
Got page 0x67
Got page 0x68
Got page 0x69
Got page 0x6a
Got page 0x6b
Got page 0x6c
Got page 0x6d
Got page 0x6e
Got page 0x6f
Got page 0x70
Got page 0x71
Got page 0x72
Got page 0x73
Got page 0x74
Got page 0x75
Got page 0x76
Got page 0x77
Got page 0x78
Got page 0x79
Got page 0x7a
Got page 0x7b
Got page 0x7c
Got page 0x7d
Got page 0x7e
Got page 0x7f
Got page 0x80
Got page 0x81
Got page 0x82
Got page 0x83
Got page 0x84
Got page 0x85
Got page 0x86
Got page 0x87
Got page 0x88
Got page 0x89
Got page 0x8a
Got page 0x8b
Got page 0x8c
Got page 0x8d
Got page 0x8e
Got page 0x8f
Got page 0x90
Got page 0x91
Got page 0x92
Got page 0x93
Got page 0x94
Got page 0x95
Got page 0x96
Got page 0x97
Got page 0x98
Got page 0x99
Got page 0x9a
Got page 0x9b
Got page 0x9c
Got page 0x9d
Got page 0x9e
Got page 0x9f
Got page 0xa0
Got page 0xa1
Got page 0xa2
Got page 0xa3
Got page 0xa4
Got page 0xa5
Got page 0xa6
Got page 0xa7
Setting parameters...
Saving silead_ts.fw...
lilithe@signet:~/Downloads/flex11_6_2017$ sudo cp silead_ts.fw /lib/firmware/silead/mssl1680.fw
[sudo] password for lilithe: 
lilithe@signet:~/Downloads/flex11_6_2017$ sudo reboot
```

eanbowman, 2019-08-16:
That last attempt failed again with ``` error -121```

eanbowman, 2019-08-16:
@noizef
Do you know where I should go from here? I was able to extract the firmware from a SYS file but everything is upside down. If I run fwtool on that firmware it stops working with error 121.

eanbowman, 2019-08-16:
```lilithe@signet:~$ xinput_calibrator 
Calibrating standard Xorg driver "silead_ts"
	current calibration values: min_x=0, max_x=65535 and min_y=0, max_y=65535
	If these values are estimated wrong, either supply it manually with the --precalib option, or run the 'get_precalib.sh' script to automatically get it (through HAL).
	--> Making the calibration permanent <--
  copy the snippet below into '/etc/X11/xorg.conf.d/99-calibration.conf' (/usr/share/X11/xorg.conf.d/ in some distro's)
Section "InputClass"
	Identifier	"calibration"
	MatchProduct	"silead_ts"
	Option	"MinX"	"683"
	Option	"MaxX"	"30993"
	Option	"MinY"	"24004"
	Option	"MaxY"	"177"
	Option	"SwapXY"	"0" # unless it was already set to 1
	Option	"InvertX"	"0"  # unless it was already set
	Option	"InvertY"	"0"  # unless it was already set
EndSection
lilithe@signet:~$ 
```

onitake, 2019-08-18:
Please don't use fwtool any more. It's for my older driver, as indicated in the readme.
silead.ko needs "vanilla" firmware that was simply extracted, but not converted with fwtool.