onitake/gsl-firmware #133: Extract firmware from system.img for Colorfly i898A

aannenko, 2019-10-13:
Hi there,

Any chance to get help on extracting a proper `silead_ts.fw` from [Colorfly_i898_3G_HY_V1.0.0.zip](https://mega.nz/#!o9YXHaiS!m6BfGR7yoI9Mvqf-xFknVz1vRI9BgNRJy02T1Hj1s4M) or [system.img_extracted.zip](https://mega.nz/#!kw5ESQLA!jeT2elqvyJR2kbjtLrZyIY23uo9RQ-zjS9Mv7iRvSxg) (Windows-friendly)? :)
I'm trying to get touchscreen working on my Colorfly i898A with Android-x86.

Big thanks in advance!

braewoods, 2020-02-28:
Do you still have the device?

aannenko, 2020-02-29:
Yes, two pieces of it actually. One working, with its original BIOS untouched, the other one with improperly flashed BIOS, hence – non-functional.

From: braewoods
Sent: Saturday, February 29, 2020 0:55
To: onitake/gsl-firmware
Cc: Andrii Annenko; Author
Subject: Re: [onitake/gsl-firmware] Extract firmware from system.img forColorfly i898A (#133)

Do you still have the device?

braewoods, 2020-02-29:
I can't find any kernel modules in the system image nor any reference to Silead. Does this actually have a Silead touchscreen?

aannenko, 2020-03-01:
I don't know that.

I know though that it uses [this touch screen](https://www.aliexpress.com/item/32804781506.html) (PB90JG1891 to be precise) with an EDT Polytouch controller FT5x06, for which I also found a set of drivers on Github: [1](https://github.com/focaltech-systems/drivers-input-touchscreen-FTS_driver), [2](https://github.com/UlrichBecker/EDT-Polytouch-linux-driver), [3](https://github.com/torvalds/linux/blob/master/drivers/input/touchscreen/edt-ft5x06.c)

As for the system image, I'm not familiar with kernel modules and don't know what to look for. That's why I initially asked for help :)

There's also a silead_ts.fw for a [very similar tablet](https://github.com/onitake/gsl-firmware/tree/master/firmware/onda/v891w) in this repo, though it seems to have a [totally different touch screen](https://www.aliexpress.com/item/32831039456.html).

aannenko, 2020-03-01:
Closed it accidentally.

braewoods, 2020-03-01:
Are you able to boot the device with a regular Linux distribution? One way to know is to see if the kernel complains about missing firmware when it loads the mainline silead driver. If it does then there is a detected silead touchscreen. Otherwise it means a different touchscreen is present. This is how I was able to work on support for the NES11.

aannenko, 2020-03-01:
Tried Ubuntu and MX Linux, both don't seem to be noticing any missing drivers. I guess it's not Silead then.

closed 2020-03-01 by aannenko

onitake, 2020-03-02:
Manufacturers change their touchscreen controllers often, sometimes even in the same product line. That's why some BIOSes contain multiple configurations in their DSDT. Can be a bit hard to find out which one is actually active.

The FT5x06 already has kernel support: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/input/touchscreen/edt-ft5x06.c

aannenko, 2020-03-03:
@onitake, so the controller support is there. I'm certain I have [FT5506eeg](http://www.zxlcd.com/issuance//uploadpic/2013813104837730.pdf).

[image]

Does it mean my touch screen should work or at least be recognized by the OS (it's Linux Mint now)?
I understand that this is not related to Silead question but I'll be grateful if you guys give me some direction on this.

onitake, 2020-03-04:
Is this an ARM device?
As far as I can see, the driver only has DeviceTree support but no ACPI.

It looks like the chip doesn't require any special firmware or vendor-specific initialisation parameters, so it should load automatically as long as the DeviceTree contains a mapping for it.
Did you check your DT for a match?

aannenko, 2020-03-05:
It's x86 - a cheap Chinese tablet with Intel Atom Z3735F and x64 UEFI.
I'll post its `xrandr` and `xinput list` during the weekend.

onitake, 2020-03-05:
In that case, you may be out of luck.
The driver only contains data structures for simple I2C (SMBUS?) and DeviceTree matching.
Since I2C buses don't provide a reliable way to autodetect devices, the driver will simply not autoload. See here for more info on how I2C drivers work: https://www.kernel.org/doc/Documentation/i2c/writing-clients

To support ACPI platforms (i.e. x86), you will need to extend the driver so it can be instantiated by an entry on the DSDT.
Look at this patch for a different driver, for example: https://lore.kernel.org/patchwork/patch/738267/
It (roughly) shows what you need to do to enable ACPI detection.