onitake/gsl-firmware #140: Unable to swap/invert touch axes, Onda V820W_32

nroskden, 2020-04-07:
I got working a tablet which works with the same driver of Onda V820W/32GB.  

The original driver extracted from Windows is functional, and the *mssl1680.fw* available here works the same way too. 
  
The mouse cursor moves in a tiny area, after a calibration matrix it clicks and drags on the entire screen.  
Command:  
`xinput  set-prop "silead_ts" "libinput Calibration Matrix" 2.4, 0.0, -0.001, 0.0, 3.7065, -0.0111, 0.0, 0.0, 1.0`
  
...but the axes are swapped or mirrored, depending on the orientation.
Apparently this axis settings can be done with libinput props or adding some lines in */etc/X11/xorg.conf.d/XX-XXXX.conf*, but nothing works - "silead_ts" hasn't options related to that in xinput/libinput.  

[image]

In the README of this repository there is a tool to make custom .fw files, but it is only for *gslx680-acpi* which is deprecated.

antoniobaiao, 2023-03-17:
I have the same problem, the touchscreen looks like inverted. I have tried a lot of things, but nothing works. Actually looks like the OS is not reading anything in the config file. I even tried firmware for another devices, but everyone has the same behaviour.

manotroll, 2023-12-31:
I have this tablet, how exactly do I load this firmware into Debian?

kairotlei, 2024-02-09:
Sorry if late to answer you all, but got it some weeks ago!

kairotlei, 2024-02-09:
>

kairotlei, 2024-02-09:
Sorry for being late, but finally got it some weeks ago!
All below done on Debian:  
- Get this repo and your Windows driver:
- `./unscramble SileadTouch.fw mssl1680.fw`
- Move the new fw to its directory.
- `sudo mkdir /usr/lib/firmware/silead && sudo mv mssl1680.fw /usr/lib/firmware/silead/.`
- Reboot so the driver gets loaded.
- The touch may work, but badly. Then:
- Get this: https://github.com/reinderien/xcalibrate
- `sudo ./xcalibrate`
- Follow the questions. At the end some config lines will echoed, along some instructions. At that moment the calibration should work, if it's fine keep reading to save it (it is temporary until system gets powered off). If not, try changing the "Use rotation?" answer option, or the number of calibration points. 
- To save settings, at first it may be confusing as it's explained on console output - just copypaste **your config** into a file as in the following picture:
- 
[image]

onitake, 2024-02-10:
Glad that you could get it working!

If possible, could you take the time and submit a kernel patch as well? This will make it much easier for others to use this tablet. Follow these instructions: https://github.com/onitake/gsl-firmware?tab=readme-ov-file#silead_ts
You can use the properties defined in https://www.kernel.org/doc/Documentation/devicetree/bindings/input/touchscreen/touchscreen.yaml to flip/swap the axes as needed.

Unfortunately, testing this requires recompiling the kernel and rebooting (or kexec'ing) into it.
Once that is done, please submit the kernel patch as explained in the readme, and open a pull request on this repo here, so the converted firmware can be added to https://github.com/onitake/gsl-firmware/tree/master/firmware/linux

One thing though - we already have a firmware there called "gsl1680-onda-v820w-32g.fw". Is this for a different device, or did the manufacturer change device IDs? The corresponding device parameters are here: https://elixir.bootlin.com/linux/v6.7/source/drivers/platform/x86/touchscreen_dmi.c#L669
And it matches the following DMI properties: https://elixir.bootlin.com/linux/v6.7/source/drivers/platform/x86/touchscreen_dmi.c#L1456
Maybe it's enough to just add a matching entry to `touchscreen_dmi_table`?

onitake, 2024-04-21:
Since you managed to get your device working, I'm going to close this issue.
Feel free to reopen if you still have problems.

closed 2024-04-21 by onitake