onitake/gsl-firmware #147: Hi10Pro Swapped axis for Windows

ZramZero, 2020-06-14:
Hi, 
I've change the touch screen of my Hi10pro, and I tried >30 different drivers but touchscreen axis are swaped. 
I can see drivers for Hi10pro with sapped and not swapped axis, but how can I install it for Windows 10 x64?  (because there is no inf file only fw)

onitake, 2020-06-15:
Please be specific: Do you need a firmware or driver for Linux or Windows?

ZramZero, 2020-06-15:
I need a driver for Windows, and I thought that it's possible to convert fw - sys+inf

onitake, 2020-06-15:
That may be possible, as the Windows drivers seem to be able to load firmware from a separate file.

But this is out of scope for the gsl-firmware repository. We know almost nothing about the Windows driver and how to modify it to work with different tablets. Please contact your hardware vendor for a Windows driver. Or, try asking on xda-developers or other mobile device hacking forums.

closed 2020-06-15 by onitake

ZramZero, 2020-06-15:
I'm a little bit confused I can see  firmware for H10 pro

https://github.com/onitake/gsl-firmware/tree/master/firmware/chuwi/hi10_pro
Axes swapped | No

and there is no way how to use it as a driver or upload to the controller from windows, right?

onitake, 2020-06-15:
Looking at some of the Windows drivers we have in the repo, there may be a way: See https://github.com/onitake/gsl-firmware/tree/master/firmware/chuwi/vi8/PG-HK70DR2503 for example.

I think some versions of the driver will read firmware from `c:\Windows\System32\Drivers\SileadTouch.fw` (or similar), but I don't think there's any way to inject device parameters, as those are compiled into the driver. I don't know how and where these parameters are stored, so even if the firmware loads, the calibration data may be totally wrong, axes are swapped or you may have other problems. But there's no harm in trying, I guess.

ZramZero, 2020-06-15:
Thank you for the explanation!