onitake/gsl-firmware #152: XVIEW Quantum Xenon: The pointer glitching on touch events..

matiasdelellis, 2020-10-08:
Hi everyone,
Got extract the firmware for this device. I don't know if I can share it here since it is one of many devices "To be filled by o.e.m".. :see_no_evil: :sweat_smile: :disappointed: 

Well .. The touch itself seems to work, but it seems to be very sensitive and vibrates/glit when I select something.

[image]

OS: Fedora 31
Kernel: 5.8.13

This is a common error? Any way to improve?
Thank you very much.

onitake, 2020-10-11:
There are two HID device properties that control jitter: touchscreen-fuzz-x and touchscreen-fuzz-y
If you experiment with them a bit, you should get much less random movement.

See here for an example how they're used: https://patchwork.kernel.org/patch/10835721/

I think there's also a way to set them from user space, but I don't know how.

matiasdelellis, 2020-10-11:
Hi @onitake 
Thank you very much for the info. It seems to be what I need.. :grimacing:,
...but note that I cannot identify the device by ACPI since his tables are wrong (Everything says "To be filled by oem").

Try passing the parameters on modprobe, but it ignores it.

```
[ 2601.508812] silead: unknown parameter 'touchscreen-fuzz-x' ignored
[ 2601.508813] silead: unknown parameter 'touchscreen-fuzz-y' ignored
```

There is another way?.

Thanks again.

onitake, 2020-10-11:
I really don't know...
@jwrdegoede Do you have an idea how to instantiate the module manually with custom parameters?