onitake/gsl-firmware #154: Detected input completely off for NES 11

Lootron, 2020-11-12:
Whew, it's been quite an adventure for me and I'm learning as I go, so I apologize if I'm poor at writing an issue. I'm new to this side of linux.
Long story short, I install android x86 on this thing and finally found the touchscreen driver, then "installed" (not sure if I did it properly, following someone else's example). However, the input inside android is completely inaccurate. At some touch points, its mirrored vertically, then others it's not. Touching and dragging anywhere will not bring the registered input anywhere past the left half of the screen. It's hard to describe the situation without a picture or a video. 

I was wondering if I completed the process incorrectly. This is the driver I used.

https://github.com/onitake/gsl-firmware/blob/master/firmware/nuvision/nes11/silead.fw

Please correct me if I failed to provide necessary details. I can provide them.

onitake, 2020-11-12:
It would be nice to know what tutorial you used and which driver.
The firmware in this is not very useful unless you also have the correct device configuration in your kernel.

On the other hand, the issues you're describing suggest that the firmware is the wrong one for your device.
Note that some manufacturers replace parts silently or source them from different manufacturers. This can cause issues when firmware/drivers don't match.

Lootron, 2020-11-12:
I actually fumbled my way through the same process this person did.

https://hackaday.io/project/83212-liberating-a-50-windows-tablet/log/116445-testing-the-touchscreen

It appears at the end he had a similar problem to mine, but that was a different device and he intentionally used the wrong firmware. However, mine referenced the same missing file as his, 'mssl1670.fw'. All I did was take the firmware file I linked, copied it to the proper directory, and renamed it to this. Surprisingly, it detected input right away without a reboot. I did go ahead and reboot anyway, with the same problem persisting.

I'm unsure how to affirm I have the correct device configuration in my kernel, nor am I educated on how to change it.

However, your wording suggests that drivers and firmware are separate things? Is this true, or am I misreading your post? I compared the configuration file for the Windows driver for my device (that I got off of my manufacturer's website). The file references the same file that I mentioned earlier, so I believe its the correct firmware, Thanks for your help so far. Again, I apologize for not really knowing what I'm doing. I've never done anything with drivers at this level. Feel free to write as if you were addressing a professional, I can go learn anything I don't understand.