onitake/gsl-firmware #178: Jittery Touchscreen

0x0000ff, 2021-10-03:
Hello there,

I've managed to get the touchscreen working on my x86 tablet (Viglen Connect 10) with the silead_ts driver, but the touch screen is very jittery when dragging windows and scrolling, and I don't think it can properly register GUI button presses. I see there's some 'touchscreen-fuzz' attributes that are used in the touchscreen_dmi source code that i've used, but that doesn't seem to remedy it. 

Perhaps there's some option to filter it out with hwdb rules libinput? I'm not sure where to look to see if there is.

Thanks,
0x0000ff

jwrdegoede, 2021-10-03:
On 10/3/21 5:59 PM, 0x0000ff wrote:
> I've managed to get the touchscreen working on my x86 tablet (Viglen Connect 10) with the silead_ts driver, but the touch screen is very jittery when dragging windows and scrolling, and I don't [...]

Is this with a charger connected to the tablet ?

Some (cheap/brandless) chargers generate a lot of electrical noise ontop of the 5V they are supplying and Silead touchscreens are quite sensitive to this.

Try using the touchscreen without the charger connected, and if that helps, try using a different charge / buy another charger. I've good luck with using the Dell 5V/2A chargers from the Dell Venue 8 pro models. These chargers seem to supply a nice clean 5V.

0x0000ff, 2021-10-03:
It's not charging and it still persists. Does touchscreen-fuzz help with remedying this?

jwrdegoede, 2021-10-03:
> It's not charging and it still persists. Does touchscreen-fuzz help with remedying this?

If this still happens when no charger is connected then touchscreen-fuzz might help a bit.

0x0000ff, 2021-10-03:
Ok. I have the coordinates for the fuzz set at 30. Is that maybe too high?

Thank you for working on the drivers btw.

jwrdegoede, 2021-10-03:
On 10/3/21 6:23 PM, 0x0000ff wrote:
> Ok. I have the coordinates for the fuzz set at 30. Is that maybe too high?

30 indeed sounds quite high, 5 - 8 or something like that is probably best.

0x0000ff, 2021-10-03:
Ok. I used another firmware file and it seems to have sorted it. I've kept the fuzz because it still flickers a bit. I'll try to upload my progress so far with some pull requests. Thanks for your help.

closed 2021-10-03 by 0x0000ff

Lehmaning, 2022-11-11:
Could you tell me how did you fix it in more detail? I have no idea about setting the touchscreen-fuzz...