onitake/gsl-firmware #187: I can't invert the X and Y axes.

hiagocmendes, 2022-05-14:
[Hello how are you?
I have a How Max Quad Tablet, from Brazil. The company does not provide the ROM and there is no other 100% compatible.
I ended up finding one that works 90%, but the touch is mirrored.
The resolution is 800(x) x 1280(y)
What remains is just to invert the axes for it to work 100%.
Can you help me?

Link of .ko file: https://cdn.discordapp.com/attachments/717027224020058172/974880998707855440/gslX680.ko

hiagocmendes, 2022-05-14:
I used the tool for windows `scanwindrv`, it extracted firmware_00.fw successfully. 
But when trying to use the command `fwtool -s -f xflip,yflip firmware_00.fw ` it returns the following error: "Loading xflip,yflip...
Can't load firmware xflip,yflip: Invalid file handle at fwtool line 104."

Could anyone help?

hiagocmendes, 2022-05-14:
I used the tool for windows scanwindrv, it extracted firmware_00.fw successfully.
But when trying to use the command fwtool -s -f xflip,yflip firmware_00.fw it returns the following error: "Loading firmware_00.fw....
Can't load firmware firmware_00.fw.: Invalid magic     at fwtool line 104."

Could anyone help?

hiagocmendes, 2022-05-18:
I'm just trying to reverse the touch axes that are incorrect.
I'm pretty noob on the subject and I'm trying to fix the tablet that doesn't have ROM available for Flash.

Can you give an example of how to compile the .ko file using your driver? I'm using Windows with WSL, but I don't know how to use the commands to generate the file very well.

I use the file gslX680.ko and touch works, but inverted on both axes.

onitake, 2022-05-20:
The build instructions for gslx680-acpi are in the README: https://github.com/onitake/gslx680-acpi#build-instructions
It's relatively straight-forward. You need to install a suitable toolchain (native or cross, depending on the target architecture), plus the sources of the exact kernel that you're targeting. Then, just run make, passing it the path to the kernel sources:
```shell
make KSRC=/path/where/you/unpacked/the/kernel/sources/for/your/target/device/and/os
```
It's a bit easier when you're compiling directly on the target device, simply install the package containing the kernel headers for the active kernel, then type `make` in the driver source code directory.
Which kernel headers package you need to install, depends on the Linux distribution you want to use and the target architecture.

I do not know anything about gslX680.ko. Where did you find that driver?