onitake/gsl-firmware #188: Talius Zaphyr 8004W/8005W

miklosakos, 2022-05-28:
I have a no-name spanish tablet where I got touch working but it only works in a tiny rectangle and the axis seem to be swapped. Tried recompiling the kernel module, using evtest to find out the resolution but none of those seemed to help. Tried xinput_calibrator as well but the changes don't seem to apply.
Sadly the OEM doesn't provide the drivers for the tablet however I was able to extract the functioning ones from the factory Windows installation.
Screen resolution: 1200x1920
Digitizer resolution: ????
Digitizer chip: GSL1680
Manufacturer: Talius Technologies S.L.
Model: Zaphyr 8005W (however DMI info indicates it's an 8004W
I attached the extracted drivers that work perfectly fine under Windows: [sileadtouch.inf_amd64_9d69a50d38792845.zip](https://github.com/onitake/gsl-firmware/files/8790440/sileadtouch.inf_amd64_9d69a50d38792845.zip)
)
Other topic that should be raised somewhere else but maybe you could clue me in: the bosch accelerometer seems to be flipped as well, works fine under windows as well.

onitake, 2022-05-28:
xinput_calibrator should normally help, but it's better to use it for finding our the boundaries, and then adding them to the kernel.

You can use the scripts in the tools/ directory to extract the firmware from the Windows driver.

I know absolutely nothing about accelerometer use on Linux, but this page indicates that there's at least support for one Bosch device: https://linux-sunxi.org/Accelerometer
Unfortunately, they only provide a DeviceTree example, which is useless on x86.
If the manufacturer has done a good job, there should be an entry in the ACPI DSDT of your tablet mentioning the sensor. In that case, the driver may simply not be picking up the entry and should be fixed. If there's nothing in the DSDT, you'll have to figure out how to load and configure the driver manually. You may also try your luck with a DSDT override. Here's a good starting point: https://wiki.archlinux.org/title/DSDT

miklosakos, 2022-05-31:
So some update: I did get the digitizer calibrated somewhat, however I did notice that on the right side it's skewed/angled so what i touch doesn't correspond to what the OS interpreted what I touched.
I have to boot up the tablet to get the exact calibration data, will do that later. Probably I'm messing up the calibration however I can't figure out how could I make it more accurate.