onitake/gsl-firmware #199: Fusion5 FWIN232 PRO

syco, 2022-11-22:
Hi,
I've extracted the firmware from windows, and I've successfully converted it, but I have one question:
the values I was expecting for w and h don't work, after a good bit of trial and error I found some values that work, but is there a proper way to find them?
Thanks

onitake, 2022-11-22:
You can set the values to the maximum (4095) and then use evtest or xinput_calibrator to find the boundary values.

There's unfortunately no way to find the exact values used by the panel and controller. At least, no known way. Perhaps there is a command to ask the controller or dump the limits from the firmware, but nobody has found out how yet...

syco, 2022-11-23:
Thanks, evtest did the job!

closed 2022-11-23 by syco