onitake/gsl-firmware #211: tools/fw_tool is not available

RickyMV, 2023-07-11:
Hello, I was trying to fetch a firmware file for my tablet, but then reading further instructions I've found out that this fw_tool that it mentions is not on the firmware/ directory. What can I do?

onitake, 2023-07-11:
That's an error in the readme. The tool is in `tools/fwtool`, not in `firmware/fwtool`.
Gonna correct that, thanks for reporting.

Also, are you sure you need it? I stopped maintaining gslx680-acpi a long time ago, and fwtool not useful for anything else.

RickyMV, 2023-07-11:
> Also, are you sure you need it? I stopped maintaining gslx680-acpi a long time ago, and fwtool not useful for anything else.

Oh, because I'm trying to run postmarketOS on an Allwinner tablet, but the touchscreen driver that was included in the image wasn't working. So after finding out the driver my tablet needs was gsl1680 and after researching I found this project. I don't know if there is a cleaner way to get that driver...

RickyMV, 2023-07-11:
off topic: first time using github issues oops

onitake, 2023-07-12:
The correct way is to use the driver that's included in the mainline Linux kernel (silead.ko).

If your device is already supported, all you need to do is download the ready-to-use firmware and it should work. See https://github.com/onitake/gsl-firmware/tree/master/firmware/linux for more info.

Unfortunately, adding support for a new device is much more work. The readme describes what you need to do: https://github.com/onitake/gsl-firmware#usage